Genesis-Embodied-AI/genesis-world #1038: [Bug]: difference between control_dofs_position and set_dofs_position

milong26, 2025-04-18:
### Bug Description

When Im trying to set a robot's dofs to given data, `set_dofs_position` is better, since `control_dofs_position` need sometime to reach given qpos.

But now I'm trying to grasp an object using merged robot. I can set_dofs_position of arm then need to grasp the object using hand gripper. Problem is that when arm moved after grasping, the gripper cannot left the object.

### Steps to Reproduce

1. I checked that the merged arm+hand can grasp and lift object when using `contro_dofs_position` to control arm by IK. so it's not friction or gripper problem.
2. Given data doesn't offer gripper data, it only gives 0 and 1 to show whether gripper is open or not.
3. The arm and gripper can work well when I `set_dofs_position` before grasp. I use `control_dofs_force` and do some `scene.step()` to let it grasp.  But when the arm lifted, the hand which i think can grasp object together failed. the object looks like slipped down :(

### Expected Behavior

Can I `set_dofs_position` to arm and keep the gripper lifting object up? I tried to use `control_dofs_position` but it needs run some scene.setp() after that to keep qpos near targeted one. :(

### Relevant log output

```shell

```

### Environment

linux(I don't think it's os problem)

### Release version or Commit ID

[[19baf7d](https://github.com/Genesis-Embodied-AI/Genesis/commit/19baf7d6a75f55afc0b0b8068f6bf672e012b65a)]

milong26, 2025-04-18:
Or is there any way for me to use `control_dofs_position` but need fewer steps for it to move to targeted qpos? Since  by using given qpos, they're close.(comprared to let robot control qpos from 0 to 100,my work need it control qpos from 0 to 1,1 to 2...and so on)

duburcqa, 2025-04-18:
I will take a look during the time (hopefully!).

milong26, 2025-04-18:
> I will take a look during the time (hopefully!).

Tell me if you need code (im afraid that my expression not clear)

duburcqa, 2025-05-11:
The issue is probably that the grip is not firm enough. You either need to increase the position gain of the gripper to make the controller more stiff, or offset the target position a bit more. For instance, you can use negative target position of the gripper of franka. This position will never be reached in practice because it is out of bounds, but it will increase the gripper force. I would recommend the first approach though, which is more intuitive and easier to tune.

closed 2025-05-11 by duburcqa