Genesis-Embodied-AI/genesis-world #1042: [Feature]: AMD GPU Support

MrCharlieJ200212, 2025-04-19:
### What feature or enhancement are you proposing?

Improve GPU select logic for supporting AMD GPU

### Motivation

Hi! Recently  I was reviewing issues and trying to get a method to run parallel sim on my AMD GPU.  I saw this https://github.com/Genesis-Embodied-AI/Genesis/issues/1024 
 It said  taichi is used for scene.step and Torch is used at init and for public data accessors. I have tested torch3.8nightly+rocm6.4 on my PC (9700x+9070xt) and it`s working.  AND I noticed this issue of taichi https://github.com/taichi-dev/taichi/issues/6434 because they have already introduced amdgpu backend into taichi and I have checked taichi code to confirm it. Is it possible to introduce amdgpu backend to genesis?

### Potential Benefit

Enable AMD gfx9\10\11\12 video card to run parallel sims

### What is the expected outcome of the implementation work?

- [ ]

MrCharlieJ200212, 2025-04-19:
https://github.com/taichi-dev/taichi/issues/6434 Some one has tested the AMDGPU backend (TI_ARCH=amdgpu) is already working with ROCm 6.0 on gfx1032 GPU

duburcqa, 2025-04-19:
Do you think it would improve performance over vulkan ?

duburcqa, 2025-04-19:
What I could do is automatically detect amd gpu and use the appropriate taichi and torch backends. But it would be nice to make sure it is worthwhile beforehand.

MrCharlieJ200212, 2025-04-19:
> Do you think it would improve performance over vulkan ?

I am sure rocm may reach better performance on 9070xt (have special ai cores) but not sure on other gpu(which i don`t have).
reference to https://www.chiphell.com/thread-2683824-1-1.html and https://github.com/ggml-org/llama.cpp/discussions/10879

MrCharlieJ200212, 2025-04-19:
Sorry https://llminfo.tech/posts/1iw9m8r/ this one should get a clearer picture(You might need translater).

duburcqa, 2025-04-19:
I can create a PR if you can run the benchmarks for both Vulkan and AMDGPU backends to make sure it is worth merging.

MrCharlieJ200212, 2025-04-19:
I`ll  try run taichi benchmark tomorrow

> I can create a PR if you can run the benchmarks for both Vulkan and AMDGPU backends to make sure it is worth merging.

duburcqa, 2025-04-19:
There are already Genesis benchmarks available that would be more representative. I will send you the command to run them.

duburcqa, 2025-04-19:
`pytest -v -m 'benchmarks' "./tests"` inside the root folder of genesis repository.

MrCharlieJ200212, 2025-04-19:
Thank you I will try tomorrow

duburcqa, 2025-04-19:
Here is the PR https://github.com/Genesis-Embodied-AI/Genesis/pull/1043 that enables AMDGPU / Torch RoCm backend.

MrCharlieJ200212, 2025-04-20:
It seems that test file cannot find gpu :
`========================================================================================================================= test session starts ==========================================================================================================================
platform linux -- Python 3.10.16, pytest-8.3.5, pluggy-1.5.0 -- /home/changyuan/anaconda3/envs/RL/bin/python
cachedir: .pytest_cache
Using --random-order-bucket=global
Using --random-order-seed=0

rootdir: /home/changyuan/桌面/Genesis
configfile: pyproject.toml
plugins: random-order-1.1.1, xdist-3.6.1
collected 80 items / 66 deselected / 14 selected                                                                                                                                                                                                                       

tests/test_rigid_benchmarks.py::test_cubes[gpu-8192-True-1-CG] SKIPPED (No GPU available on this machine)                                                                                                                                                        [  7%]
tests/test_rigid_benchmarks.py::test_cubes[gpu-8192-False-1-CG] SKIPPED (No GPU available on this machine)                                                                                                                                                       [ 14%]
tests/test_rigid_benchmarks.py::test_speed[gpu-30000-Newton-random] SKIPPED (No GPU available on this machine)                                                                                                                                                   [ 21%]
tests/test_rigid_benchmarks.py::test_speed[gpu-30000-CG-anymal_c] SKIPPED (No GPU available on this machine)                                                                                                                                                     [ 28%]
tests/test_rigid_benchmarks.py::test_cubes[gpu-8192-False-10-CG] SKIPPED (No GPU available on this machine)                                                                                                                                                      [ 35%]
tests/test_rigid_benchmarks.py::test_cubes[gpu-8192-True-10-Newton] SKIPPED (No GPU available on this machine)                                                                                                                                                   [ 42%]
tests/test_rigid_benchmarks.py::test_speed[gpu-30000-Newton-anymal_c] SKIPPED (No GPU available on this machine)                                                                                                                                                 [ 50%]
tests/test_rigid_benchmarks.py::test_cubes[gpu-8192-True-10-CG] SKIPPED (No GPU available on this machine)                                                                                                                                                       [ 57%]
tests/test_rigid_benchmarks.py::test_speed[gpu-30000-CG-random] SKIPPED (No GPU available on this machine)                                                                                                                                                       [ 64%]
tests/test_rigid_benchmarks.py::test_speed[gpu-30000-Newton-batched_franka] SKIPPED (No GPU available on this machine)                                                                                                                                           [ 71%]
tests/test_rigid_benchmarks.py::test_speed[gpu-30000-CG-batched_franka] SKIPPED (No GPU available on this machine)                                                                                                                                               [ 78%]
tests/test_rigid_benchmarks.py::test_cubes[gpu-8192-True-1-Newton] SKIPPED (No GPU available on this machine)                                                                                                                                                    [ 85%]
tests/test_rigid_benchmarks.py::test_cubes[gpu-8192-False-1-Newton] SKIPPED (No GPU available on this machine)                                                                                                                                                   [ 92%]
tests/test_rigid_benchmarks.py::test_cubes[gpu-8192-False-10-Newton] SKIPPED (No GPU available on this machine)                                                                                                                                                  [100%]

========================================================================================================================== slowest durations ===========================================================================================================================

(28 durations < 40s hidden.  Use -vv to show these durations.)
================================================================================================================= 14 skipped, 66 deselected in 24.83s ==================================================================================================================`

MrCharlieJ200212, 2025-04-20:
I`ve checked my environment and torch works fine? I am not sure the backend choosing logic is correct

duburcqa, 2025-04-20:
> I`ve checked my environment and torch works fine? I am not sure the backend choosing logic is correct

Indeed, you are right. Let me check what I can do about it.

duburcqa, 2025-04-20:
@MrCharlieJ200212 You can now specify the desired backend for running the benchmarks, ie.
```python
pytest -v -m 'benchmarks' --backend gpu ./tests
```

MrCharlieJ200212, 2025-04-20:
[Test result for vulkan.txt](https://github.com/user-attachments/files/19825767/Test.result.for.vulkan.txt)

Ah...... It seems that taichi developers support amdgpu by hardcoding so that it only support gfx10 card  and my card is newer than that.....
I also tried vulkan backend soming wrong happend. Although multible test cases failed (outcomes is pasted in the txt file above) but it reaches great performances on some cases like : 
`tests/test_rigid_benchmarks.py::test_cubes[8192-True-1-CG] 
True island 	| 1 cubes 	| <CG: 0> 	| 4,859,749.51 fps 	| 8192 envs
`
Thanks for your effort , and hopefully taichi developers can fix supporting problem in the future.

Oh! There are also some tips for amdgpu user (maybe worth to put into docs?). 

1. AMDGPU users must use Ubuntu24.04+rocm6.4+torch3.8nightly+mesa(RDVA) driver combination. 
2. Amdvlk driver doesn`t support vulkan backend of taichi (Can`t allocate memory)
3. Ubuntu22.04+rocm6.4 may casues vulkanapi ICD problems and sometimes ubuntu even doesn`t recognize gpu.

MrCharlieJ200212, 2025-04-20:
Maybe genesis should allow vulkan backend for taichi and cuda(rocm) for torch?

duburcqa, 2025-04-20:
> Maybe genesis should allow vulkan backend for taichi and cuda(rocm) for torch?

Why not, but I doubt it would bring any benefit :/ I will do it.

duburcqa, 2025-04-20:
done.

MrCharlieJ200212, 2025-04-20:
> done.

Thanks you! This will provide  a temporary  solution  for newer amdgpu users like me 🥰

duburcqa, 2025-04-20:
I'm afraid taichi is not actively developed anymore, and barely maintained. Do not expect your GPU to be properly supported anytime soon... Typically no Python 3.13 is available to this day. The idea it is took over the development of taichi at some point, or move to an alternative solution if possible (less likely I would say).

MrCharlieJ200212, 2025-04-20:
> I'm afraid taichi is not actively developed anymore, and barely maintained. Do not expect your GPU to be properly supported anytime soon... Typically no Python 3.13 is available to this day. The [...]

Alright.... Maybe I should get an NVIDIA GPU.... Thanks for your explain you could close this issue .....

closed 2025-04-20 by MrCharlieJ200212

duburcqa, 2025-04-20:
So in the end, do you think it is worth merging AMDGPU support ? I don’t think so. But torch cuda for vulkan is worth it I would say.

MrCharlieJ200212, 2025-04-21:
> So in the end, do you think it is worth merging AMDGPU support ? I don’t think so. But torch cuda for vulkan is worth it I would say.

Me too