Genesis-Embodied-AI/genesis-world #1044: [Bug]: Syntax error in commit in April 18

OwenOwl, 2025-04-19:
### Bug Description

Commit 3fba766, file genesis/utils/particle.py, line 23.
Double quotes were used in double quotes.

```
os.environ["LD_LIBRARY_PATH"] = f"{os.environ.get("LD_LIBRARY_PATH", "")}:{LD_LIBRARY_PATH}"
```

This syntax error prevents import.

### Steps to Reproduce

Simply running `import genesis as gs` crashes on the corresponding line.

### Expected Behavior

Expected to import successfully.

### Relevant log output

```shell

```

### Environment

- OS: Mac OS
- GPU/CPU: Apple M4

### Release version or Commit ID

Using 29e1ed5

Kashu7100, 2025-04-20:
Thank you. Fixing the issue.

duburcqa, 2025-04-20:
I already have a branch fixing this locally. I will push it soon.

closed 2025-04-20 by duburcqa