Genesis-Embodied-AI/genesis-world #1049: [Feature]: Enhanced Visualization Tools

KafuuChikai, 2025-04-20:
### What feature or enhancement are you proposing?

- Virtual sensor representations (e.g., camera frustums, lidar beams)
- Virtual targets and waypoints with customizable appearance
- Trajectory visualization with history trails and predicted paths
- Virtual coordinate frames and axes for spatial reference
- Customizable visual markers for important points or regions

### Motivation

Adding comprehensive visualization tools to the framework that would enable users to render various virtual elements critical for debugging and demonstration.

### Potential Benefit

This enhancement would provide several key benefits:

- **Improved debugging experience**: Visualizing sensor coverage and agent perception aids in troubleshooting
- **Enhanced training analysis**: Trajectory visualization helps identify patterns in agent behavior
- **Better demonstration capabilities**: Visual elements make presentations and videos more informative
- **Accelerated development**: Spatial references and virtual guides reduce time spent analyzing logs
- **Intuitive parameter tuning**: Visual feedback on waypoints and targets simplifies configuration

These visualization capabilities would greatly enhance both development workflows and result presentations, making it easier to understand agent behavior and sensor interactions during simulation.

### What is the expected outcome of the implementation work?

The implementation would deliver the following outcomes:

- **Fully integrated visualization API**: A comprehensive library of visualization functions that integrate seamlessly with the existing simulation environment
- **User-friendly interface**: Simple API calls to create and modify visual elements with minimal code changes to existing projects
- **Customizable visual elements**:
  - Color-coded trajectory visualization with adjustable history length
  - Transparent sensor coverage volumes (camera frustums, lidar range)
  - Coordinate frame axes with adjustable size and opacity
  - Target markers with customizable shapes and colors
  - Path visualizations with varying styles (solid, dashed, etc.)
- **Documentation and examples**: Comprehensive documentation with usage examples for common visualization scenarios

This enhancement would transform the framework into a more powerful tool for development, debugging, and presenting results, while maintaining its current performance characteristics for simulation.

### Additional information

### Example Visualizations
The visualization capabilities would be similar to those demonstrated in these examples:

### The visualization effects in Genesis:
[image]

### Simple demo for drones:
https://github.com/user-attachments/assets/1e0c0307-f4e2-4cb0-b6dc-8d42821d5b49

### Isaac Lab Projects:
https://github.com/user-attachments/assets/c93d754d-7571-476c-81d9-11b22351b71a

https://github.com/user-attachments/assets/d4a98e04-6289-4030-aaa2-3d9227f6edfa

https://github.com/user-attachments/assets/267dee3b-7cce-4f59-9008-8b58c65c802a

KafuuChikai, 2025-04-20:
This feature request is related to the following existing issues:

1. [[Feature]: Visualizing Marking Objects in the Preview Window Without Affecting Depth or Semantic Segmentation Maps Acquired by the Camera #942](https://github.com/Genesis-Embodied-AI/Genesis/issues/942)
2. [[Feature]: Add Arrows for Command Velocities #975](https://github.com/Genesis-Embodied-AI/Genesis/issues/975)

duburcqa, 2026-03-22:
@Mehak261124 What are you trying to solve here? You are mentioning issues that have already been addressed. Genesis already supports markers only visible to the interactive viewer and cameras flagged as debug.

duburcqa, 2026-03-23:
Improving visualization is not a priority at the moment. You can extended what existing to support debug draw of much objects as you suggest, but you don’t need a high-level marker drawing API for this. Every manual draw is already adding markers only.