Genesis-Embodied-AI/genesis-world #1055: [Bug]: CUDA Error CUDA_ERROR_OUT_OF_MEMORY

guntgbtgv, 2025-04-21:
### Bug Description

I've loaded unitree g1 humanoid urdf for locomotion example code, and this happened. When I run the go2 urdf exmalple, it is fine. I tried loading anymal_c.urdf, it was also fine. 

### Steps to Reproduce

If possible, provide a script triggering the bug, e.g.
```python
# code snippet triggering the bug
import genesis as gs
gs.init()
scene = gs.Scene(show_viewer=False)
broken = scene.add_entity(
    gs.morphs.MJCF(file='attachement.xml'),
)
scene.build()
```
Make sure to attached any needed assets (here `attachement.xml`)!

### Expected Behavior

training starts without CUDA error

### Relevant log output

```shell
21:28:04.869] [CoACD] [info] Processing [12.5%]
[2025-04-21 21:28:04.869] [CoACD] [info] Processing [0.0%]
[2025-04-21 21:28:04.869] [CoACD] [info] Processing [62.5%]
[2025-04-21 21:28:04.869] [CoACD] [info] Processing [87.5%]
[2025-04-21 21:28:04.869] [CoACD] [info] Processing [25.0%]
[2025-04-21 21:28:04.869] [CoACD] [info] Processing [75.0%]
[2025-04-21 21:28:04.880] [CoACD] [info] Processing [100.0%]
[2025-04-21 21:28:04.880] [CoACD] [info]  - Merge Convex Hulls
[2025-04-21 21:28:05.169] [CoACD] [info] Compute Time: 9.151077202999886s
[2025-04-21 21:28:05.169] [CoACD] [info] # Convex Hulls: 29
[E 04/21/25 21:28:17.863 8096] [cuda_driver.h:operator()@92] CUDA Error CUDA_ERROR_OUT_OF_MEMORY: out of memory while calling malloc_async_impl (cuMemAllocAsync)

Traceback (most recent call last):
  File "/home/kist/Documents/genesis-rl-test/g1_train.py", line 218, in <module>
    main()
  File "/home/kist/Documents/genesis-rl-test/g1_train.py", line 208, in main
    env = G1Env(
  File "/home/kist/Documents/genesis-rl-test/g1_env.py", line 67, in __init__
    self.scene.build(n_envs=num_envs)
  File "/home/kist/Documents/genesis-rl-test/Genesis/genesis/utils/misc.py", line 62, in wrapper
    return method(self, *args, **kwargs)
  File "/home/kist/Documents/genesis-rl-test/Genesis/genesis/engine/scene.py", line 591, in build
    self._sim.build()
  File "/home/kist/Documents/genesis-rl-test/Genesis/genesis/engine/simulator.py", line 172, in build
    solver.build()
  File "/home/kist/Documents/genesis-rl-test/Genesis/genesis/engine/solvers/rigid/rigid_solver_decomp.py", line 205, in build
    self._init_constraint_solver()
  File "/home/kist/Documents/genesis-rl-test/Genesis/genesis/engine/solvers/rigid/rigid_solver_decomp.py", line 1309, in _init_constraint_solver
    self.constraint_solver = ConstraintSolver(self)
  File "/home/kist/Documents/genesis-rl-test/Genesis/genesis/engine/solvers/rigid/constraint_solver_decomp.py", line 26, in __init__
    + np.logical_not(np.isinf(self._solver.dofs_info.limit.to_numpy()[:, 0])).sum()
  File "/home/kist/miniconda3/envs/genesis_env/lib/python3.10/site-packages/taichi/lang/util.py", line 351, in wrapped
    return func(*args, **kwargs)
  File "/home/kist/miniconda3/envs/genesis_env/lib/python3.10/site-packages/taichi/lang/matrix.py", line 1271, in to_numpy
    matrix_to_ext_arr(self, arr, as_vector)
  File "/home/kist/miniconda3/envs/genesis_env/lib/python3.10/site-packages/taichi/lang/kernel_impl.py", line 1113, in wrapped
    return primal(*args, **kwargs)
  File "/home/kist/miniconda3/envs/genesis_env/lib/python3.10/site-packages/taichi/lang/kernel_impl.py", line 1043, in __call__
    key = self.ensure_compiled(*args)
  File "/home/kist/miniconda3/envs/genesis_env/lib/python3.10/site-packages/taichi/lang/kernel_impl.py", line 1011, in ensure_compiled
    self.materialize(key=key, args=args, arg_features=arg_features)
  File "/home/kist/miniconda3/envs/genesis_env/lib/python3.10/site-packages/taichi/lang/kernel_impl.py", line 637, in materialize
    self.runtime.materialize()
  File "/home/kist/miniconda3/envs/genesis_env/lib/python3.10/site-packages/taichi/lang/impl.py", line 471, in materialize
    self.materialize_root_fb(not self.materialized)
  File "/home/kist/miniconda3/envs/genesis_env/lib/python3.10/site-packages/taichi/lang/impl.py", line 406, in materialize_root_fb
    root.finalize(raise_warning=not is_first_call)
  File "/home/kist/miniconda3/envs/genesis_env/lib/python3.10/site-packages/taichi/_snode/fields_builder.py", line 170, in finalize
    return self._finalize(raise_warning, compile_only=False)

[Genesis] [21:28:17] [ERROR] RuntimeError: [cuda_driver.h:operator()@92] CUDA Error CUDA_ERROR_OUT_OF_MEMORY: out of memory while calling malloc_async_impl (cuMemAllocAsync)
```

### Environment

- OS: Ubuntu 20.04 
- GPU/CPU:  RTX 4070 SUPER
- GPU-driver version: 550.90.07
- CUDA / CUDA-toolkit version 12.4

### Release version or Commit ID

bbea562

AliBehdar, 2025-04-21:
I think maybe there would be a issue with your loaded unitree g1 humanoid

duburcqa, 2025-04-21:
Can you share a complete reproduction script (including all the assets in zip file)?

You could try disable self collision, I'm not sure why yet but it tends to lower memory consumption:
```python
    scene = gs.Scene(
        [...]
        rigid_options=gs.options.RigidOptions(
            [...]
            enable_self_collision=False,
        ),
        [...]
    )
```

duburcqa, 2025-04-21:
For G1, I would recommend disabling convex decomposition, which is mostly irrelevant and significantly increasing the memory footprint cost and runtime cost:
```python
    robot = scene.add_entity(
        morph=gs.morphs.URDF(
            file=[...],
            decompose_error_threshold=float("inf"),
        ),
    )
```

guntgbtgv, 2025-04-22:
I get the same error when I use the example locomotion code, the only line I changed is following:

[g1_29dof_rev_1_0.zip](https://github.com/user-attachments/files/19842907/g1_29dof_rev_1_0.zip)

```python
        self.robot = self.scene.add_entity(
            gs.morphs.URDF(
                # file="urdf/go2/urdf/go2.urdf",
                file="g1_description/g1_29dof_rev_1_0.urdf",
                pos=self.base_init_pos.cpu().numpy(),
                quat=self.base_init_quat.cpu().numpy(),
            ),
        )
```
I'll try the workarounds you've recommended

guntgbtgv, 2025-04-22:
@duburcqa I disabled convex deomposition and it worked! disabling self collision did not work.  I think disabling convex decomposition needs to be documented in the API docs, since I don't see 'decompose_error_threshold' argument.

duburcqa, 2025-04-22:
Glad to hear it worked. I will disable it by default for articulated robots, as it is generally useless in these cases.

> I think disabling convex decomposition needs to be documented in the API docs, since I don't see 'decompose_error_threshold' argument.

Hum... This is strange because it is properly documented:
```python
    decompose_nonconvex: bool, optional
        This parameter is deprecated. Please refers to 'convexify' and 'decompose_error_threshold' instead.
    decompose_error_threshold : bool, optional:
        Skip decompose if the relative difference between the volume of original mesh and its convex hull is lower than
        this threashold. 0.0 to enforce decomposition, float("inf") to disable it completly. Defaults to 0.15 (15%).
    coacd_options : CoacdOptions, optional
        Options for configuring coacd convex decomposition. Needs to be a `gs.options.CoacdOptions` object.
```