Genesis-Embodied-AI/genesis-world #1072: [Bug]: FEM and Rigid Entity Collision Bug

bahadir14, 2025-04-25:
### Bug Description

Hello everyone, I tried to model collision in between FEM and Rigid Entity. But, there seems to be a bug, which I couldnt solve. Do you guys have any idea about problem?

### Steps to Reproduce

import argparse

import numpy as np
import os
import torch

from dask.order import order
from mpmath import euler
from pygel3d.hmesh import minimize_dihedral_angle
from os import path
os.environ['PYOPENGL_PLATFORM'] = 'glx'
import genesis as gs

def main():
    parser = argparse.ArgumentParser()
    parser.add_argument("-v", "--vis", action="store_true", default=False)
    args = parser.parse_args()

    ########################## init ##########################
    gs.init(seed=0, precision="32", logging_level="debug")

    ########################## create a scene ##########################

    scene = gs.Scene(
        sim_options=gs.options.SimOptions(
            dt=4e-3,
            substeps=10,
        ),
        coupler_options=gs.options.CouplerOptions(
            rigid_fem=True,
        ),
        rigid_options=gs.options.RigidOptions(
            dt=0.01,
            gravity=(0,0,-9.8),
            enable_collision=True,
            constraint_solver=gs.constraint_solver.Newton,
            iterations=200,
            tolerance=1e-7,
            integrator=gs.integrator.implicitfast,
            use_contact_island=True,
        ),
        fem_options=gs.options.FEMOptions(
            damping=0.1,
        ),
        viewer_options=gs.options.ViewerOptions(
            camera_pos=(9.6, 4.8, 14.0),
            camera_lookat=(0.0, 0.0, 0.0),
            camera_fov=35,
            max_FPS=30,
        ),
        show_viewer=args.vis,
        vis_options=gs.options.VisOptions(
            #visualize_mpm_boundary=True,
        ),
    )
    cam = scene.add_camera(
        res=(1280, 800),
        pos=(5.5, 6.5, 3.2),
        lookat=(0.5, 1.5, 1.5),
        fov=30,
        GUI=True,
    )

    plane = scene.add_entity(
        material=gs.materials.Rigid(),
        morph=gs.morphs.URDF(file="urdf/plane/plane.urdf", fixed=True),
    )
    E, nu = 1.e7,0.3
    rho = 100.0

    obj1 = scene.add_entity(
        material=gs.materials.Rigid(
            rho=rho,
            friction=1.0,
        ),
        morph=gs.morphs.Sphere(
            pos=(0.0, 2.0, 0.5),
            radius=0.2,
        ),
        surface=gs.surfaces.Iron(),
    )

    obj2 = scene.add_entity(
        material=gs.materials.FEM.Elastic(
            E=E,
            nu=nu,
            rho=rho,
            model="linear",
        ),
        morph=gs.morphs.Box(
            pos=(0.0, 0.0, 0.5),
            size=(0.5,0.5,0.5),
        ),
        surface=gs.surfaces.Metal(
            color=(0.6, 1.0, 0.8, 1.0),
            vis_mode="visual",
        ),
    )

    scene.build()
    cam.start_recording()

    horizon = 1000
    for i in range(horizon):
            if i < 400:
                obj1.set_dofs_velocity([0.0, -2.0, 0.0,0.0,0.0,0.0])

            scene.step()
            cam.render()
    cam.stop_recording(save_to_filename="fem_rigid_entity.mp4", fps=60)

if __name__ == "__main__":
    main()

### Expected Behavior

I expect collision in between entities

### Screenshots/Videos

https://github.com/user-attachments/assets/a8a60b66-b6ab-4209-8b72-1716ebb2e83e

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 24.04 24H2]
- GPU/CPU [e.g. QUADRO RTX 4000, M3pr, Intel I9-9900k] (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version 

### Release version or Commit ID

version 0.2.1

duburcqa, 2025-04-25:
The video is not loading on my side.

bahadir14, 2025-05-05:
https://drive.google.com/file/d/1GpmkRl621ZluS__eDgaLgjRQbhEe13q5/view?usp=drive_link

duburcqa, 2025-05-07:
Yes I do! Here it is to make sure it does not disappear because the link gets broken at some point:

https://github.com/user-attachments/assets/e4af55fd-d934-4ba7-addf-c67029c56ee2

bahadir14, 2025-05-12:
Hi @duburcqa,

Do you have suggestion for problem I have been experiencing?

bahadir14, 2025-05-20:
Hi @YilingQiao,

I tried increasing ball size and it still gives the same response. I am looking forward to new coupling system.

By the way, I also tried using 2 fem entity within this script and I got below error:

"AttributeError: 'FEMEntity' object has no attribute '_n_surfaces'. Did you mean: 'n_surfaces'?"

Is there any solution for this?

YilingQiao, 2025-05-20:
can you show more error information?

bahadir14, 2025-05-21:
Hi @YilingQiao,

Here is the complete error message:

Traceback (most recent call last):
  File "/home/bahadir/Desktop/genesis/genesis/part_1/modelling/fem_entity/fem_try.py", line 127, in <module>
    main()
  File "/home/bahadir/Desktop/genesis/genesis/part_1/modelling/fem_entity/fem_try.py", line 93, in main
    obj2 = scene.add_entity(
           ^^^^^^^^^^^^^^^^^
  File "/home/bahadir/anaconda3/lib/python3.12/site-packages/genesis/utils/misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bahadir/anaconda3/lib/python3.12/site-packages/genesis/engine/scene.py", line 359, in add_entity
    entity = self._sim._add_entity(morph, material, surface, visualize_contact)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bahadir/anaconda3/lib/python3.12/site-packages/genesis/engine/simulator.py", line 146, in _add_entity
    entity = self.fem_solver.add_entity(self.n_entities, material, morph, surface)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bahadir/anaconda3/lib/python3.12/site-packages/genesis/engine/solvers/fem_solver.py", line 176, in add_entity
    s_start=self.n_surfaces,
            ^^^^^^^^^^^^^^^
  File "/home/bahadir/anaconda3/lib/python3.12/site-packages/taichi/lang/kernel_impl.py", line 1236, in _getattr
    return x(self)
           ^^^^^^^
  File "/home/bahadir/anaconda3/lib/python3.12/site-packages/genesis/engine/solvers/fem_solver.py", line 641, in n_surfaces
    return sum([entity.n_surfaces for entity in self.entities])
                ^^^^^^^^^^^^^^^^^
  File "/home/bahadir/anaconda3/lib/python3.12/site-packages/taichi/lang/kernel_impl.py", line 1236, in _getattr
    return x(self)
           ^^^^^^^
  File "/home/bahadir/anaconda3/lib/python3.12/site-packages/genesis/engine/entities/fem_entity.py", line 454, in n_surfaces
    return self._n_surfaces
           ^^^^^^^^^^^^^^^^
  File "/home/bahadir/anaconda3/lib/python3.12/site-packages/taichi/lang/kernel_impl.py", line 1221, in _getattr
    x = super(cls, self).__getattribute__(item)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bahadir/anaconda3/lib/python3.12/site-packages/taichi/lang/kernel_impl.py", line 1221, in _getattr
    x = super(cls, self).__getattribute__(item)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'FEMEntity' object has no attribute '_n_surfaces'. Did you mean: 'n_surfaces'?

bahadir14, 2025-05-27:
Hi @YilingQiao,

Still the same error

Traceback (most recent call last):
  File "/home/bahadir/Desktop/genesis/genesis/part_1/modelling/fem_entity/fem_try2.py", line 129, in <module>
    main()
  File "/home/bahadir/Desktop/genesis/genesis/part_1/modelling/fem_entity/fem_try2.py", line 102, in main
    obj3 = scene.add_entity(
           ^^^^^^^^^^^^^^^^^
  File "/home/bahadir/anaconda3/lib/python3.12/site-packages/genesis/utils/misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bahadir/anaconda3/lib/python3.12/site-packages/genesis/engine/scene.py", line 359, in add_entity
    entity = self._sim._add_entity(morph, material, surface, visualize_contact)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bahadir/anaconda3/lib/python3.12/site-packages/genesis/engine/simulator.py", line 146, in _add_entity
    entity = self.fem_solver.add_entity(self.n_entities, material, morph, surface)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bahadir/anaconda3/lib/python3.12/site-packages/genesis/engine/solvers/fem_solver.py", line 176, in add_entity
    s_start=self.n_surfaces,
            ^^^^^^^^^^^^^^^
  File "/home/bahadir/anaconda3/lib/python3.12/site-packages/taichi/lang/kernel_impl.py", line 1236, in _getattr
    return x(self)
           ^^^^^^^
  File "/home/bahadir/anaconda3/lib/python3.12/site-packages/genesis/engine/solvers/fem_solver.py", line 641, in n_surfaces
    return sum([entity.n_surfaces for entity in self.entities])
                ^^^^^^^^^^^^^^^^^
  File "/home/bahadir/anaconda3/lib/python3.12/site-packages/taichi/lang/kernel_impl.py", line 1236, in _getattr
    return x(self)
           ^^^^^^^
  File "/home/bahadir/anaconda3/lib/python3.12/site-packages/genesis/engine/entities/fem_entity.py", line 454, in n_surfaces
    return self._n_surfaces
           ^^^^^^^^^^^^^^^^
  File "/home/bahadir/anaconda3/lib/python3.12/site-packages/taichi/lang/kernel_impl.py", line 1221, in _getattr
    x = super(cls, self).__getattribute__(item)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bahadir/anaconda3/lib/python3.12/site-packages/taichi/lang/kernel_impl.py", line 1221, in _getattr
    x = super(cls, self).__getattribute__(item)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'FEMEntity' object has no attribute '_n_surfaces'. Did you mean: 'n_surfaces'?

YilingQiao, 2025-05-27:
```
File "/home/bahadir/anaconda3/lib/python3.12/site-packages/genesis/engine/entities/fem_entity.py", line 454, in n_surfaces
return self._n_surfaces
```
It's not the latest commit. Line454 in `fem_entity.py` is like that

https://github.com/Genesis-Embodied-AI/Genesis/blob/a313d457ea84b55be97b804c723cfe34c504e136/genesis/engine/entities/fem_entity.py#L454

Can you install the latest commit?

```
pip install -e .
```

bahadir14, 2025-05-27:
Hi @YilingQiao,

I installed latest version and successfully run it. Here is the latest video of sim.

https://drive.google.com/file/d/1t8Cx9Lmp_H0OapduFHPkanp3EfVioQm4/view?usp=sharing

But, there are still rigid-fem and fem-fem entity collision bugs. Is there any way to solve the problem?

YilingQiao, 2025-06-19:
@Libero0809 is working on it. We will add a better collision detection and resolution

bahadir14, 2025-08-08:
Hi @duburcqa and @Libero0809,

I still get same behavior shown in video that I uploaded. What should I do to overcome this problem?