Genesis-Embodied-AI/genesis-world #1076: [Feature]: Parallel motion planning soon ?

jgleyze, 2025-04-25:
### What feature or enhancement are you proposing?

Hey guys,
Hope you're doing well !

I was wondering if parallel motion planning was comming soon ? we're planning to use genesis to create datasets of different grasping motions however the current limitation is that we can't plan trajectories on multiple envs to do diverse grasps. Having this functionnality would greatly benefit genesis for creating dataset towards training stuff like behavioral cloning in manipulation scenarios. 

Have a nice day !

### Motivation

n/a

### Potential Benefit

n/a

### What is the expected outcome of the implementation work?

n/a

duburcqa, 2025-05-22:
Unfortunately it is not planned "soon". If obstacle avoidance is not an issue, I would suggest defining a straight line trajectory of the end effector in cartesian space, then create a few evenly spaced keypoints along the way (say 10cm apart from each others), use inverse kinematics to compute the configuration of the robot for each of these keypoints (do not forget to specify `init_qpos` with the solution from the previous keypoint), and then use linear interpolation in joint space between each successive keypoint. This works very well in practice and is very cheap to evaluate. But this approach is no longer applicable as soon as you have obstacles.

jgleyze, 2025-05-22:
In our case we may have very cluttered envs, after some discussions we in fact didn't really need the parallelization of ompl just being able to do it sequentially over all envs would be fine, you just pass the index of the env you re willing to do the planning in. Would that be feasible ?

MRiabov, 2025-06-02:
Hello,
Folks, it would be great to have either a description of how this could be achieved or something similar.

At the moment, the script that I've written is too wiggly - and I'm trying to draw a square trajectory around the franka arm such that it goes from something like (-0.75, -0.75) to (.75,-.75) to (.75,.75) to (.-75,.75) and back to (-.75, -.75) on a static Z height.

Could devs please implement a working stub version like the one @duburcqa suggested in his last comment and mark it as "linear_interpolation=True" in api? Or, the "use_iterative: bool = True" would also work well enough, simply by using for loops as a stub for the parallel implementation.

Technically I don't _need_ it for a demo, but it would be very useful to be able to train the RL policy with it.

https://github.com/user-attachments/assets/56a368a9-35a1-45db-8df7-a46d7fefcae4

Perhaps somebody will find useful:
Code:
```py

def execute_straight_line_trajectory(
    franka,
    scene,
    pos,
    quat,
    gripper,
    keypoint_distance=0.1,
    num_steps_between_keypoints=10,
    camera: Camera | None = None,
):
    """
    Execute a straight-line trajectory for a robot arm in Cartesian space.

    This implements a simple motion planning approach:
    1. Creates a straight line from current position to target in Cartesian space
    2. Generates evenly spaced keypoints along that line
    3. Computes IK for each keypoint
    4. Interpolates in joint space between keypoints

    Args:
        franka: The Franka robot entity
        scene: The simulation scene
        pos: Target position tensor with shape (num_envs, 3)
        quat: Target orientation quaternion tensor with shape (num_envs, 4)
        gripper: Gripper command tensor with shape (num_envs,) - 0 for close, 1 for open
        keypoint_distance: Distance between keypoints in meters (default: 0.1m or 10cm)
        num_steps_between_keypoints: Number of interpolation steps between keypoints

    Returns:
        None: The function directly executes the trajectory in the simulation
    """
    # Get the number of environments from the position tensor
    num_envs = pos.shape[0]
    device = pos.device

    # Get the current joint positions
    current_state = franka.get_dofs_position()
    end_effector = franka.get_link("hand")

    # Execute the trajectory for each environment separately (batch issue workaround)
    for env_idx in range(num_envs):
        # Get the current joint positions for this environment
        current_qpos = current_state[env_idx].clone()

        # Get the current end-effector pose
        # Create a full tensor with zeros for all environments
        all_envs_qpos = torch.zeros((num_envs, current_qpos.shape[0]), device=device)
        # Set the position for the current environment
        all_envs_qpos[env_idx] = current_qpos

        # # Get the position and orientation of the end effector for all environments
        # all_pos = [
        #     link.pos for link in franka.links
        # ]  # note: this is a hacky workaround. you can just get_links("hand"), although need to know the exact name.
        # all_quat = [link.quat for link in franka.links]
[... 262 lines cut ...]
            pos=pos_batch,
            quat=quat_batch,
            gripper=gripper_batch,
            keypoint_distance=0.05,  # 5cm between keypoints for smoother motion
            num_steps_between_keypoints=20,
            camera=cam,
        )

        # Render a frame after reaching the target
        cam.render()

    # Additional cooldown steps to allow robot to settle
    for _ in range(20):
        scene.step()
        cam.render()

    # Stop recording and save video
    cam.stop_recording(
        save_to_filename="/workspace/RepairsComponents-v0/motion_planning_test.mp4",
        fps=50,
    )

    print("Motion planning test completed. Video saved to motion_planning_test.mp4")

```

duburcqa, 2025-07-16:
Fixed!