Genesis-Embodied-AI/genesis-world #1079: [Feature]: Get world positions of all joints all at once

sr71684, 2025-04-27:
### What feature or enhancement are you proposing?

A way to obtain the world position of all joints in a robot, without having to use a python-based for loop to invoke `.get_pos()` on each `gs.RigidJoint`.

### Motivation

As a [e.g. civil engineering researcher working on flood simulation]
I want to [e.g. be able to simulate city-scale fluid / terrain simulations]
Because [e.g. this enables the efficient design and validation of flood protection structures].

### Potential Benefit

Vectorized operation would be significantly faster than a range-based for loop

### What is the expected outcome of the implementation work?

Instead of:
```
dofs_pos_w = torch.zeros((env.n_envs, len(env.dofs_idx), 3))
for i in range(len(env.dofs_idx)):
    dofs_pos_w[:, i, :] = env.robot.joints[i].get_pos()
```

should be able to do:
```
dofs_pos_w = env.robot.joints[:].get_pos()
```

duburcqa, 2025-04-27:
Please, bear in mind that `RigidJoint.get_pos` has been deprecated and will be removed in the future. Why are you trying to get the joint position ? It has been deprecated because it was deemed useless in practice and confusing. What you can do is access the link position or the complete configuration vector of the system. Do you this, you can use either `RigidLink.get_pos` for a single link, or `RigidEntity.get_links_pos` to get them all at once, or `RigidEntity.get_qpos` for the full configuration vector.

sr71684, 2025-04-27:
Understood, thanks for the quick reply. I'll give that a go.

duburcqa, 2025-04-27:
Why were you querying the joint positions ? As a proxy for the link positions are you really need the positions of the joints for some reason ? I’m curious about your use-case to know if my assumption that it was irrelevant is legitimate.

closed 2025-05-08 by duburcqa