Genesis-Embodied-AI/genesis-world #1083: [Bug]: Rendering issues

echoopen, 2025-04-28:
### Bug Description

Rendering issues: Depending on the viewing angle, the rendering may appear normal or abnormal

eg:

[image]

[image]

[image]

[image]

[image]

### Steps to Reproduce

import genesis as gs
gs.init(backend=gs.cuda)

scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())
franka = scene.add_entity(
    gs.morphs.MJCF(file='/home/aikou/project/Genesis/h1/unitree_ros/robots/h1_2_description/h1_2_handless.xml'),
)

scene.build()
while True:
    scene.step()

XML：https://github.com/unitreerobotics/unitree_ros/tree/master/robots/h1_2_description

### Expected Behavior

None

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 24.04, Windows 11 24H2]
- GPU/CPU [e.g. A100, RTX 4090, M3pr, Intel I9-9900k, Ryzen 5900x] (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

Name: genesis-world
Version: 0.2.1

duburcqa, 2025-04-28:
Can you fill the information relative to the environment ? Also, you should tray to install the latest version (not the one available PyPI) following the README instructions.

YilingQiao, 2025-04-29:
Is it because your asset has a plane and you also add a
```
    plane = scene.add_entity(
        gs.morphs.Plane(),
    )
```

Try the urdf file (which doesn't contain a plane), or remove the plane in your code?

closed 2025-04-29 by duburcqa