Genesis-Embodied-AI/genesis-world #1091: [Feature]: Hybrid & Custom material for URDF/XML joints

saulgooodman, 2025-04-30:
### What feature or enhancement are you proposing?

Custom materials for each joint when loading URDF/XML

### Motivation

As the title suggests, the end-effector of a rigid robotic arm is likely to be elastic, which makes the gripper very powerful. I saw in the documentation that when loading a URDF/XML file, only a single rigid material can be defined for the entire robotic arm. I'm not sure if it's possible to define the material for each joint later on. Thank you!

### Potential Benefit

It is common for the end-effector of a robotic arm to be flexible.

### What is the expected outcome of the implementation work?

When loading a URDF/XML file, it is possible to customize the material for each joint, such as defining the mass for each joint individually. I also hope to be able to define the material for each joint separately.

duburcqa, 2025-05-22:
Basically, different solvers are used for different materials. You should have a look to hybrid entities example(s).

duburcqa, 2025-09-27:
Related to https://github.com/Genesis-Embodied-AI/Genesis/issues/417