Genesis-Embodied-AI/genesis-world #1092: [Bug]: Crash on link-frame visualization in hello_genesis.py tutorial

trns1997, 2025-04-30:
### Bug Description

When attempting to display the Franka arm’s link frames in the `hello_genesis.py` tutorial, the renderer crashes with an assertion in Taichi’s Python API. Specifically, pressing i then l to toggle link-frame visualization triggers the following error:
```
AssertionError: __getitem__ cannot be called in Taichi-scope
```
This occurs deep inside the visualization rasterizer when iterating over contact data, indicating that a Python-side list or array indexing operation is being invoked within a Taichi kernel scope where it isn’t allowed. As a result, the simulation window freezes and the program exits with the traceback shown below:
```
Traceback (most recent call last):
  File "/home/thomas/Genesis/examples/tutorials/hello_genesis.py", line 20, in <module>
    scene.step()
  File "/home/thomas/Genesis/genesis/utils/misc.py", line 72, in wrapper
    return method(self, *args, **kwargs)
  File "/home/thomas/Genesis/genesis/engine/scene.py", line 726, in step
    self._visualizer.update(force=False, auto=refresh_visualizer)
  File "/home/thomas/Genesis/genesis/vis/visualizer.py", line 170, in update
    self._viewer.update(auto_refresh=auto)
  File "/home/thomas/Genesis/genesis/vis/viewer.py", line 141, in update
    buffer_updates = self.context.update()
  File "/home/thomas/Genesis/genesis/vis/rasterizer_context.py", line 808, in update
    self.update_contact(buffer_updates)
  File "/home/thomas/Genesis/genesis/vis/rasterizer_context.py", line 332, in update_contact
    for i_con in range(self.sim.rigid_solver.collider.n_contacts[batch_idx]):
          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^
  File "/home/thomas/miniconda3/envs/genesis/lib/python3.12/site-packages/taichi/lang/util.py", line 350, in wrapped
    assert in_python_scope(), f"{func.__name__} cannot be called in Taichi-scope"
AssertionError: __getitem__ cannot be called in Taichi-scope

[Genesis] [13:43:51] [ERROR] AssertionError: __getitem__ cannot be called in Taichi-scope
```
This bug prevents any link-frame visualization from rendering and forces the user to terminate the simulation.

### Steps to Reproduce

1. Open a terminal and navigate to the tutorial examples folder:
```
cd examples/tutorials
```

2. Launch the Hello Genesis tutorial:
```
python hello_genesis.py
```

3. Wait for the simulation window to appear.

4. Press i to toggle the info window.

5. Press l to enable the Franka arm’s link-frame visualization.

6. Simu Crashes

### Expected Behavior

When pressing i followed by l, the simulation should overlay the link frames of the Franka arm:
[image]

Produced the result by commenting out contents of `def update_contact(self, buffer_updates):`

### Relevant log output

```shell

```

### Environment

- OS: Ubuntu 24.04
- GPU/CPU: Intel(R) Core(TM) i5-9300HF CPU @ 2.40 GHz (using gs.cpu backend)
- GPU-driver version : N/A
- CUDA / CUDA-toolkit version : N/A

### Release version or Commit ID

branch main, commit 73c0e25846eb145ebeaa7746def6f1b1c258ec3d

### Additional Context

P.S. I’m excited to contribute to Genesis! I’m still exploring the codebase and getting familiar with the stack, so any pointers or guidance would be greatly appreciated.

duburcqa, 2025-04-30:
Could you try to update taichi to the latest release ?

duburcqa, 2025-04-30:
No it is working fine on my machine :/

trns1997, 2025-04-30:
Ah, interesting! I’ll spin up a Docker container and try to reproduce the issue there. I’ll report back once I have an update :) .

duburcqa, 2025-05-08:
@trns1997 Yes I would be great if you could add a note about this in the troubleshooting section of genesis-doc !

closed 2025-05-11 by duburcqa

trns1997, 2025-05-11:
@duburcqa I would like to keep this issue open, unless there were updates to resolve this? I cannot be the only one who is facing this issue, when I have time I'll try to figure out what's wrong with my system to cause this issue.

duburcqa, 2025-05-11:
I understand your point but non-actionnable issues cannot be left open indefinitely, otherwise the list of issues will just get longer and longer, and it will become impossible for the maintainer to sort out issues that must be investigated from the others. If you have anything new or any setup I could use to reproduce the issue I would be glad to try that and reopen if it works. Otherwise I'm afraid I have no other choice but to close this issue :/

trns1997, 2025-05-11:
Sounds fair. Noted, I'll look into it and if I have updates I'll post them here :)