Genesis-Embodied-AI/genesis-world #1099: [Bug]: genesis viewer crash when using with ros2 jazzy and docker

rocketman123456, 2025-05-03:
### Bug Description

i am using genesis with ros2 jazzy in docker, while other simulator like mujoco and pybullet work with viewer, genesis crash and gives:
```
Exception in thread Thread-2 (_init_and_start_app):
Traceback (most recent call last):
  File "/usr/lib/python3.12/threading.py", line 1073, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.12/threading.py", line 1010, in run
    self._target(*self._args, **self._kwargs)
  File "/home/rocketsky/.local/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 1161, in _init_and_start_app
    pyglet.clock.tick()
  File "/home/rocketsky/.local/lib/python3.12/site-packages/pyglet/clock.py", line 528, in tick
    return _default.tick(poll)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/rocketsky/.local/lib/python3.12/site-packages/pyglet/clock.py", line 270, in tick
    self.call_scheduled_functions(delta_t)
  File "/home/rocketsky/.local/lib/python3.12/site-packages/pyglet/clock.py", line 217, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/home/rocketsky/.local/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 955, in _time_event
    self.on_draw()
  File "/home/rocketsky/.local/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 654, in on_draw
    self._render()
  File "/home/rocketsky/.local/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 1098, in _render
    retval = renderer.render(self.scene, flags, seg_node_map=seg_node_map)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rocketsky/.local/lib/python3.12/site-packages/genesis/ext/pyrender/renderer.py", line 141, in render
    self._update_context(scene, flags)
  File "/home/rocketsky/.local/lib/python3.12/site-packages/genesis/ext/pyrender/renderer.py", line 899, in _update_context
    p._add_to_context()
  File "/home/rocketsky/.local/lib/python3.12/site-packages/genesis/ext/pyrender/primitive.py", line 359, in _add_to_context
    glVertexAttribPointer(0, 3, GL_FLOAT, GL_FALSE, FLOAT_SZ * 3, ctypes.c_void_p(0))
  File "/home/rocketsky/.local/lib/python3.12/site-packages/OpenGL/latebind.py", line 63, in __call__
    return self.wrapperFunction( self.baseFunction, *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rocketsky/.local/lib/python3.12/site-packages/OpenGL/GL/VERSION/GL_2_0.py", line 465, in glVertexAttribPointer
    contextdata.setValue( key, array )
  File "/home/rocketsky/.local/lib/python3.12/site-packages/OpenGL/contextdata.py", line 58, in setValue
    context = getContext( context )
              ^^^^^^^^^^^^^^^^^^^^^
  File "/home/rocketsky/.local/lib/python3.12/site-packages/OpenGL/contextdata.py", line 40, in getContext
    raise error.Error(
OpenGL.error.Error: Attempt to retrieve context when no valid context
```

### Steps to Reproduce

If possible, provide a script triggering the bug, e.g.
```python
# code snippet triggering the bug
import genesis as gs
gs.init()
scene = gs.Scene(show_viewer=False)
broken = scene.add_entity(
    gs.morphs.MJCF(file='attachement.xml'),
)
scene.build()
```
Make sure to attached any needed assets (here `attachement.xml`)!

### Expected Behavior

should work with no crash

### Relevant log output

```shell

```

### Environment

- OS: distrobox - Ubuntu 24.04
- GPU/CPU [e.g. A100, RTX 4090, M3pr, Intel I9-9900k, Ryzen 5900x] (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

from pip install genesis-world with latest version

duburcqa, 2025-05-03:
You should install the main branch (not the one available PyPI) following the README instructions.

duburcqa, 2025-05-03:
Ping me if the issue is not fixed after doing so and I will re-open this issue.

closed 2025-05-03 by duburcqa