Genesis-Embodied-AI/genesis-world #1104: [Bug]: Error on tutorial file

niyazikemer, 2025-05-05:
### Bug Description

Running tutorial `control_your_robot.py` produce error on line 45 regarding:
`motors_dof_idx = [franka.get_joint(name).dofs_idx_local[0] for name in joints_name]`
changing it to 
```python
motors_dof_idx = [franka.get_joint(name).dof_idx_local for name in joints_name]
```
solves it.
maybe the code base updated after this change?

### Steps to Reproduce

# running the code code snippet triggering the error
```python 
python control_your_robot.py
```

### Expected Behavior

Code run without an error.

### Relevant log output

```shell
[Genesis] [22:44:40] [INFO] ╭───────────────────────────────────────────────────────────────────╮
[Genesis] [22:44:40] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈│
[Genesis] [22:44:40] [INFO] ╰───────────────────────────────────────────────────────────────────╯
[Genesis] [22:44:40] [INFO] Running on [NVIDIA GeForce RTX 3090] with backend gs.cuda. Device memory: 23.66 GB.
[Genesis] [22:44:40] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [22:44:40] [INFO] Scene <1df3963> created.
[Genesis] [22:44:40] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <f76241f>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [22:44:40] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <e5eda73>, morph: <gs.morphs.MJCF(file='/home/niyazi/miniforge3/envs/electrogenesis/lib/python3.12/site-packages/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [22:44:40] [INFO] Building scene <1df3963>...
[Genesis] [22:44:42] [INFO] Compiling simulation kernels...
[Genesis] [22:44:45] [INFO] Building visualizer...
[Genesis] [22:44:46] [INFO] Viewer created. Resolution: 960×720, max_FPS: 60.
Traceback (most recent call last):
  File "/home/niyazi/Git/Genesis/examples/tutorials/control_your_robot.py", line 45, in <module>
    motors_dof_idx = [franka.get_joint(name).dofs_idx_local for name in joints_name]
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/niyazi/miniforge3/envs/electrogenesis/lib/python3.12/site-packages/taichi/lang/kernel_impl.py", line 1221, in _getattr
    x = super(cls, self).__getattribute__(item)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'RigidJoint' object has no attribute 'dofs_idx_local'. Did you mean: 'dof_idx_local'?
[Genesis] [22:44:46] [INFO] 💤 Exiting Genesis and caching compiled kernels...
```

### Environment

- OS: [Ubuntu 24.04 LTS]
- GPU/CPU [RTX 3090, AMD Ryzen™ 9 7950X × 32] (N/A if no GPU/CPU)
- GPU-driver version (535.230.02)   
- CUDA / CUDA-toolkit version: 12.2 

### Release version or Commit ID

23287ad859af528dfd951a2681fc074121ddfad7

niyazikemer, 2025-05-06:
Yes, it is the main branch. I believe I'd pasted the wrong commit id.  I've corrected it, you can check now.

duburcqa, 2025-05-06:
This commit is fairly "old" already. The example script is only compatible with the latest commit on the main branch. The API has changed since the commit you are using, so you cannot run a more recent example. You should run the version of the example that corresponds to your specific commit [here](https://github.com/Genesis-Embodied-AI/Genesis/tree/8ffd810d2f14bb91ad965ddbf285d96d794b857b/examples).

niyazikemer, 2025-05-06:
There may be a misunderstanding here—I’m not a GitHub expert (obviously 😊). I tried pasting the latest commit of the related file.
I assumed that a file in the main branch should work with the pip-installed version of Genesis (v0.2.1). Since I forked the main branch and ran the example file from it, I expected the example to either work as-is or require updates to match the latest Genesis release. Could you verify this?
If a update needed then I can make a pull request.

duburcqa, 2025-05-06:
> I assumed that a file in the main branch should work with the pip-installed version of Genesis (v0.2.1)

Yes. And this is a wrong assumption. Unfortunately, the only setup that is supposed to work is  installing the version of Genesis corresponding the commit of the example. If you are looking at the main branch, then you need to update your install of Genesis right before running any example script.

closed 2025-05-08 by duburcqa