Genesis-Embodied-AI/genesis-world #1113: [Feature]: Env-wise gravity vector

nootnootpengupengu, 2025-05-06:
### What feature or enhancement are you proposing?

The ability to set a unique gravity vector for each of my simulation environments, not a global vector for all envs. 

### Motivation

Part of my domain randomization scheme requires varying the gravity vector between all envs during intraining. It is currently not possible to do so as it's a global parameter in `SimOptions`. 

### Potential Benefit

Improved policy training and robustness due to increased resilience from different gravity force. 

### What is the expected outcome of the implementation work?

The ability to set a `SimOptions.gravity = gravity_tensor`, where `gravity_tensor.shape = (num_envs, 3)`

bb7332193, 2025-05-20:
This would be very useful, would a [PerlinNoiseField](https://genesis-world.readthedocs.io/en/latest/_modules/genesis/engine/force_fields.html#PerlinNoiseField) work for this purpose?

Milotrince, 2025-08-01:
PR #1324 added this, but there is a bug. I will fix this.

closed 2025-08-04 by duburcqa