Genesis-Embodied-AI/genesis-world #1123: [Bug]: scene.step() gets stuck indefinitely

spen2005, 2025-05-08:
### Bug Description

While simulating a robot hand interacting with objects, the call to scene.step() sometimes gets stuck indefinitely. When this happens, the process becomes unresponsive and cannot be terminated even with Ctrl-C.

This bug can be reproduced using the provided script in the earlier version, but not in the latest version. However, the issue still occurs in the latest version—just under different simulation conditions, especially when the robot hand has complex contact interactions with objects.

### Steps to Reproduce

Please download the action sequence and object files in bug_report2.zip

[bug_report2.zip](https://github.com/user-attachments/files/20103788/bug_report2.zip)

```python
import genesis as gs
import numpy as np
import torch

gs.init(seed=42, backend=gs.gpu)

scene = gs.Scene(
            show_viewer = True,
            sim_options=gs.options.SimOptions(
                dt       = 1 / 60,
                substeps = 4,
            ),
            viewer_options = gs.options.ViewerOptions(
                camera_pos =  (0, 0.3, 1.6),
                camera_lookat = (0, 0.47, 0.7),
            ),
            rigid_options = gs.options.RigidOptions(
                dt = 1 / 60,
                enable_self_collision = False,
                use_contact_island=False,  # Note: for multiple-contact
                constraint_solver=gs.constraint_solver.Newton,
                tolerance=1e-8,
                iterations=1000,
                ls_iterations=500,
                max_collision_pairs=1000,
            ),
            vis_options = gs.options.VisOptions(
                env_separate_rigid = True,
            )
        )

plane = scene.add_entity(
            gs.morphs.Plane(),
        )

object = scene.add_entity(
            morph=gs.morphs.URDF(
                file="./assets/C12001/C12001.urdf",
                pos=(2, 0, 0), # Initial pos set during reset
                fixed=False,
                convexify=True,
                coacd_options=gs.options.CoacdOptions(
                    threshold=0.3,
                ),
            ),
        )

robot = scene.add_entity(
            gs.morphs.URDF(
                file="./assets/g1_shadow/g1_29dof_shadow.urdf",
                pos=(0, 0, 0.75),
                quat = (0.7071, 0, 0, 0.7071),
                fixed=True,
                merge_fixed_links=True,
                convexify=True,
                decompose_robot_error_threshold=float("inf"),
            ),
        )

robot.set_friction(0.5)

table = scene.add_entity(
            gs.morphs.Box(
                pos = (0, 0.9, 0.02),
                size = (1.5, 1.5, 0.04),
                fixed = True,
            )
        )

scene.build(n_envs = 1, env_spacing=(0,0), center_envs_at_origin = False)

robot_kp = np.load("robot_kp.npy")
robot_kv = np.load("robot_kv.npy")
robot.set_dofs_kp(torch.tensor(robot_kp, device=gs.device))
robot.set_dofs_kv(torch.tensor(robot_kv, device=gs.device))

robot_init_action = np.load("init_action.npy")
robot.set_dofs_position(torch.tensor(robot_init_action, device=gs.device))
print("init action", torch.tensor(robot_init_action, device=gs.device))

init_table_pos = np.load("init_table_pos.npy")
table.set_pos(torch.tensor(init_table_pos, device=gs.device))

init_bowl_pos = np.load("init_bowl_pos.npy")
object.set_pos(torch.tensor(init_bowl_pos, device=gs.device))

init_bowl_quat = np.load("init_bowl_quat.npy")
object.set_quat(torch.tensor(init_bowl_quat, device=gs.device))

for i in range(16):
    action = np.load(f"action{i+1}.npy")
    robot.control_dofs_position(torch.tensor(action, device=gs.device))
    scene.step()
```

### Expected Behavior

scene.step() should proceed normally without freezing.

### Relevant log output

```shell
The process appears to hang at the following line:

File "collider_decomp.py", line 269 in detection
```

### Environment

OS: Ubuntu 24.04.1 LTS
GPU: GeForce RTX 4090
CPU: AMD Ryzen 9 7950X 16-Core Processor
GPU-driver: 560.28.03
CUDA / CUDA-toolkit version: 12.4

### Release version or Commit ID

01f0858997767cb82b29bd0fcdcacab278dc40fc

### Additional Context

Thanks for looking into this! Really appreciate your help.

duburcqa, 2025-05-08:
You should enable FPS monitoring, and wait for a few minutes. I suspect it is compiling collisions or something the first time it becomes necessary.

spen2005, 2025-05-08:
I have tried setting show_FPS = True and waited for 20 minutes, but the scene remains stuck at timestep 16.
Please let me know if there's anything else I could try or check.

duburcqa, 2025-05-08:
Ok there is definitely something wrong. I will have a look.

spen2005, 2025-05-16:
Hi, I found that the program gets stuck at mpr_discover_portal() in mpr_decomp.py.

It happens in this loop:

```python
while self.simplex_size[i_ga, i_gb, i_b] < 4:
                        v, v1, v2 = self.compute_support(direction, i_ga, i_gb, i_b)
                        dot = v.dot(direction)
                        if dot < self.CCD_EPS:
                            ret = -1
                            break

                        cont = False

                        va = self.simplex_support[i_ga, i_gb, 1, i_b].v.cross(v)
                        dot = va.dot(self.simplex_support[i_ga, i_gb, 0, i_b].v)
                        if dot < -self.CCD_EPS:
                            self.simplex_support[i_ga, i_gb, 2, i_b].v1 = v1
                            self.simplex_support[i_ga, i_gb, 2, i_b].v2 = v2
                            self.simplex_support[i_ga, i_gb, 2, i_b].v = v
                            cont = True

                        if not cont:
                            va = v.cross(self.simplex_support[i_ga, i_gb, 2, i_b].v)
                            dot = va.dot(self.simplex_support[i_ga, i_gb, 0, i_b].v)
                            if dot < -self.CCD_EPS:
                                self.simplex_support[i_ga, i_gb, 1, i_b].v1 = v1
                                self.simplex_support[i_ga, i_gb, 1, i_b].v2 = v2
                                self.simplex_support[i_ga, i_gb, 1, i_b].v = v
                                cont = True

                        if cont:
                            va = self.simplex_support[i_ga, i_gb, 1, i_b].v - self.simplex_support[i_ga, i_gb, 0, i_b].v
                            vb = self.simplex_support[i_ga, i_gb, 2, i_b].v - self.simplex_support[i_ga, i_gb, 0, i_b].v
                            direction = va.cross(vb)
                            direction = direction.normalized()
                        else:
                            self.simplex_support[i_ga, i_gb, 3, i_b].v1 = v1
                            self.simplex_support[i_ga, i_gb, 3, i_b].v2 = v2
                            self.simplex_support[i_ga, i_gb, 3, i_b].v = v
                            self.simplex_size[i_ga, i_gb, i_b] = 4
```

Hope this helps!

duburcqa, 2025-05-19:
Interesting, I will have a look asap.

duburcqa, 2025-06-27:
I was not able to reproduce the issue on my Mac with the latest main branch.

duburcqa, 2025-06-27:
Actually I could reproduce the issue with the exact commit you shared. I will have a look.

closed 2025-06-30 by duburcqa