Genesis-Embodied-AI/genesis-world #1126: [Bug]: Segmentation mask noise | Blocky Artifacts

GuilhermeAsura, 2025-05-09:
### Bug Description

 While generating synthetic segmentation masks, I'm encountering unexpected rectangular or blocky artifacts outside the target object (in this case, a bottle). These appear as miscolored regions in the segmentation output, despite no corresponding geometry in the 3D scene. I had seen some other similar issues that recommend to disable normal from camera (set normal=false in camera.render()), but I tried and there is no effect.

### Steps to Reproduce

# Number of points in the grid
nx_points, nz_points = 100, 10

# Create a 2D grid with the given specifications 
x = np.linspace(start=-0.5, stop=0.5, num=nx_points)
z = np.linspace(start=0.16, stop=0.6, num=nz_points)
X, Z = np.meshgrid(x, z)
Y = np.full_like(X, 0.5)  # y fixed at 0.5

# Stack into (x, y, z) points
grid_points = np.stack([X.ravel(), Y.ravel(), Z.ravel()], axis=1)

print(f"Grid points: {grid_points.shape}")  # Outputs (1_000, 3)

for i, (x, y, z) in enumerate(grid_points):
    # Randomize camera lookat
    rand_lookat_x = random.uniform(-0.03, 0.03)
    rand_lookat_y = random.uniform(-0.03, 0.03)
    rand_lookat_z = random.uniform(0, 0.09)   

    cam_lookat = [rand_lookat_x, rand_lookat_y, rand_lookat_z]
    cam_pos = [x, y, z]

    # Change camera position
    cam.set_pose(
        pos= cam_pos,
        lookat= cam_lookat
    )
    
    #  View setup 
    view_id = f"view_{i}"  # Ensure consistent naming 
    view_dir = dataset_dir / view_id
    img_view_dir = image_dir / view_id
    view_dir.mkdir(parents=True, exist_ok=True)
    img_view_dir.mkdir(parents=True, exist_ok=True)

    # Get RGB-D data
    for x in range(5):
        scene.step()
        render_output = cam.render(rgb=True, depth=True, segmentation=True, normal=True, colorize_seg=False)

    rgb, depth, seg, normal_map = render_output
    bottle_pixels = np.where(np.array(seg) == 1)
    bottle_pixels = np.transpose(bottle_pixels)  # Shape: (num_pixels, 2)

    # Save data
    np.save(view_dir / "rgb.npy", rgb)
    np.save(view_dir / "depth.npy", depth)
    np.save(view_dir / "segmentation.npy", seg)
    np.save(view_dir / "normal.npy", normal_map)

### Expected Behavior

 The segmentation mask should cleanly isolate the labeled object (the bottle) with a uniform color and without any scattered or unrelated pixels. I also tried to disable normal map like recommended in other similar issues listed on this repo but I have no success

### Screenshots/Videos

[image]
[image]

### Relevant log output

```shell

```

### Environment

- OS: Ubuntu 22.04.5 LTS x86_64
- GPU/CPU: NVIDIA RTX A2000 12GB
- GPU-Driver Version: 550.144.03 
-  CUDA Version: 12.4  

### Release version or Commit ID

Genesis version: 0.2.1

duburcqa, 2025-05-09:
I think this issue has been fixed on main branch. Could you check?

GuilhermeAsura, 2025-05-10:
> I think this issue has been fixed on main branch. Could you check?

I have made a pull and tested right now, apparently the glitch do not interfere in the main mask anymore. This will really help me, thanks!!

[image]

duburcqa, 2025-05-10:
> I have made a pull and tested right now, apparently the glitch do not interfere in the main mask anymore. T

Yes but the issue is not completely fixed apparently :/

duburcqa, 2025-05-10:
Could you share a complete reproduction script, including the scene?

GuilhermeAsura, 2025-05-10:
> [@GuilhermeAsura](https://github.com/GuilhermeAsura) Could you share a complete reproduction script, including the scene?

Off course! Here it is a notebook and .stl that you can use to reproduce it  

[reproduction_script.zip](https://github.com/user-attachments/files/20140660/reproduction_script.zip)

GuilhermeAsura, 2025-05-11:
> > I have made a pull and tested right now, apparently the glitch do not interfere in the main mask anymore.
> 
> Yes but the issue is not completely fixed apparently :/

I continued my experiments after the pull and sadly the glitch come back to interfere in the mask

[image]
[image]

duburcqa, 2025-05-11:
Are you sure that you are still using the main branch ? This is very curious that sometimes it works for the bottle and sometimes it doesn’t …

duburcqa, 2025-05-11:
I ran your script and was not able to reproduce the issue on MacOS Apple Silicon M4 and Ubuntu x84 + Nvidia RTX 5090. Are you sure that you actually installed the main branch ? 

Closing for now since I cannot reproduce :/

closed 2025-05-11 by duburcqa