Genesis-Embodied-AI/genesis-world #1133: [Bug]: Missing examples/render_on_macos.py file

muyuan-ma, 2025-05-13:
### Bug Description

When I use MacOS and the interactive viewer, it prompts me with "[Genesis] [10:38:01] [WARNING] Non-linux system detected. In order to use the interactive viewer, you need to manually run simulation in a separate thread and then start viewer. See `examples/render_on_macos.py`.", but I haven't found this file.

### Steps to Reproduce

If possible, provide a script triggering the bug, e.g.
```python
# code snippet triggering the bug
import argparse

import genesis as gs

def main():
    parser = argparse.ArgumentParser()
    parser.add_argument("-v", "--vis", action="store_true", default=False)
    args = parser.parse_args()

    ########################## init ##########################
    gs.init()

    ########################## create a scene ##########################

    scene = gs.Scene(
        viewer_options=gs.options.ViewerOptions(
            camera_pos=(3.5, 0.0, 2.5),
            camera_lookat=(0.0, 0.0, 0.5),
            camera_fov=40,
        ),
        show_viewer=args.vis,
    )

    ########################## entities ##########################
    plane = scene.add_entity(
        gs.morphs.Plane(),
    )
    franka = scene.add_entity(
        gs.morphs.MJCF(file="xml/franka_emika_panda/panda.xml"),
    )
    ########################## cameras ##########################
    cam_0 = scene.add_camera(
        res=(1280, 960),
        pos=(3.5, 0.0, 2.5),
        lookat=(0, 0, 0.5),
        fov=30,
        GUI=True,
    )
    ########################## build ##########################
    scene.build()

    for i in range(100):
        scene.step()
        print(i)
        cam_0.set_pose(pos=(i / 100, 0, 2.5))
        cam_0.render(
            rgb=True,
            # depth        = True,
            # segmentation = True,
        )

if __name__ == "__main__":
    main()

```
Make sure to attached any needed assets (here `attachement.xml`)!

### Expected Behavior

Provide an example of how to use interactive viewer on MacOS

### Relevant log output

```shell

```

### Environment

- OS: [MacOS 14.6.1 (23G93)]
- GPU/CPU [M1]
- GPU-driver version (N/A)
- CUDA / CUDA-toolkit version (N/A)

### Release version or Commit ID

a0eedddf63aa13cfb0bb64f7b9446c9ca2d511d6

muyuan-ma, 2025-05-13:
After reviewing your recent history updates, I noticed that you've added support for rendering on macOS. However, I'm still encountering an issue. When I run example/rendering/moving_camera.py, I can visualize the scene correctly using `python example/rendering/moving_camera.py`. But when I use the correct flag, specifically `python example/rendering/moving_camera.py -v`, it displays the message '[Genesis] [WARNING] Non-linux system detected. In order to use the interactive viewer, you need to manually run simulation in a separate thread and then start viewer. See `examples/render_on_macos.py.`' and the visualization fails. I'm unsure if this is a bug, but I believe this could be misleading for users.

duburcqa, 2025-05-13:
You should install the latest version (which is not available on PyPI yet) to get full Windows OS support. The instructions are provided in README. Ping me on this issue if it still does not work after that and I will reopen it.