Genesis-Embodied-AI/genesis-world #1135: [Bug]: how to tune the default parameters of links&joints to make an entity(composed of multi links and joints) behave as expected

terry728, 2025-05-13:
### Bug Description

Hi everyone,

I would like to express my sincere gratitude for your exceptional contributions first!

I want to reproduce this [repo](https://github.com/jmz3/Gazebo_cable)(which is achieved in gazebo) to **simulate linear deformable objects in Genesis**. However, after adding the model generated from a .urdf file, I found the dynamics of the same model differ a lot.

Specifically, the model in gazebo behaves quite flexible and quite similar to linear deformable objects, as shown in this [repo](https://github.com/jmz3/Gazebo_cable). While the same model in Genesis behaves more like **a rigid rod**, the attachment file contains how to reproduce it.

After checking the way that how Genesis generate an entity from a .urdf file, I found there are some parameters of links&joints may have a big effect on the flexiblity of the model, including LINK(**friction**) and JOINT(**damping, armature, invweight**...). Some of these parameters have default values, so I wonder how to tune these them to behave the same as in gazebo.

I eagerly await your reply in this matter!

### Steps to Reproduce

[reproduce.zip](https://github.com/user-attachments/files/20186527/reproduce.zip)

### Expected Behavior

instructions about how to generate a more flexible entity in Genesis

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 24.04, Windows 11 24H2]
- GPU/CPU [e.g. A100, RTX 4090, M3pr, Intel I9-9900k, Ryzen 5900x] (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

Genesis Commit ID: a0eedddf63aa13cfb0bb64f7b9446c9ca2d511d6

duburcqa, 2025-05-14:
Could you share 2 videos comparing Gazebo VS Genesis? This would be helpful.

duburcqa, 2025-05-17:
I will work on this, probably on Monday. After testing on my side, I can confirm that the behaviour we observe in Genesis is unexpected.

duburcqa, 2025-05-19:
Could you check if [this PR](https://github.com/Genesis-Embodied-AI/Genesis/pull/1159) is helping? In your case, the URDF file is parsed using Mujoco, then translated to Genesis. This has proven to be reliable in the past for MJCF file, so I'm extending it to URDF files whenever Mujoco supports it.

duburcqa, 2025-05-20:
Do you have a reproduction that I could test? I guess it is related to the removal of `fixed` option?

duburcqa, 2025-05-20:
I'm still actively working on this issue. The refactoring is more tricky than I thought to deal with all the edge-cases properly.

duburcqa, 2025-05-20:
Ok thank you for pointing that out. It is now fixed. There is still a bug somewhere that I need to fix but it is getting closer.

duburcqa, 2025-05-20:
Waiting for the CI to pass. Once this is OK I will have a closer look at your issue, finally!

duburcqa, 2025-05-21:
Can you tell me how it goes with https://github.com/Genesis-Embodied-AI/Genesis/pull/1169 ?

duburcqa, 2025-05-21:
Let me have a look. Running your script.

duburcqa, 2025-05-21:
I found one first issue. Will fix it and see how better it is afterward.

duburcqa, 2025-05-21:
Fixed ! I updated PR#1169. You should decrease the tilmestep to improve numerical stability, and make sure you are starting with the drone and chain above the ground.

duburcqa, 2025-05-22:
I cannot access the video. Do you some testing procedure I could use to iterate on it when trying to fix the issue ?

duburcqa, 2025-05-22:
I would be surprise if there is any “bug” could you just try to set “merge_fixed_links” to false? I changed the default from false to true but it should not affect the physics unless there is a bug, which is possible.

duburcqa, 2025-05-22:
Noted ! I will work on fixed link merging and add a unit tests to make sure it does not happen again.

duburcqa, 2025-05-22:
With fixed link merging disabled, does it still works for your initial stiffness issue ?

duburcqa, 2025-05-22:
> Besides, I come up with some numerical problems during RL training using

Yes, I'm not surprised. Your system has very low inertia, which is hard to simulate. Did you decrease the simulation timestep 'dt' already? If this is a big issue, I suggest adding virtual dof armature for the revolute joints of the chain. By the way, you may consider switching to spherical unbounded joints. This should be supported in Genesis even for URDF files.

duburcqa, 2025-05-22:
Here is a very basic snippet demonstrating how to do this:
```python
    dofs_idx_local = [...]
    robot.set_dofs_armature(robot.get_dofs_armature(dofs_idx_local) + KD * scene.sim._substep_dt, dofs_idx_local)
```
where `dofs_idx_local` are all the indices of the joints of the chain, i.e. `range(5, -1)`.

duburcqa, 2025-05-22:
I fixed it. I will summit a PR soon.

duburcqa, 2025-05-22:
By the way, there was some self-collision that also needed to be fixed on your side. Here is the updated URDF (changed the extension to '.txt' to be able to upload it on Github issue).

[drone_sus.txt](https://github.com/user-attachments/files/20400571/drone_sus.txt)

duburcqa, 2025-05-22:
[Here](https://github.com/Genesis-Embodied-AI/Genesis/pull/1179) is the PR. I recommend that you check the unit test I added. it is based on your reproduction script.

duburcqa, 2025-05-23:
> I still encounter with the abnormal oscillations/jittering

Did you updated the URDF as suggested? Decreased the tilmestep and  added armature as in the unit test?

duburcqa, 2025-05-23:
Both links are pointing to the same abnormal video.

duburcqa, 2025-05-23:
> Files to reproduce are [here](https://drive.google.com/file/d/1-ok1QyU9ynShKshFNS-judUL0AwRMQhK/view?usp=sharing)

This is not a minimal reproductible example, I cannot download a full repository and run it.

At this point, the unit test is passing and everything is fine on my side, so I cannot help you anymore unless you can provide another reproduction script that triggers the issue.

duburcqa, 2025-05-23:
What you could do is store the sequence of actions (propellels_rpm) in a pickle, then provide a script showing the discrepancy before and after my work on your issue.

duburcqa, 2025-05-23:
```bash
pip install -e ".[dev]"
```

```python
pytest -svn 0 --vis "tests/test_rigid_physics.py::test_drone_advanced"
```

closed 2025-05-30 by duburcqa