Genesis-Embodied-AI/genesis-world #1139: [Bug]: Changing friction vector doesn't affect the contact behavior

VahidDanesh, 2025-05-13:
### Bug Description

First of all, thanks to the Genesis team for their great contribution!

I’m trying to enable pure rotational slippage (about normal vector of contact) at the Panda robot’s fingertip contacts so that grasped objects can freely spin under the pads without resistance.

According to MuJoCo’s MJCF reference, the `friction` field accepts up to five coefficients. I set the second element (torsional friction) to a near-zero value, but saw no change when grasping and lifting an object.

I also changed `condim` to `1` for all fingertip pads, yet the object still resists rotation.

–––

Any suggestions for achieving true rotational freedom around the contact-normal—whether by tweaking contact friction parameters or by adding hinge joints to the robot’s fingers (how?) —would be greatly appreciated. 

### Steps to Reproduce

If possible, provide a script triggering the bug, e.g.
```python
# code snippet triggering the bug
import genesis as gs
gs.init()
franka = scene.add_entity(
    gs.morphs.MJCF(file="xml/franka_emika_panda/panda.xml")
                   
)
cube = scene.add_entity(
    gs.morphs.Box(
        size = (0.1, 0.07, 0.03),
        pos  = (0.7, 0, 0.015),
    ))
 
scene.build()
motors_dof = np.arange(7)
fingers_dof = np.arange(7, 9)
end_effector = franka.get_link('hand')
above_box = cube.get_pos().cpu().numpy() + np.array([-cube_size[0]/2 + 0.02, 0, 0.11 + 0.1])
# move to pre-grasp pose
qpos = franka.inverse_kinematics(
    link = end_effector,
    pos  = above_box,
    quat = np.array([0, 1, 0, 0]),
)
# gripper open pos
qpos[-2:] = 0.04
path = franka.plan_path(
    qpos_goal     = qpos,
    num_waypoints = 200, # 2s duration
)
# execute the planned path
for waypoint in path:
    franka.control_dofs_position(waypoint)
    scene.step()
side_box = above_box + np.array([0.0, 0.0, -0.1])
qpos_reach = franka.inverse_kinematics(
    link = end_effector,
    pos  = side_box,
    quat = np.array([0, 1, 0, 0]),
)
path = franka.plan_path(
    qpos_goal     = qpos_reach,
    num_waypoints = 100, # 1s duration
)
for waypoint in path:
    franka.control_dofs_position(waypoint)
    scene.step()
# grasp
franka.control_dofs_position(qpos_reach[:-2], motors_dof)
franka.control_dofs_force(np.array([-5, -5]), fingers_dof)
# lift
# move to pre-grasp pose
qpos = franka.inverse_kinematics(
    link = end_effector,
    pos  = above_box,
    quat = np.array([0, 1, 0, 0]),
)
franka.control_dofs_position(qpos[:-2], motors_dof)
for _ in range(100):
    scene.step()
```

Here panda.xml has the following changes:

```
        <default class="fingertip_pad_collision_1">
          <geom type="box" size="0.0085 0.004 0.0085" pos="0 0.0055 0.0445" friction="0.0 0.0 0.0"/>
        </default>
        <default class="fingertip_pad_collision_2">
          <geom type="box" size="0.003 0.002 0.003" pos="0.0055 0.002 0.05" friction="0.0 0.0 0.0"/>
        </default>
        <default class="fingertip_pad_collision_3">
          <geom type="box" size="0.003 0.002 0.003" pos="-0.0055 0.002 0.05" friction="0.0 0.0 0.0"/>
        </default>
        <default class="fingertip_pad_collision_4">
          <geom type="box" size="0.003 0.002 0.0035" pos="0.0055 0.002 0.0395" friction="0.0 0.0 0.0"/>
        </default>
        <default class="fingertip_pad_collision_5">
          <geom type="box" size="0.003 0.002 0.0035" pos="-0.0055 0.002 0.0395" friction="0.0 0.0 0.0"/>
        </default>
```

### Expected Behavior

With `condim=3` and torsional friction coefficients set to zero, the solver should allow nearly frictionless rotation about the contact normal. However, it appears that when two surfaces come into contact, the solver defaults to the object’s friction settings (or only keeps a one dimension float number for modelling friction) and ignores those of the fingertip pads.  

I expect the following screenshot happens right after lifting object.

### Screenshots/Videos

[image]

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 22.04]
- GPU/CPU [RTX 3070Ti Laptop, i7 13700k] (N/A if no GPU/CPU)
- GPU-driver version 550
- CUDA / CUDA-toolkit version 12.4

### Release version or Commit ID

current version

### Additional Context

Is that related to the following part in `genesis/utils/mjcf.py`?
```python
    info = {
        "type": gs_type,
        "pos": mj_geom.pos * scale,
        "quat": mj_geom.quat,
        "contype": mj_geom.contype[0],
        "conaffinity": mj_geom.conaffinity[0],
        "group": mj_geom.group[0],
        "data": geom_data,
        "friction": mj_geom.friction[0], # only keeps first entity of friction vector?
        "sol_params": np.concatenate((mj_geom.solref, mj_geom.solimp)),
    }
```

duburcqa, 2025-05-14:
The implementation for friction in genesis is not as advanced that in Mujoco. There is only 1 scalar parameter in Genesis, and condim cannot be changed (only condim = 3 is implemented). There is no torsional friction at the moment.

duburcqa, 2025-05-14:
You could try disabling multi-contact:
```python
    scene = gs.Scene(
       [...],
        rigid_options=gs.options.RigidOptions(
           [...],
            enable_multi_contact=False,
           [...],
        ),
       [...],
    )
```

duburcqa, 2025-05-14:
I think the "torsional" friction you see here is due to multi-contact. Here, the interaction between the gripper and the object is kinda surfacing. This results in a torsional friction effect even if in practice there is no torsional force that is applied on each individual contact point. That is basically the same than what happens in reality.

closed 2025-05-14 by duburcqa