Genesis-Embodied-AI/genesis-world #1142: [Feature]: Robo-arm RL example

NicolasKuske, 2025-05-14:
### What feature or enhancement are you proposing?

Hi Genesis team,

Great work!

Mayb you could add to the tutorial a minimal RL example of how to teach the Franka robo-arm to reach for the cube?

Possibly with the addition of including a camera sensor and, finally attaching the sensor to the arm?

Thank you!

Best, Nicolas

### Motivation

I am currently working on a multimodal setup with Genesis for a robotic research project on continual learning and would like my code to be as clean and efficient as possible.

### Potential Benefit

Fits with the prior flow of the tutorial and would help beginners (including me) to get started with a simple robotic RL example.

### What is the expected outcome of the implementation work?

- [ ]

AcademeMaster, 2025-05-23:
Hi，I have finish a demo，but it has need to improve。Can you improve it further?
https://github.com/AcademeMaster/Genesis-robot-gym

NicolasKuske, 2025-05-24:
Hi @AcademeMaster, thank you for your efforts!

However, your example is far from being minimal :) 

I am afraid, currently it has a complexity which does not help my understanding much. 

For example, there is no need to include a random position generator. Just one cube position which the arm learns to reach towards would suffice for a minimal functional example.

Thanks again, Nicolas