Genesis-Embodied-AI/genesis-world #1155: [Bug]: scene.build() on Windows takes significantly longer than before

MahiBoroji, 2025-05-17:
### Bug Description

Compiling kernels, whether in `scene.build()` or in `scene.step()` with `scene.build(compile_kernels=False)` takes nearly 14 minutes, whereas the same scene previously built in around 7 minutes in earlier versions, last week. 

### Steps to Reproduce

```python
import torch

import genesis as gs

########################## init ##########################
gs.init(backend=gs.gpu)

########################## create a scene ##########################
scene = gs.Scene(
    show_viewer=False,
    viewer_options=gs.options.ViewerOptions(
        camera_pos=(3.5, -1.0, 2.5),
        camera_lookat=(0.0, 0.0, 0.5),
        camera_fov=40,
        res=(1920, 1080),
    ),
    rigid_options=gs.options.RigidOptions(
        dt=0.01,
    ),
)

########################## entities ##########################
plane = scene.add_entity(
    gs.morphs.Plane(),
)

franka = scene.add_entity(
    gs.morphs.MJCF(file="xml/franka_emika_panda/panda.xml"),
)

########################## build ##########################

B = 4
scene.build(n_envs=B, env_spacing=(1.0, 1.0))

# control all the robots
# with the following control: 43M FPS
# without the following control (arm in collision with the floor): 32M FPS
franka.control_dofs_position(
    torch.tile(torch.tensor([0, 0, 0, -1.0, 0, 0, 0, 0.02, 0.02], device=gs.device), (B, 1)),
)

for i in range(100):
    scene.step()
```

### Expected Behavior

I was expecting compiling kernels in shorter time rather than 14 minutes. Compiling kernel on similar Ubuntu 22.04 machine takes 4 minutes (previously 2 minutes). I've tested both gpu and cpu init. With different `n_envs`. 

### Relevant log output

```shell
[Genesis] [21:07:20] [INFO] Building scene <28cce0b>...
[Genesis] [21:07:25] [WARNING] Reference robot position exceeds joint limits.
[Genesis] [21:07:49] [INFO] Compiling simulation kernels...
[Genesis] [21:23:09] [INFO] Building visualizer...
[Genesis] [21:23:13] [INFO] Viewer created. Resolution: 1280×960, max_FPS: 60.
```

### Environment

- OS: [Windows 11]
- GPU/CPU [RTX 3500,  I7-13800H] 
- GPU-driver version (573.06)
- CUDA / CUDA-toolkit version (12.8)

### Release version or Commit ID

[]

duburcqa, 2025-05-17:
I’m assuming you are using the main branch. So yes, compilation time is a big issue. There is a current effort on trying to reduce it. Lately is has significantly increased to fix a compilation issue that was only occurring on Apple Metal. Apparently the flag to disable compilation is not effective enough, and should be improved or removed. There is no much you can do on you side I’m afraid, this should be fixed in Genesis itself.

stillonearth, 2025-05-19:
The same is on linux. Build times increassing for same configuration from 3-5 to 20 minutes.

duburcqa, 2025-05-19:
Besides, which robot model are you loading? This may be helpful to find the most appropriate fix.

stillonearth, 2025-05-20:
@duburcqa it's crazyflie example.