Genesis-Embodied-AI/genesis-world #1156: [Bug]: 'too many values to unpack' error due to breaking changes of libigl from 2.6.0

dieu-detruit, 2025-05-17:
### Bug Description

`scene.add_entity` causes an error below.

```
File "/xxx/.venv/lib/python3.12/site-packages/genesis/engine/entities/rigid_entity/rigid_geom.py", line 224, in _compute_sd
    sd, _, = igl.signed_distance(query_points, self._sdf_verts, self._sdf_faces)
    ^^^^^^
ValueError: too many values to unpack (expected 2)
```

It's caused by breaking changes of libigl from v2.6.0 (released 2025-05-15)
* https://pypi.org/project/libigl/2.6.0/

The version of libigl is not fixed in pyproject.toml:
https://github.com/Genesis-Embodied-AI/Genesis/blob/37c1ce62c70a0a4917684a6db5ae5b64174e660d/pyproject.toml

Installing libigl==2.5.1 manually fixed this issue.

### Steps to Reproduce

If possible, provide a script triggering the bug, e.g.

```shell
uv init
uv add genesis-world torch
uv run main.py # script below
```

```python
# code snippet triggering the bug
import genesis as gs
gs.init()
scene = gs.Scene(show_viewer=False)
broken = scene.add_entity(
    gs.morphs.MJCF(file='attachement.xml'),
)
scene.build()
```
Make sure to attached any needed assets (here `attachement.xml`)!

### Expected Behavior

I think it's better to fix the libigl version or follow the new signature.

### Relevant log output

```shell
File "/xxx/.venv/lib/python3.12/site-packages/genesis/engine/entities/rigid_entity/rigid_geom.py", line 224, in _compute_sd
    sd, _, = igl.signed_distance(query_points, self._sdf_verts, self._sdf_faces)
    ^^^^^^
ValueError: too many values to unpack (expected 2)
```

### Environment

- OS: [e.g. Ubuntu 24.04, Windows 11 24H2]
- GPU/CPU [e.g. A100, RTX 4090, M3pr, Intel I9-9900k, Ryzen 5900x] (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

genesis-world v0.2.1

duburcqa, 2025-05-18:
This issue has already been fixed on main branch.

duburcqa, 2025-05-18:
Hum... apparently I was wrong. Let me double check.

duburcqa, 2025-05-18:
Yes, It has been fixed already (I will improve the current patch later as it is quite ugly and inefficient).

closed 2025-05-18 by duburcqa

teresanguyen01, 2025-05-19:
I am also receiving this issue running the ```hello_genesis.py```file.: 

```Traceback (most recent call last):
  File "/Users/xxx/Genesis/examples/tutorials/hello_genesis.py", line 7, in <module>
    plane = scene.add_entity(
            ^^^^^^^^^^^^^^^^^
  File "/Users/xxx/miniconda3/lib/python3.12/site-packages/genesis/utils/misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/xxx/miniconda3/lib/python3.12/site-packages/genesis/engine/scene.py", line 359, in add_entity
    entity = self._sim._add_entity(morph, material, surface, visualize_contact)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/xxx/miniconda3/lib/python3.12/site-packages/genesis/engine/simulator.py", line 134, in _add_entity
    entity = self.rigid_solver.add_entity(self.n_entities, material, morph, surface, visualize_contact)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/xxx/miniconda3/lib/python3.12/site-packages/genesis/engine/solvers/rigid/rigid_solver_decomp.py", line 64, in add_entity
    entity = entity_class(
             ^^^^^^^^^^^^^
  File "/Users/xxx/miniconda3/lib/python3.12/site-packages/genesis/engine/entities/rigid_entity/rigid_entity.py", line 68, in __init__
    self._load_model()
  File "/Users/xxx/miniconda3/lib/python3.12/site-packages/genesis/engine/entities/rigid_entity/rigid_entity.py", line 87, in _load_model
    self._load_primitive(self._morph, self._surface)
  File "/Users/xxx/miniconda3/lib/python3.12/site-packages/genesis/engine/entities/rigid_entity/rigid_entity.py", line 189, in _load_primitive
    link._add_geom(
  File "/Users/xxx/miniconda3/lib/python3.12/site-packages/genesis/engine/entities/rigid_entity/rigid_link.py", line 173, in _add_geom
    geom = RigidGeom(
           ^^^^^^^^^^
  File "/Users/xxx/miniconda3/lib/python3.12/site-packages/genesis/engine/entities/rigid_entity/rigid_geom.py", line 94, in __init__
    self._preprocess()
  File "/Users/xxx/miniconda3/lib/python3.12/site-packages/genesis/engine/entities/rigid_entity/rigid_geom.py", line 146, in _preprocess
    sdf_val = self._compute_sd(query_points)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/xxx/miniconda3/lib/python3.12/site-packages/genesis/engine/entities/rigid_entity/rigid_geom.py", line 224, in _compute_sd
    sd, _, _ = igl.signed_distance(query_points, self._sdf_verts, self._sdf_faces)
    ^^^^^^^^
```

Macbook Pro M3 Sequoia 15.4.1

duburcqa, 2025-05-19:
Please install the main branch instead.

teresanguyen01, 2025-05-19:
@duburcqa I have installed the main branch with all its new edits and I am still receiving the same issue.

duburcqa, 2025-05-19:
There must be something wrong, because the error is clearly pointing to a line of code that does not exist:
```python
  File "/Users/xxx/miniconda3/lib/python3.12/site-packages/genesis/engine/entities/rigid_entity/rigid_geom.py", line 224, in _compute_sd
    sd, _, _ = igl.signed_distance(query_points, self._sdf_verts, self._sdf_faces)
```

You can check yourself [here](https://github.com/Genesis-Embodied-AI/Genesis/blob/37c1ce62c70a0a4917684a6db5ae5b64174e660d/genesis/engine/entities/rigid_entity/rigid_geom.py#L224).

OpenJarvisAI, 2025-07-03:
Also hitting this issue.

Please help developers solve their first hello world , your Genesis sounds good but not even runable for a hello world

OpenJarvisAI, 2025-07-03:
Hi, I would suggest please upgrade your pypi version, it's been 3 months. Otherwise new comers could be very suffering.

OpenJarvisAI, 2025-07-17:
I would suggest add requirements to genesis requirement.txt