Genesis-Embodied-AI/genesis-world #1158: [Bug]: Inverse Kinematics (IK) Multiple Solutions Control in Genesis AI

coghks625, 2025-05-19:
### Bug Description

Hello,

I have a question about the inverse kinematics (IK) solver in Genesis AI. As far as I know, the current implementation of the IK solver finds a solution that falls within a specified error margin and then outputs the corresponding forward kinematics (FK) values.

However, since IK generally has multiple solutions, is there an option in Genesis AI to control the solution by specifying the values of joints such as the shoulder, elbow, or wrist? This would allow me to achieve the desired configuration for the robot arm.

Thank you for your time.

### Steps to Reproduce

        qpos = kuka_rb.inverse_kinematics(
            link=kuka_end_effector, 
            pos=target_pos_tensor, 
            quat=target_quat_tensor, 
            max_solver_iters=1000, 
            rot_tol=0.0000001,
            pos_tol=0.0000001
        )

### Expected Behavior

If the IK solver is provided with a target position and orientation, it should return an array of possible solutions (multiple joint configurations) instead of just one.

The user should be able to specify a preferred joint angle range for certain joints (like shoulder, elbow, or wrist) to guide the solver towards the desired solution.

Alternatively, the user should be able to set a priority or weighting for each joint to influence the final solution selection.

### Relevant log output

```shell

```

### Environment

- OS: [Ubuntu 20.04]
- GPU/CPU [RTX 3090]

### Release version or Commit ID

0.2.1

duburcqa, 2025-05-19:
There is a way to specify the initial position. You should use that to encourage the solver find a solution close to it.

coghks625, 2025-05-19:
How can I ensure that, in certain special cases, the first joint of a 6-axis robotic arm is strictly fixed at 180 degrees while allowing it to move freely in normal cases?

If I set the initial position like [3.14, 0, 0, 0, 0, 0], will this guarantee that the first joint always stays at 180 degrees? According to your answer, it seems the IK solver will find the closest solution to the target position, which means the first joint value may change if the closest solution does not maintain 180 degrees.

Is there a way to strictly lock the first joint at 180 degrees in such cases?

duburcqa, 2025-05-21:
Yes, just specify both `init_qpos` and `dofs_idx_local`. The later is a mask that specifies which dof can be modified to reach the desired target, i.e. `range(1, entity.n_dofs)` in your case.