Genesis-Embodied-AI/genesis-world #1167: [Bug]: New version of Libigl conflicts

qtcc-uoft, 2025-05-20:
### Bug Description

Hi all, 

When you install Genesis through pip, it installs the newest version of libigl (2.6.1) when this conflicts with the version that Genesis uses (2.5.x). This ends up causing an issue pretty quickly - notably when you try to add an entity you'll get an error. It's easy to fix by just downgrading your libigl to 2.5.1, but this should be done by Genesis by default. Thanks!

### Steps to Reproduce

just try installing genesis right now 

### Expected Behavior

for genesis to work and not throw errors when you add entities

### Relevant log output

```shell

```

### Environment

tried on both Ubuntu native and Windows through WSL, happens in both

### Release version or Commit ID

latest branch

duburcqa, 2025-05-20:
> latest branch

I don't think this is correct. This issue has already been fixed on main branch.

closed 2025-05-20 by duburcqa

qtcc-uoft, 2025-05-20:
Don't think so - I just checked the project.tomyl file in main and it still has just "libigl". Regardless this happens when you install via the recommended way with pip so unless I'm doing something very wrong with my install this is still an issue

duburcqa, 2025-05-20:
Both version are supported by the main branch.