Genesis-Embodied-AI/genesis-world #1172: [Bug]:  Error when using force fields with MPM entities

RiccardoMengozzi, 2025-05-21:
### Bug Description

I tried to visualize the behaviour of a MPM entity (cube) when a force field is applied to it. I initially tried with a Wind force field, but then discovered that the same problem happens with other force fields. The problem is the following:
```
taichi.lang.exception.TaichiSyntaxError: 
File "/home/mengo/miniconda3/envs/gs/lib/python3.10/site-packages/genesis/engine/coupler.py", line 202, in mpm_grid_op:
                    vel_mpm += self.mpm_solver._ffs[i_ff].get_acc(pos, vel_mpm,
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Parameter `i` missing.
```

### Steps to Reproduce

Just run the following script. Note that if the force field is commented, everything works fine.

```python
import genesis as gs

def main():
    ########################## init ##########################
    gs.init(backend=gs.gpu)

    ########################## create a scene ##########################

    scene = gs.Scene(
        sim_options=gs.options.SimOptions(
            dt=5e-3,
            substeps=15,
        ),
        viewer_options=gs.options.ViewerOptions(
            camera_pos=(3, -1, 1.5),
            camera_lookat=(0.0, 0.0, 0.0),
            camera_fov=30,
            max_FPS=60,
        ),
        show_viewer=True,
        vis_options=gs.options.VisOptions(
            visualize_mpm_boundary=True,
        ),
        mpm_options=gs.options.MPMOptions(
            lower_bound=(-1.0,-1.0, -0.05),
            upper_bound=(1.0, 1.0, 1.0),
            grid_density=128,
        ),
    )

    ########################## entities ##########################
    plane = scene.add_entity(
        gs.morphs.Plane(),
    )
    cube = scene.add_entity(
        material=gs.materials.MPM.Elastic(),
        morph=gs.morphs.Box(
            pos=(0.0, 0.0, 0.5),
            euler=(0, 0, 0),
            size=(0.1, 0.1, 0.1),
        ),
        surface=gs.surfaces.Default(
            color=(0.8, 0.2, 0.2),
            vis_mode="particle",    
        ),
    )

    wind_field = scene.add_force_field(
        gs.force_fields.Wind(
            direction=(1.0, 0.0, 0.0),
            strength=1000.0,
            radius=1.0,
            center=(0.0, 0.0, 1.0),
        ),
    )
    wind_field.activate()

    scene.build()
    

    while True:
        scene.step()

if __name__ == "__main__":
    main()
```

### Expected Behavior

Following error:

```
[Genesis] [14:14:45] [INFO] ╭───────────────────────────────────────────────────────────────────────╮
[Genesis] [14:14:45] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈│
[Genesis] [14:14:45] [INFO] ╰───────────────────────────────────────────────────────────────────────╯
[Genesis] [14:14:45] [INFO] Running on [NVIDIA GeForce GTX 1650] with backend gs.cuda. Device memory: 3.63 GB.
[Genesis] [14:14:45] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [14:14:45] [INFO] Scene <f8d80d5> created.
[Genesis] [14:14:45] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <472460b>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [14:14:45] [INFO] Adding <gs.MPMEntity>. idx: 1, uid: <4eeb670>, morph: <gs.morphs.Box>, material: <gs.materials.MPM.Elastic>.
[Genesis] [14:14:45] [INFO] Sampled 8,182 particles.
[Genesis] [14:14:45] [INFO] Adding <gs.RigidEntity>. idx: 2, uid: <7f01b5d>, morph: <gs.morphs.Mesh(file='/home/mengo/Research/LLM_DOM/dlo_ws/models/SimpleTable/SimpleTable.obj')>, material: <gs.materials.Rigid>.
[Genesis] [14:14:46] [INFO] Building scene <f8d80d5>...
[Genesis] [14:14:50] [WARNING] Current `substep_dt` (0.000333333) is greater than suggested_dt (0.00015625, calculated based on `grid_density`). Simulation might be unstable.
[Genesis] [14:14:50] [INFO] Compiling simulation kernels...
Traceback (most recent call last):
  File "/home/mengo/Research/LLM_DOM/dlo_ws/prova.py", line 88, in <module>
    main()
  File "/home/mengo/Research/LLM_DOM/dlo_ws/prova.py", line 80, in main
    scene.build()
  File "/home/mengo/miniconda3/envs/gs/lib/python3.10/site-packages/genesis/utils/misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "/home/mengo/miniconda3/envs/gs/lib/python3.10/site-packages/genesis/engine/scene.py", line 602, in build
    self._sim.step()
  File "/home/mengo/miniconda3/envs/gs/lib/python3.10/site-packages/genesis/engine/simulator.py", line 240, in step
    self.substep(self.cur_substep_local)
  File "/home/mengo/miniconda3/envs/gs/lib/python3.10/site-packages/genesis/engine/simulator.py", line 272, in substep
    self._coupler.couple(f)
  File "/home/mengo/miniconda3/envs/gs/lib/python3.10/site-packages/genesis/engine/coupler.py", line 553, in couple
    self.mpm_grid_op(f, self.sim.cur_t)
  File "/home/mengo/.local/lib/python3.10/site-packages/taichi/lang/kernel_impl.py", line 1178, in __call__
    raise type(e)("\n" + str(e)) from None
taichi.lang.exception.TaichiSyntaxError: 
File "/home/mengo/miniconda3/envs/gs/lib/python3.10/site-packages/genesis/engine/coupler.py", line 202, in mpm_grid_op:
                    vel_mpm += self.mpm_solver._ffs[i_ff].get_acc(pos, vel_mpm,
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Parameter `i` missing.
[Genesis] [14:14:58] [INFO] 💤 Exiting Genesis and caching compiled kernels...

```

### Relevant log output

```shell

```

### Environment

- OS: **Ubuntu 22.04.5**
- GPU/CPU **GeForce GTX 1650 Mobile, Intel i7-9750H**
- GPU-driver version: **570.133.20**
- CUDA / CUDA-toolkit version: **V12.4.131**

### Release version or Commit ID

genesis 0.2.1

duburcqa, 2025-05-22:
Could you check whether this fix is doing what you want? https://github.com/Genesis-Embodied-AI/Genesis/pull/1176

duburcqa, 2025-05-22:
Great!

RiccardoMengozzi, 2025-05-22:
It is also behaving as I would expect

closed 2025-05-22 by Genesis-Embodied-AI