Genesis-Embodied-AI/genesis-world #1182: [Bug]: Mesh with many vertices disappears when using FEM material

yar2001, 2025-05-23:
### Bug Description

Mesh with many vertices (as shown in the obj file in attachment below) disappears in viewer when using FEM material 

### Steps to Reproduce

```python
import os

import genesis as gs

os.environ["PYOPENGL_PLATFORM"] = "glx"

########################## init ##########################
gs.init(
    backend=gs.cpu,
    debug=True,
)

########################## create a scene ##########################
dt = 4e-3
scene = gs.Scene(
    sim_options=gs.options.SimOptions(
        dt=dt,
        substeps=10,
    ),
    fem_options=gs.options.FEMOptions(
        dt=dt,
        damping=45.0,
    ),
    viewer_options=gs.options.ViewerOptions(
        camera_fov=30,
        res=(1280, 720),
        max_FPS=60,
        run_in_thread=False,
    ),
    show_viewer=True,
)

########################## entities ##########################
plane = scene.add_entity(
    morph=gs.morphs.Plane(),
)

cloth_2 = scene.add_entity(
    material=gs.materials.FEM.Elastic(),
    morph=gs.morphs.Mesh(
        file="./Cube_Simple_SubD.obj",
        scale=0.3,
        pos=(0, 0, 2.0),
        euler=(0.0, 0, 0.0),
    ),
    surface=gs.surfaces.Default(
        color=(0.8, 0.4, 0.2, 1.0),
        vis_mode="visual",
    ),
)

########################## build ##########################
scene.build()

horizon = 1000
for i in range(horizon):
    scene.step()

```

Model: [Cube_Simple_SubD.obj](https://github.com/user-attachments/files/20403722/Cube_Simple_SubD.zip)

### Expected Behavior

The cube should appear in the viewer.

### Screenshots/Videos

[image]

### Relevant log output

```shell
python test.py 
[Genesis] [11:58:45] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [11:58:45] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [11:58:45] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [11:58:46] [INFO] Beware running Genesis in debug mode dramatically reduces runtime speed.
[Genesis] [11:58:46] [INFO] Running on [AMD Ryzen 7 6800U with Radeon Graphics] with backend gs.cpu. Device memory: 14.94 GB.
[Genesis] [11:58:46] [DEBUG] [Taichi] version 1.7.3, llvm 15.0.4, commit 5ec301be, linux, python 3.11.11
[Genesis] [11:58:46] [DEBUG] [Taichi] Starting on arch=x64
[Genesis] [11:58:46] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: True, 🎨 theme: 'dark'.
[Genesis] [11:58:46] [INFO] Scene <1383465> created.
[Genesis] [11:58:46] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <871454e>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [11:58:46] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 0.
[Genesis] [11:58:46] [INFO] Adding <gs.FEMEntity>. idx: 1, uid: <6ca64df>, morph: <gs.morphs.Mesh(file='/home/yar2001/Downloads/Cube_Simple_SubD.obj')>, material: <gs.FEM.Elastic>.
[Genesis] [11:58:46] [DEBUG] Tetrahedra file (`.tet`) found in cache.
[Genesis] [11:58:46] [INFO] Building scene <1383465>...
[Genesis] [11:58:46] [DEBUG] Constraint solver time constant not specified. Using minimum value (`0.0008`).
[W 05/23/25 11:58:46.864 322670] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[Genesis] [11:58:46] [DEBUG] Constraint solver time constant not specified. Using minimum value (`0.0008`).
[W 05/23/25 11:58:46.953 322670] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 05/23/25 11:58:47.389 322670] [frontend_ir.cpp:begin_frontend_struct_for_on_external_tensor@1694] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 05/23/25 11:58:48.735 322670] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[Genesis] [11:58:48] [INFO] Entity <6ca64df> added. class: FEMEntity, morph: Mesh, size: (1924, 533), material: <gs.FEM.Elastic>.
[Genesis] [11:58:48] [INFO] Compiling simulation kernels...
[Genesis] [11:58:58] [INFO] Building visualizer...
[Genesis] [11:58:58] [DEBUG] Trying to create OpenGL Context for PYOPENGL_PLATFORM='glx'...
[Genesis] [11:58:59] [INFO] Viewer created. Resolution: 1280×720, max_FPS: 60.
[Genesis] [11:58:59] [INFO] Running at 254.82 FPS.
[Genesis] [11:58:59] [INFO] Running at 156.87 FPS.
[Genesis] [11:58:59] [INFO] Running at 111.28 FPS.
[Genesis] [11:58:59] [INFO] Running at 89.79 FPS.
[Genesis] [11:58:59] [INFO] Running at 77.39 FPS.
[Genesis] [11:58:59] [INFO] Running at 68.29 FPS.
[Genesis] [11:58:59] [INFO] Running at 61.74 FPS.
[Genesis] [11:58:59] [INFO] Running at 56.28 FPS.
[Genesis] [11:58:59] [INFO] Running at 51.79 FPS.
[Genesis] [11:58:59] [INFO] Running at 48.25 FPS.
[Genesis] [11:58:59] [INFO] Running at 45.37 FPS.
[Genesis] [11:58:59] [INFO] Running at 41.95 FPS.
[Genesis] [11:58:59] [INFO] Running at 39.78 FPS.
[Genesis] [11:58:59] [INFO] Running at 37.92 FPS.
[Genesis] [11:59:00] [INFO] Running at 36.43 FPS.
[Genesis] [11:59:00] [INFO] Running at 35.01 FPS.
[Genesis] [11:59:00] [INFO] Running at 33.89 FPS.
[Genesis] [11:59:00] [INFO] Running at 32.84 FPS.
```

### Environment

- OS: Linux 6.12.21-amd64 Debian 6.12.21-1 (2025-03-30) x86_64 GNU/Linux
- CPU: AMD Ryzen 7 6800U

### Release version or Commit ID

49ce8c746e7ce1255cdd97b904434b70f8fd4e40

Tryoe, 2025-05-23:
me too,have u fixed it？

yar2001, 2025-05-23:
> me too,have u fixed it？

Not yet :(

duburcqa, 2025-05-23:
@ Tryoe You should provide all the assets, otherwise it is impossible to help you.

duburcqa, 2025-05-23:
@yar2001 I can confirm that this simple script is already failing. The mesh gets extremely distorted and "disappears".

```python
import genesis as gs

gs.init(backend=gs.cpu, precision="64", logging_level="debug")

scene = gs.Scene(
    sim_options=gs.options.SimOptions(
        dt=0.001,
    ),
    show_viewer=True,
)

scene.add_entity(
    material=gs.materials.FEM.Elastic(),
    morph=gs.morphs.Mesh(
        file="./Cube_Simple_SubD.obj",
        scale=0.1,
        pos=(0, 0, 1.0),
    ),
)
scene.build()

for i in range(10):
    scene.step()
```

duburcqa, 2025-05-23:
Here is the asset `"./Cube_Simple_SubD.obj"` with `.txt` extension instead of `.obj`:
[Cube_Simple_SubD.txt](https://github.com/user-attachments/files/20413042/Cube_Simple_SubD.txt)

Tryoe, 2025-05-26:
> [@Tryoe](https://github.com/Tryoe) You should provide all the assets, otherwise it is impossible to help you.

[temp.zip](https://github.com/user-attachments/files/20437077/temp.zip)
This is all of my asset files. I tried loading the first 20 for testing, but found that all the objects disappeared. I hope to get your help. Thank you.

Libero0809, 2025-05-27:
The disappearance is caused by the failure of the fem solver. It is currently using an explicit Euler integration and is prone to divergence. A temporary fix would be using a smaller dt or more substeps.  I tried using dt=1e-3 /substeps=10 and dt=4e-3\substeps=40  and the cube worked fine. We will be improving our solver soon to be more robust.

closed 2025-06-07 by Genesis-Embodied-AI