Genesis-Embodied-AI/genesis-world #1183: [Bug]: why I can't run the code?

Tryoe, 2025-05-23:
### Bug Description

import genesis as gs
import trimesh

def main():
    # 初始化Genesis
    gs.init(backend=gs.gpu)  # 使用GPU后端
    
    # 创建场景
    scene = gs.Scene(
        viewer_options=gs.options.ViewerOptions(
            camera_pos=(2.0, 2.0, 2.0),  # 相机位置
            camera_lookat=(0.0, 0.0, 0.0),  # 看向原点
            camera_fov=40,  # 视场角
        ),
        show_viewer=True,  # 显示查看器
        rigid_options=gs.options.RigidOptions(
            enable_collision=True,  # 启用碰撞
            gravity=(0, 0, -9.81),  # 添加重力
        ),
    )
    
    # 加载三棱柱体模型
    prism_file = "examples/obj/result_prisms.obj"
    print(f"正在加载模型: {prism_file}")
    
    # 使用trimesh加载模型以计算体积
    mesh = trimesh.load(prism_file)
    print(mesh.is_watertight)  # True 表示封闭
    volume = mesh.volume
    mass = volume * 10.0  # 假设密度为10
    print(f"模型体积: {volume:.4f}, 质量: {mass:.4f}")
    
    # 添加模型到场景
    scene.add_entity(
        morph=gs.morphs.Mesh(
            file=prism_file,
            pos=(0, 0, 0),  # 位置
            fixed=True,     # 固定位置
        ),
        surface=gs.surfaces.Default(
            color=(0.8, 0.8, 0.8),  # 浅灰色
            smooth=True,  # 平滑渲染
        ),
    )
    
    # 添加一个平面作为地面
    scene.add_entity(
        morph=gs.morphs.Box(
            pos=(0.0, 0.0, -0.5),  # 位置
            size=(10.0, 10.0, 0.1),  # 大小
            fixed=True,  # 固定地面
        ),
        surface=gs.surfaces.Default(
            color=(0.8, 0.8, 0.8),  # 浅灰色
            smooth=True,
        ),
    )
    
    # 添加坐标轴
    # X轴（红色）
    scene.add_entity(
        gs.morphs.Cylinder(
            pos=(0.5, 0.0, 0.0),
            radius=0.005,
            height=1.0,
            euler=(0, 90, 0),  # 绕Y轴旋转90度
            fixed=True,
        ),
        surface=gs.surfaces.Default(color=(1.0, 0.0, 0.0)),  # 红色
    )
    
    # Y轴（绿色）
    scene.add_entity(
        gs.morphs.Cylinder(
            pos=(0.0, 0.5, 0.0),
            radius=0.005,
            height=1.0,
            euler=(90, 0, 0),  # 绕X轴旋转90度
            fixed=True,
        ),
        surface=gs.surfaces.Default(color=(0.0, 1.0, 0.0)),  # 绿色
    )
    
    # Z轴（蓝色）
    scene.add_entity(
        gs.morphs.Cylinder(
            pos=(0.0, 0.0, 0.5),
            radius=0.005,
            height=1.0,
            fixed=True,
        ),
        surface=gs.surfaces.Default(color=(0.0, 0.0, 1.0)),  # 蓝色
    )
    
    # 构建场景
    scene.build()
    
    print("场景已构建完成！")
    print("操作说明：")
    print("- 按住鼠标左键并拖动可以旋转视图")
    print("- 滚动鼠标滚轮可以缩放视图")
    print("- 按住鼠标右键并拖动可以平移视图")
    
    # 运行场景
    while True:
        scene.step()

if __name__ == "__main__":
    main() 

### Steps to Reproduce

If possible, provide a script triggering the bug, e.g.
```python
# code snippet triggering the bug
import genesis as gs
gs.init()
scene = gs.Scene(show_viewer=False)
broken = scene.add_entity(
    gs.morphs.MJCF(file='attachement.xml'),
)
scene.build()
```
Make sure to attached any needed assets (here `attachement.xml`)!

### Expected Behavior

Hope u can help me 

### Relevant log output

```shell

```

### Environment

Genesis 0.2.1 
windows 

### Release version or Commit ID

/

### Additional Context

PS D:\khalil\Genesis\Genesis> & D:/conda/envs/Genesis/python.exe d:/khalil/Genesis/Genesis/examples/obj/render_prisms.py
[Genesis] [13:21:27] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [13:21:27] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [13:21:27] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [13:21:27] [INFO] Running on [NVIDIA GeForce RTX 3090] with backend gs.cuda. Device memory: 24.00 GB.
[Genesis] [13:21:28] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [13:21:28] [INFO] Scene <6877c44> created.
正在加载模型: examples/obj/result_prisms.obj
False
模型体积: 62.8938, 质量: 628.9382
[Genesis] [13:21:28] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <724eb5a>, morph: <gs.morphs.Mesh(file='D:\khalil\Genesis\Genesis\examples\obj\result_prisms.obj')>, material: <gs.materials.Rigid>.
[Genesis] [13:21:28] [INFO] Preprocessing geom idx 0.
Traceback (most recent call last):
  File "d:\khalil\Genesis\Genesis\examples\obj\render_prisms.py", line 109, in <module>    
    main() 
    ^^^^^^
  File "d:\khalil\Genesis\Genesis\examples\obj\render_prisms.py", line 34, in main
    scene.add_entity(
  File "D:\conda\envs\Genesis\Lib\site-packages\genesis\utils\misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\conda\envs\Genesis\Lib\site-packages\genesis\engine\scene.py", line 359, in add_entity
    entity = self._sim._add_entity(morph, material, surface, visualize_contact)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\conda\envs\Genesis\Lib\site-packages\genesis\engine\simulator.py", line 134, in _add_entity
    entity = self.rigid_solver.add_entity(self.n_entities, material, morph, surface, visualize_contact)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\conda\envs\Genesis\Lib\site-packages\genesis\engine\solvers\rigid\rigid_solver_decomp.py", line 64, in add_entity
    entity = entity_class(
             ^^^^^^^^^^^^^
  File "D:\conda\envs\Genesis\Lib\site-packages\genesis\engine\entities\rigid_entity\rigid_entity.py", line 68, in __init__
    self._load_model()
  File "D:\conda\envs\Genesis\Lib\site-packages\genesis\engine\entities\rigid_entity\rigid_entity.py", line 75, in _load_model
    self._load_mesh(self._morph, self._surface)
  File "D:\conda\envs\Genesis\Lib\site-packages\genesis\engine\entities\rigid_entity\rigid_entity.py", line 258, in _load_mesh
    link._add_geom(
  File "D:\conda\envs\Genesis\Lib\site-packages\genesis\engine\entities\rigid_entity\rigid_link.py", line 173, in _add_geom
    geom = RigidGeom(
           ^^^^^^^^^^
  File "D:\conda\envs\Genesis\Lib\site-packages\genesis\engine\entities\rigid_entity\rigid_geom.py", line 94, in __init__
    self._preprocess()
  File "D:\conda\envs\Genesis\Lib\site-packages\genesis\engine\entities\rigid_entity\rigid_geom.py", line 146, in _preprocess
    sdf_val = self._compute_sd(query_points)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\conda\envs\Genesis\Lib\site-packages\genesis\engine\entities\rigid_entity\rigid_geom.py", line 224, in _compute_sd
    sd, _, _ = igl.signed_distance(query_points, self._sdf_verts, self._sdf_faces)
    ^^^^^^^^
ValueError: too many values to unpack (expected 3)
[Genesis] [13:21:28] [INFO] 💤 Exiting Genesis and caching compiled kernels...

duburcqa, 2025-05-23:
Please install the main branch following the README instructions.

closed 2025-05-23 by duburcqa

Tryoe, 2025-05-23:
I've created a new workspace and synchronized the latest code, but when running the example file, it still throws an error.
 Windows system
 Python 3.11
Genesis 0.2.1

duburcqa, 2025-05-23:
There must be something wrong. I'm 100% sure this is fixed. And you line the error is pointing out does not even exist on main branch.

Tryoe, 2025-05-23:
I found that the issue was caused by using the Genesis package from the virtual environment. If I’ve downloaded the source code, do I still need the Genesis library?

duburcqa, 2025-05-23:
> If I’ve downloaded the source code, do I still need the Genesis library?

What do you mean? The source code? It depends if Genesis was installed in editable mode.

Could you provide more details about what you did BEFORE (failure) and AFTER (success) ? I would add this information to the troubleshooting section of the official documentation.

Tryoe, 2025-05-23:
I searched through blogs and found that many people solved the issue by copying the files from the genesis directory in the source code into the Genesis folder in the virtual environment to replace the files.But when I replaced it with the latest version of the genesis folder, the following error occurred: cannot import name ‘_replay’ from partially initialized module ‘genesis.ext.fast_simplification’ (most likely due to a circular import). The blogs mentioned that an older version of the Genesis files is needed, so I am now downloading the older version and will try again.

  

  
Great!

duburcqa, 2025-05-23:
This is clearly the wrong way of doing it. There must be something wrong on your setup. A simple pip install (using `git@`) without even downloading the source mainly should work out of the box. Did you try cleaning your pip cache of something ?

Tryoe, 2025-05-23:
I didn't, but I reinstalled the Genesis package via pip. It worked fine when I followed the above steps on my previous 4060 computer, and everything ran well. However, now I’m using a 3090 computer for rendering, and after reconfiguring the Genesis development environment, it failed.

duburcqa, 2025-05-23:
You used this command?
`pip install git+https://github.com/Genesis-Embodied-AI/Genesis.git`

Tryoe, 2025-05-23:
I used git clone https://github.com/Genesis-Embodied-AI/Genesis.git