Genesis-Embodied-AI/genesis-world #1184: [Bug]: Tetrahedral meshing of obj file

bahadir14, 2025-05-23:
### Bug Description

Hi, I am struggling with tetrahedral meshing of obj file geometry. Is there any better way to have volumetric mesh of certain geometry using genesis?

### Steps to Reproduce

If possible, provide a script triggering the bug, e.g.
```python
# code snippet triggering the bug
import genesis as gs
gs.init()
scene = gs.Scene(show_viewer=False)
broken = scene.add_entity(
    gs.morphs.MJCF(file='attachement.xml'),
)
scene.build()
```
Make sure to attached any needed assets (here `attachement.xml`)!

### Expected Behavior

Well meshed obj file geometry from genesis

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 24.04]
- GPU/CPU [e.g. A100, RTX 4090, M3pr, Intel I9-9900k, Ryzen 5900x] (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

0.2.1

duburcqa, 2025-05-23:
You should provide the assets, i.e. `"/home/bahadir/Desktop/genesis/genesis/part_1/tensorflow_model/container_new.obj"` otherwise it is impossible to help you. That being said, try installing the main branch following the README instructions. I think this issue has been fixed since latest release on pypi.

bahadir14, 2025-05-27:
Hi @duburcqa,

I installed latest version of genesis and it still does not show meshed geometry.

Here is the geometry file:

https://drive.google.com/file/d/13IcODtOAwrfXYd9grEpuSmd3v3FmmOk2/view?usp=sharing

Is there any way to solve problem?

LeoHsuProgrammingLab, 2025-08-07:
Hi @bahadir14, have you solved the problem?

bahadir14, 2025-08-08:
Hi @LeoHsuProgrammingLab, unfortunately no.
Hi @duburcqa, Could you help me about this bug?

duburcqa, 2025-08-11:
Basically the FEM solver is failing, which "breaks" the mesh. Nothing to do with rendering. See related issue https://github.com/Genesis-Embodied-AI/Genesis/issues/1182

duburcqa, 2025-08-11:
This script is running and the mesh does not disappear. The result is definitely bad, but that is another issue:
```python
import argparse

import os

import genesis as gs

gs.init(seed=0, precision="32", logging_level="debug")

scene = gs.Scene(
    sim_options=gs.options.SimOptions(
        dt=0.01,
    ),
    fem_options=gs.options.FEMOptions(
        use_implicit_solver=True,
        damping=0.1,
    ),
    viewer_options=gs.options.ViewerOptions(
        camera_pos=(9.6, 4.8, 14.0),
        camera_lookat=(0.0, 0.0, 0.0),
        camera_fov=35,
    ),
    show_viewer=True,
    show_FPS=False,
)
plane = scene.add_entity(
    morph=gs.morphs.URDF(file="urdf/plane/plane.urdf", fixed=True),
    material=gs.materials.Rigid(),
)
obj1 = scene.add_entity(
    morph=gs.morphs.Mesh(
        file="/Users/alexis.duburcq/Downloads/container_new.obj",
        scale=0.05,
        euler=(-225, -45, 90),
        pos=(0.0, 5.0, 5.0),
    ),
    material=gs.materials.FEM.Elastic(
        E=1.0e7,
        nu=0.3,
        rho=100.0,
        model="linear",
    ),
    surface=gs.surfaces.Iron(
        vis_mode="visual",
    ),
)

scene.build()
for i in range(1000):
    scene.step()
```

closed 2025-08-11 by duburcqa

duburcqa, 2025-08-12:
> I still get no meshed geometry and use_implicit_solver command is not working for me. Do you have any suggestions?

I guess you need to update Genesis. You are using an outdated version probably.