Genesis-Embodied-AI/genesis-world #1231: Simulation hangs indefinitely at "Sampling particles with pbs-32 sampler and generating `.ptc` file"

Maxwell-Zhao, 2025-06-04:
### Bug Description

When I try to simulate a obj file with MPM, the simulation workflow freezes without error.

[image]

### Steps to Reproduce

```python
my_model = scene.add_entity(
    material=gs.materials.MPM.Elastic(),
    morph=gs.morphs.Mesh( 
        file="woman.obj",  
        pos=(0.0, 0.5, 0.0),           
        euler=(0.0, 0.0, 45.0),         
        scale=1.5,                     
        collision=True,            
        visualization=True      
    ),
    surface=gs.surfaces.Default(
        color=(0.2, 0.8, 0.3),        
        vis_mode='visual',            
    ),
)
```
Make sure to attached any needed assets (here `attachement.xml`)!

### Expected Behavior

The simulation workflow freezes without error.

[image]

### Screenshots/Videos

[image]

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 20.04,]
- I5 14400f+5060ti 16G
- GPU-driver version 575
- CUDA / CUDA-toolkit version 12.6

### Release version or Commit ID

2.1.0

duburcqa, 2025-06-04:
Can you provide a reproduction script with all the assets? Otherwise it is impossible to help you unfortunately.

Maxwell-Zhao, 2025-06-05:
## issue_reproduce_package/
## reproduce_bug.py
```python
# code snippet triggering the bug
import genesis as gs

gs.init()

scene = gs.Scene(
    sim_options=gs.options.SimOptions(
        dt       = 0.0003125,
        substeps = 10,
    ),
    # mpm_options=gs.options.MPMOptions(
    #     lower_bound   = (-0.5, -1.0, 0.0),
    #     upper_bound   = (0.5, 1.0, 1),
    # ),
    vis_options=gs.options.VisOptions(
        visualize_mpm_boundary = True,
    ),
    viewer_options=gs.options.ViewerOptions(
        camera_fov=30,
    ),
    show_viewer = True,
)
plane = scene.add_entity(
    morph=gs.morphs.Plane(),
)

my_model = scene.add_entity(
    material=gs.materials.MPM.Elastic(),
    morph=gs.morphs.Mesh(  
        file="woman.obj", 
        pos=(0.0, 0.5, 0.0),          
        euler=(0.0, 0.0, 45.0),      
        scale=1.5,                   
        collision=True,            
        visualization=True           
    ),
    surface=gs.surfaces.Default(
        color=(0.2, 0.8, 0.3),       
        vis_mode='visual',           
    ),
)
scene.build()
```

## mesh.obj:
The image of this obj file is as followes. It is just a fat woman.

## environment:
```python
Package                  Version
------------------------ ------------------------
absl-py                  2.3.0
annotated-types          0.7.0
black                    25.1.0
certifi                  2025.4.26
charset-normalizer       3.4.2
click                    8.2.1
coacd                    1.0.7
colorama                 0.4.6
contourpy                1.3.2
cycler                   0.12.1
dataclasses-json         0.6.7
decorator                5.2.1
Deprecated               1.2.18
dill                     0.4.0
etils                    1.12.2
filelock                 3.16.1
fonttools                4.58.1
freetype-py              2.5.1
fsspec                   2025.5.1
genesis-world            0.2.1
glfw                     2.9.0
idna                     3.10
imageio                  2.37.0
imageio-ffmpeg           0.6.0
importlib_resources      6.5.2
Jinja2                   3.1.4
kiwisolver               1.4.8
lazy_loader              0.4
libigl                   2.6.1
llvmlite                 0.44.0
lxml                     5.4.0
markdown-it-py           3.0.0
MarkupSafe               2.1.5
marshmallow              3.26.1
matplotlib               3.10.3
mdurl                    0.1.2
moviepy                  2.2.1
mpmath                   1.3.0
mujoco                   3.2.5
mypy_extensions          1.1.0
networkx                 3.4.2
numba                    0.61.2
numpy                    1.26.4
nvidia-cublas-cu12       12.8.4.1
nvidia-cuda-cupti-cu12   12.8.90
nvidia-cuda-nvrtc-cu12   12.8.93
nvidia-cuda-runtime-cu12 12.8.90
nvidia-cudnn-cu12        9.8.0.87
nvidia-cufft-cu12        11.3.3.83
nvidia-cufile-cu12       1.13.1.3
nvidia-curand-cu12       10.3.9.90
nvidia-cusolver-cu12     11.7.3.90
nvidia-cusparse-cu12     12.5.8.93
nvidia-cusparselt-cu12   0.6.3
nvidia-nccl-cu12         2.26.5
nvidia-nvjitlink-cu12    12.8.93
nvidia-nvtx-cu12         12.8.90
ompl                     1.7.0
opencv-python            4.11.0.86
OpenEXR                  3.3.3
packaging                25.0
pathspec                 0.12.1
pillow                   11.2.1
pip                      25.1
platformdirs             4.3.8
plotly                   5.24.1
pooch                    1.8.2
proglog                  0.1.12
psutil                   7.0.0
py-cpuinfo               9.0.0
pycollada                0.9
pydantic                 2.7.1
pydantic_core            2.18.2
PyGEL3D                  0.6.1
pyglet                   2.1.6
pygltflib                1.16.0
Pygments                 2.19.1
pymeshlab                2023.12.post3
PyOpenGL                 3.1.9
pyparsing                3.2.3
python-dateutil          2.9.0.post0
python-dotenv            1.1.0
pytorch-triton           3.3.1+gitc8757738
pyvista                  0.45.2
requests                 2.32.3
rich                     14.0.0
rtree                    1.4.0
scikit-image             0.25.2
scipy                    1.15.3
scooby                   0.10.1
screeninfo               0.8.1
setuptools               78.1.1
six                      1.17.0
sympy                    1.13.3
taichi                   1.7.3
tenacity                 9.1.2
tetgen                   0.6.4
tifffile                 2025.5.10
tomli                    2.2.1
torch                    2.8.0.dev20250602+cu128
torchaudio               2.8.0.dev20250602+cu128
torchvision              0.23.0.dev20250602+cu128
tqdm                     4.67.1
trimesh                  4.6.10
typing_extensions        4.13.2
typing-inspect           0.9.0
urllib3                  2.4.0
vtk                      9.4.2
wheel                    0.45.1
wrapt                    1.17.2
z3-solver                4.15.0.0
zipp                     3.22.0
```

Maxwell-Zhao, 2025-06-06:
> Ok, so I guess the viewer is falling back on CPU. There must be some something wrong with your setup. Can you share the whole log, after passing `logging_level="debug"` to `gs.init` ?

This is the whole log:

```python
[Genesis] [10:49:27] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [10:49:27] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [10:49:27] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [10:49:27] [INFO] Running on [NVIDIA Graphics Device] with backend gs.cuda. Device memory: 15.44 GB.
[Genesis] [10:49:27] [DEBUG] [Taichi] version 1.7.3, llvm 15.0.4, commit 5ec301be, linux, python 3.10.16
[Genesis] [10:49:27] [DEBUG] [Taichi] Starting on arch=cuda
[Genesis] [10:49:27] [INFO]  Genesis initialized.  version: 0.2.1,  seed: None,  precision: '32',  debug: False,  theme: 'dark'.
[Genesis] [10:49:28] [INFO] Scene <05b6834> created.
[Genesis] [10:49:28] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <5ed4365>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 0.
[Genesis] [10:49:28] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <fe147c3>, morph: <gs.morphs.Box>, material: <gs.materials.Rigid>.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 1.
[Genesis] [10:49:28] [INFO] Adding <gs.RigidEntity>. idx: 2, uid: <951da09>, morph: <gs.morphs.MJCF(file='/home/anaconda3/envs/genesis/lib/python3.10/site-packages/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 2.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 3.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 4.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 5.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 6.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 7.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 8.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 9.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 10.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 11.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 12.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 13.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 14.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 15.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 16.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 17.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 18.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 19.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 20.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 21.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 22.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 23.
[Genesis] [10:49:28] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 24.
[Genesis] [10:49:28] [INFO] Building scene <05b6834>...
[Genesis] [10:49:31] [INFO] Compiling simulation kernels...
[Genesis] [10:49:35] [INFO] Building visualizer...
[Genesis] [10:49:36] [INFO] Viewer created. Resolution: 1440×1080, max_FPS: 60.
[Genesis] [10:49:38] [WARNING] `qpos_start` exceeds joint limit. Relaxing joint limit to contain `qpos_start` for planning.
[Genesis] [10:49:38] [INFO] Path solution found successfully.
[Genesis] [10:49:39] [INFO] Running at 8.19 FPS.
[Genesis] [10:49:39] [INFO] Running at 8.25 FPS.
[Genesis] [10:49:39] [INFO] Running at 8.60 FPS.
[Genesis] [10:49:39] [INFO] Running at 8.99 FPS.
[Genesis] [10:49:39] [INFO] Running at 9.39 FPS.
[Genesis] [10:49:39] [INFO] Running at 9.80 FPS.
[Genesis] [10:49:39] [INFO] Running at 10.23 FPS.
[Genesis] [10:49:39] [INFO] Running at 10.25 FPS.
[Genesis] [10:49:39] [INFO] Running at 10.69 FPS.
```
This bug (the code runs in cpu) is one of my question, the most important bug is I can not simulate an object with MPM. The code gets stuck at "Sampling particles with pbs-32 sampler and generating .ptc file".

duburcqa, 2025-06-06:
> This bug (the code runs in cpu) is one of my question, the most important bug is I can not simulate an object with MPM. The code gets stuck at "Sampling particles with pbs-32 sampler and generating .ptc file".

Yes I get that, but always better to have the full picture.

Maxwell-Zhao, 2025-06-06:
> Yes I get that, but always better to have the full picture.

The log and code are mentioned above:

[image]

duburcqa, 2025-06-22:
@Maxwell-Zhao I wanted to start debugging this issue but I cannot download your asset anymore