Genesis-Embodied-AI/genesis-world #1282: [Bug]: Unreliable value output from get_links_net_contact_force() function

scottie-6d, 2025-06-15:
### Bug Description

When the gripper lifts and holds the cube at a fixed position, the reported net contact force in Z no longer balances the cube’s weight—even though the cube remains perfectly static. This appears when shear forces occures. When the cube simply rests on the ground, its net contact force closely matches its weight.

### Steps to Reproduce

```python
import numpy as np
import time
import genesis as gs

########################## init ##########################
gs.init(backend=gs.gpu, precision="32")
########################## create a scene ##########################
scene = gs.Scene(
    viewer_options=gs.options.ViewerOptions(
        camera_pos=(3, -1, 1.5),
        camera_lookat=(0.0, 0.0, 0.5),
        camera_fov=30,
        res=(960, 640),
        max_FPS=60,
    ),
    sim_options=gs.options.SimOptions(
        dt=0.01,
    ),
    rigid_options=gs.options.RigidOptions(
        box_box_detection=True,
    ),
    show_viewer=True,
    show_FPS=False
)

########################## entities ##########################
plane = scene.add_entity(
    gs.morphs.Plane(),
)
franka = scene.add_entity(
    gs.morphs.MJCF(file="xml/franka_emika_panda/panda.xml"),
)

cube = scene.add_entity(
    gs.morphs.Box(
        size=(0.04, 0.04, 0.04),
        pos=(0.65, 0.0, 0.02),
    )
    ,visualize_contact=True
)
########################## build ##########################
scene.build()

motors_dof = np.arange(7)
fingers_dof = np.arange(7, 9)
qpos = np.array([-1.0124, 1.5559, 1.3662, -1.6878, -1.5799, 1.7757, 1.4602, 0.04, 0.04])
franka.set_qpos(qpos)
scene.step()

end_effector = franka.get_link("hand")
qpos = franka.inverse_kinematics(
    link=end_effector,
    pos=np.array([0.65, 0.0, 0.135]),
    quat=np.array([0, 1, 0, 0]),
)

franka.control_dofs_position(qpos[:-2], motors_dof)

# hold
for i in range(100):
    print("hold", i)
    print("Cube weight: ", -9.81*cube.get_mass(), "  Cube Net Force: ",cube.get_links_net_contact_force()[0,2])
    scene.step()

# grasp
finder_pos = -0.0
for i in range(100):
    franka.control_dofs_position(qpos[:-2], motors_dof)
    franka.control_dofs_position(np.array([finder_pos, finder_pos]), fingers_dof)
    scene.step()

# lift
qpos = franka.inverse_kinematics(
    link=end_effector,
    pos=np.array([0.65, 0.0, 0.3]),
    quat=np.array([0, 1, 0, 0]),
)
for i in range(200):
    print("lift", i)
    franka.control_dofs_position(qpos[:-2], motors_dof)
    franka.control_dofs_position(np.array([finder_pos, finder_pos]), fingers_dof)
    print("Cube weight: ", -9.81*cube.get_mass(), "  Cube Net Force: ",cube.get_links_net_contact_force()[0,2])
    scene.step()
```

### Expected Behavior

* The net contact force in Z should approximately equal `-9.81 × mass` of the cube after lifting it with gripper.

### Relevant log output

```shell

```

### Environment

OS: Ubuntu 22.04
GPU: RTX4080

### Release version or Commit ID

4e40e61af064fa293a3c06da8bf41d0e5194a939

duburcqa, 2025-06-15:
It looks like `get_links_net_contact_force` is broken indeed. I will have a look at it.

duburcqa, 2025-06-15:
I found the bug, I will fix it ASAP.

closed 2025-06-15 by duburcqa