Genesis-Embodied-AI/genesis-world #1345: [Bug]: Building visualizer - Blocked here

marino-multipla, 2025-07-02:
### Bug Description

Hello,
i try to run a simple code taken from here:
https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/hello_genesis.html

When i start the code i get blocked forever at the "[Genesis] [11:08:55] [INFO] Building visualizer...".

How to fix this?

### Steps to Reproduce

```python
import genesis as gs
gs.init(backend=gs.cpu)

scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())
franka = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)

scene.build()

for i in range(1000):
    scene.step()
```

### Expected Behavior

Once the scene is built, an interactive viewer will pop up to visualize the scene. 

### Relevant log output

```shell
[Genesis] [11:08:49] [INFO] ╭───────────────────────────────────────────────────────────────────────────────╮
[Genesis] [11:08:49] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈│
[Genesis] [11:08:49] [INFO] ╰───────────────────────────────────────────────────────────────────────────────╯
[Genesis] [11:08:50] [INFO] Consider setting 'performance_mode=True' in production to maximise runtime speed, if significantly increasing compilation time is not a concern.
[Genesis] [11:08:50] [INFO] Running on [Intel(R) Core(TM) i9-14900HX] with backend gs.cpu. Device memory: 29.07 GB.
[Genesis] [11:08:50] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [11:08:51] [INFO] Scene <6545ece> created.
[Genesis] [11:08:51] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <8439745>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [11:08:51] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <89871c4>, morph: <gs.morphs.MJCF(file='/home/gmaccagni/Workspace/genesis/Genesis/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [11:08:51] [WARNING] (MJCF) Approximating tendon by joint actuator for `finger_joint1`
[Genesis] [11:08:51] [WARNING] (MJCF) Actuator control gain and bias parameters cannot be reduced to a unique PD control position gain. Using max between gain and bias for joint `finger_joint1`.
[Genesis] [11:08:51] [WARNING] (MJCF) Approximating tendon by joint actuator for `finger_joint2`
[Genesis] [11:08:51] [WARNING] (MJCF) Actuator control gain and bias parameters cannot be reduced to a unique PD control position gain. Using max between gain and bias for joint `finger_joint2`.
[Genesis] [11:08:51] [INFO] Applying offset to base link's pose with user provided value in morph.
[Genesis] [11:08:51] [INFO] Building scene <6545ece>...
[Genesis] [11:08:52] [WARNING] Reference robot position exceeds joint limits.
[Genesis] [11:08:52] [WARNING] Constraint solver time constant should be greater than 2*substep_dt. timeconst is changed from `0.005` to `0.02`). Decrease simulation timestep or increase timeconst to avoid altering the original value.
[Genesis] [11:08:53] [INFO] Compiling simulation kernels...
[Genesis] [11:08:55] [INFO] Building visualizer...
```

### Environment

Ubuntu Ubuntu 20.04.6 LTS
Python 3.12.11
PyTorch 2.7.1+cu128 
Installed with pip
NVIDIA-SMI 570.133.20
NVIDIA GeForce RTX 4090
Driver Version: 570.133.20
CUDA Version: 12.8
Genesis installed in editable mode locally.
conda version : 25.1.1

### Release version or Commit ID

Installed genesis-world 0.2.1 locally

MRiabov, 2025-07-03:
I'm not a collaborator on Genesis, however, what logs do you see when you Ctrl+C your program?
If you can see what happens in the stack of KeyboardInterrupt, you'll pinpoint your issue.

duburcqa, 2025-07-03:
Could you try again with debug logging level ? `gs.init(backend=gs.cpu, logging_level="debug")`

marino-multipla, 2025-07-03:
Hey @duburcqa ,

here the log with DEBUG level:

[Genesis] [16:48:07] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [16:48:07] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [16:48:07] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [16:48:08] [INFO] Consider setting 'performance_mode=True' in production to maximise runtime speed, if significantly increasing compilation time is not a concern.
[Genesis] [16:48:08] [INFO] Running on [Intel(R) Core(TM) i9-14900HX] with backend gs.cpu. Device memory: 29.07 GB.
[Genesis] [16:48:08] [DEBUG] [Taichi] version 1.7.3, llvm 15.0.4, commit 5ec301be, linux, python 3.10.18
[Genesis] [16:48:08] [DEBUG] [Taichi] Starting on arch=x64
[Genesis] [16:48:08] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [16:48:08] [INFO] Scene <bbb535f> created.
[Genesis] [16:48:08] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <cd54f4f>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [16:48:08] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 0.
[Genesis] [16:48:08] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <970169c>, morph: <gs.morphs.MJCF(file='/home/gmaccagni/Workspace/genesis/Genesis/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [16:48:09] [WARNING] (MJCF) Approximating tendon by joint actuator for `finger_joint1`
[Genesis] [16:48:09] [WARNING] (MJCF) Actuator control gain and bias parameters cannot be reduced to a unique PD control position gain. Using max between gain and bias for joint `finger_joint1`.
[Genesis] [16:48:09] [WARNING] (MJCF) Approximating tendon by joint actuator for `finger_joint2`
[Genesis] [16:48:09] [WARNING] (MJCF) Actuator control gain and bias parameters cannot be reduced to a unique PD control position gain. Using max between gain and bias for joint `finger_joint2`.
[Genesis] [16:48:09] [INFO] Applying offset to base link's pose with user provided value in morph.
[Genesis] [16:48:09] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 1.
[Genesis] [16:48:09] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 2.
[Genesis] [16:48:09] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 3.
[Genesis] [16:48:09] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 4.
[Genesis] [16:48:09] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 5.
[Genesis] [16:48:09] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 6.
[Genesis] [16:48:09] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 7.
[Genesis] [16:48:09] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 8.
[Genesis] [16:48:09] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 9.
[Genesis] [16:48:09] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 10.
[Genesis] [16:48:09] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 11.
[Genesis] [16:48:09] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 12.
[Genesis] [16:48:09] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 13.
[Genesis] [16:48:09] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 14.
[Genesis] [16:48:09] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 15.
[Genesis] [16:48:09] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 16.
[Genesis] [16:48:09] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 17.
[Genesis] [16:48:09] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 18.
[Genesis] [16:48:09] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 19.
[Genesis] [16:48:09] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 20.
[Genesis] [16:48:09] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 21.
[Genesis] [16:48:09] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 22.
[Genesis] [16:48:09] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 23.
[Genesis] [16:48:09] [INFO] Building scene <bbb535f>...
[Genesis] [16:48:09] [DEBUG] Constraint solver time constant not specified. Using minimum value (`0.02`).
[Genesis] [16:48:09] [DEBUG] Constraint solver time constant not specified. Using minimum value (`0.02`).
[Genesis] [16:48:09] [WARNING] Reference robot position exceeds joint limits.
[Genesis] [16:48:09] [WARNING] Constraint solver time constant should be greater than 2*substep_dt. timeconst is changed from `0.005` to `0.02`). Decrease simulation timestep or increase timeconst to avoid altering the original value.
[Genesis] [16:48:10] [INFO] Compiling simulation kernels...
[Genesis] [16:48:13] [INFO] Building visualizer...
[Genesis] [16:48:13] [DEBUG] Trying to create OpenGL Context for PYOPENGL_PLATFORM='native'...

marino-multipla, 2025-07-03:
Hey,
no rendering was happening after this point. After some investigation, I found that the issue was related to OpenGL. Specifically, it seems that the OpenGL context was not properly set for rendering. It was set to **OpenGL vendor string: Intel**

To resolve the issue, I followed these steps:

> Install necessary dependencies for OpenGL:
```
sudo apt-get install freeglut3 freeglut3-dev
pip install PyOpenGL PyOpenGL_accelerate
```

> verify the current OpenGL setup
```
sudo apt install mesa-utils
glxinfo | grep "OpenGL vendor"
```

> Switch to the NVIDIA graphics driver
```
sudo prime-select nvidia
sudo reboot
```

> Verify the vendor switch equal to **OpenGL vendor string: NVIDIA Corporation**
`glxinfo | grep "OpenGL vendor"`

pschmutz, 2025-07-03:
I have the same issue on my machine, but the suggested fixes don't work. Do you have any suggestions how to debug this?

### System Information
- OS: Linux Ubuntu 24.04.1 LTS (x86_64)
- Kernel: 6.11.0-29-generic
- Python: 3.12.3
- GPU: NVIDIA GeForce RTX 4060 (8GB VRAM)
- NVIDIA Driver: 570.133.07
- CUDA: 12.8

pschmutz, 2025-07-04:
Here is the output (please note that I used uv to install the package, which I forgot to mention earlier) I can try to install it with pip as well, but it should not make a difference):
```
$ uv run hello_genesis.py
Built genesis-world @ file:///home/philipp/repos/genesis_workspace/Genesis
Uninstalled 1 package in 0.87ms
Installed 1 package in 1ms
[Genesis] [19:58:57] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [19:58:57] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [19:58:57] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [19:58:58] [INFO] Running on [NVIDIA GeForce RTX 4060 Laptop GPU] with backend gs.cuda. Device memory: 7.63 GB.
[Genesis] [19:58:58] [DEBUG] [Taichi] version 1.7.3, llvm 15.0.4, commit 5ec301be, linux, python 3.12.3
[Genesis] [19:58:58] [DEBUG] [Taichi] Starting on arch=cuda
[Genesis] [19:58:58] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [19:58:58] [INFO] Scene <d3c1dab> created.
[Genesis] [19:58:58] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <a8201cf>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [19:58:58] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 0.
[Genesis] [19:58:58] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <ea472b3>, morph: <gs.morphs.MJCF(file='/home/philipp/repos/genesis_workspace/Genesis/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [19:58:59] [WARNING] (MJCF) Tendon not supported
[Genesis] [19:58:59] [DEBUG] (MJCF) No actuator found for joint `world`
[Genesis] [19:58:59] [DEBUG] (MJCF) No actuator found for joint `link0`
[Genesis] [19:58:59] [DEBUG] (MJCF) No actuator found for joint `hand`
[Genesis] [19:58:59] [WARNING] (MJCF) Approximating tendon by joint actuator for `finger_joint1`
[Genesis] [19:58:59] [WARNING] (MJCF) Actuator control gain and bias parameters cannot be reduced to a unique PD control position gain. Using max between gain and bias.
[Genesis] [19:58:59] [WARNING] (MJCF) Approximating tendon by joint actuator for `finger_joint2`
[Genesis] [19:58:59] [WARNING] (MJCF) Actuator control gain and bias parameters cannot be reduced to a unique PD control position gain. Using max between gain and bias.
[Genesis] [19:58:59] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 1.
[Genesis] [19:58:59] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 2.
[Genesis] [19:58:59] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 3.
[Genesis] [19:58:59] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 4.
[Genesis] [19:58:59] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 5.
[Genesis] [19:58:59] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 6.
[Genesis] [19:58:59] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 7.
[Genesis] [19:58:59] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 8.
[Genesis] [19:58:59] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 9.
[Genesis] [19:58:59] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 10.
[Genesis] [19:59:00] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 11.
[Genesis] [19:59:00] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 12.
[Genesis] [19:59:00] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 13.
[Genesis] [19:59:00] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 14.
[Genesis] [19:59:00] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 15.
[Genesis] [19:59:00] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 16.
[Genesis] [19:59:00] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 17.
[Genesis] [19:59:00] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 18.
[Genesis] [19:59:00] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 19.
[Genesis] [19:59:00] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 20.
[Genesis] [19:59:00] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 21.
[Genesis] [19:59:00] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 22.
[Genesis] [19:59:00] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 23.
[Genesis] [19:59:00] [INFO] Building scene <d3c1dab>...
[Genesis] [19:59:00] [DEBUG] Constraint solver time constant not specified. Using minimum value (`0.02`).
[Genesis] [19:59:01] [DEBUG] Constraint solver time constant not specified. Using minimum value (`0.02`).
[Genesis] [19:59:01] [WARNING] Reference robot position exceeds joint limits.
[Genesis] [19:59:01] [WARNING] Constraint solver time constant was increased to avoid numerical instability (from `0.005` to `0.02`). Decrease simulation timestep to avoid altering the original value.
[W 07/04/25 19:59:02.641 193878] [frontend_ir.cpp:begin_frontend_struct_for_on_external_tensor@1694] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[Genesis] [19:59:05] [INFO] Compiling simulation kernels...
[Genesis] [19:59:17] [INFO] Building visualizer...
[Genesis] [19:59:18] [DEBUG] Trying to create OpenGL Context for PYOPENGL_PLATFORM='native'...
```

pschmutz, 2025-07-05:
Some observations

- `hello_genesis.py` now seems to work but unfortunately I am not aware of anything that I changed
- `IK_motion_planning_grasp.py` still did not work but I tried switching to Xorg, then it works as well
- Installing with `pip` or `uv` does not make a difference

nonlinearjunkie, 2025-12-19:
```
export __NV_PRIME_RENDER_OFFLOAD=1 \
export __GLX_VENDOR_LIBRARY_NAME=nvidia \
python hello_genesis.py
```

For me this worked!!