Genesis-Embodied-AI/genesis-world #1364: [Bug]: Make `Tensor had to be re-allocated because of incorrect dtype/device or non-contiguous memory` error print stack trace.

MRiabov, 2025-07-06:
### Bug Description

Hello,
I'm getting "Tensor had to be re-allocated because of incorrect dtype/device or non-contiguous memory. This may dramatically impede performance if it occurs in the critical path of your application." error in many places in my application. It is time-consuming to debug in which places it happens.

### Steps to Reproduce

Pass a torch float32 tensor to any function that does not expect it.

### Expected Behavior

It would be convenient if this would print a method or trace that this is called from.
e.g. 
Tensor had to be re-allocated because of incorrect dtype/device or non-contiguous memory. This may dramatically impede performance if it occurs in the critical path of your application. Happened at: {last 3 functions}

It looks like it is possible to do with `extract_tb` and related methods. See: https://docs.python.org/3/library/traceback.html#traceback.print_stack

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 24.04, Windows 11 24H2]
- GPU/CPU [e.g. A100, RTX 4090, M3pr, Intel I9-9900k, Ryzen 5900x] (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

git commit: 6479cf646de475206e85c9050dd283cae91c43d0 (latest as of 6th July)

TechArtistG, 2025-08-07:
I'm getting this as well:
Win 11, py 3.11, cuda 12.9, torch 2.8.0+cu129

duburcqa, 2025-08-11:
We should probably don't enable debug by default in the doc.