Genesis-Embodied-AI/genesis-world #1367: [Bug]:

chunantian, 2025-07-07:
### Bug Description

In Windows OS, when the genesis is installed through "pip install genesis-world", one error "AttributeError: function 'glGetUniformLocation' not found" occurs. According to the existing recommended solution, the latest version of the genesis is installed by "pip install -e ".[dev]"  ". However, the new error occurs as the follows:

Building editable for genesis-world (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building editable for genesis-world (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [23 lines of output]
      running editable_wheel
      creating C:\Users\XXX\AppData\Local\Temp\pip-wheel-qo0ow54z\.tmp-mwws7_mp\genesis_world.egg-info
      writing C:\Users\XXX\AppData\Local\Temp\pip-wheel-qo0ow54z\.tmp-mwws7_mp\genesis_world.egg-info\PKG-INFO
      writing dependency_links to C:\Users\XXX\AppData\Local\Temp\pip-wheel-qo0ow54z\.tmp-mwws7_mp\genesis_world.egg-info\dependency_links.txt      writing entry points to C:\Users\XXX\AppData\Local\Temp\pip-wheel-qo0ow54z\.tmp-mwws7_mp\genesis_world.egg-info\entry_points.txt        
      writing requirements to C:\Users\XXX\AppData\Local\Temp\pip-wheel-qo0ow54z\.tmp-mwws7_mp\genesis_world.egg-info\requires.txt
      writing top-level names to C:\Users\XXX\AppData\Local\Temp\pip-wheel-qo0ow54z\.tmp-mwws7_mp\genesis_world.egg-info\top_level.txt        
      writing manifest file 'C:\Users\XXX\AppData\Local\Temp\pip-wheel-qo0ow54z\.tmp-mwws7_mp\genesis_world.egg-info\SOURCES.txt'
      reading manifest file 'C:\Users\XXX\AppData\Local\Temp\pip-wheel-qo0ow54z\.tmp-mwws7_mp\genesis_world.egg-info\SOURCES.txt'
      reading manifest template 'MANIFEST.in'
      adding license file 'LICENSE'
      writing manifest file 'C:\Users\XXX\AppData\Local\Temp\pip-wheel-qo0ow54z\.tmp-mwws7_mp\genesis_world.egg-info\SOURCES.txt'
      creating 'C:\Users\XXX\AppData\Local\Temp\pip-wheel-qo0ow54z\.tmp-mwws7_mp\genesis_world-0.2.1.dist-info'
      C:\Users\XXX\AppData\Local\Temp\pip-build-env-0vkoiaum\overlay\Lib\site-packages\setuptools\command\bdist_wheel.py:103: RuntimeWarning: 
Config variable 'Py_DEBUG' is unset, Python ABI tag may be incorrect
        if get_flag("Py_DEBUG", hasattr(sys, "gettotalrefcount"), warn=(impl == "cp")):
      creating C:\Users\XXX\AppData\Local\Temp\pip-wheel-qo0ow54z\.tmp-mwws7_mp\genesis_world-0.2.1.dist-info\WHEEL
      running build_py
      running build_ext
      Compiling genesis/ext/fast_simplification/_simplify.pyx because it changed.
      [1/1] Cythonizing genesis/ext/fast_simplification/_simplify.pyx
      building 'genesis.ext.fast_simplification._simplify' extension
      b'*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00\r\x00\n\x00*\x00*\x00 \x00V\x00i\x00s\x00u\x00a\x00l\x00 \x00S\x00t\x00u\x00d\x00i\x00o\x00 \x002\x000\x001\x009\x00 \x00D\x00e\x00v\x00e\x00l\x00o\x00p\x00e\x00r\x00 \x00C\x00o\x00m\x00m\x00a\x00n\x00d\x00 \x00P\x00r\x00o\x00m\x00p\x00t\x00 \x00v\x001\x006\x00.\x001\x001\x00.\x002\x002\x00\r\x00\n\x00*\x00*\x00 \x00C\x00o\x00p\x00y\x00r\x00i\x00g\x00h\x00t\x00 \x00(\x00c\x00)\x00 \x002\x000\x002\x001\x00 \x00M\x00i\x00c\x00r\x00o\x00s\x00o\x00f\x00t\x00 \x00C\x00o\x00r\x00p\x00o\x00r\x00a\x00t\x00i\x00o\x00n\x00\r\x00\n\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00*\x00\r\x00\n\x00\x93\x8feQL\x88*Y\x7f\x95\x020\r\x00\n\x00}T\xe4N\xed\x8b\xd5l\rNcknx\x020\r\x00\n\x00'   
      error: Error executing cmd /u /c "C:\Software\MVS2019\Community\VC\Auxiliary\Build\vcvarsall.bat" x86_amd64 && set
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building editable for genesis-world
Failed to build genesis-world
ERROR: Failed to build installable wheels for some pyproject.toml based projects (genesis-world)

### Steps to Reproduce

pip install -e ".[dev]

### Expected Behavior

running

### Relevant log output

```shell

```

### Environment

- OS: [ Windows 10 22H2]
- GPU/CPU [RTX 3060] (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit 12.6 version (N/A if non-Nvidia)

### Release version or Commit ID

pip install -e ".[dev]

chunantian, 2025-07-09:
Sorry， the  #12 (comment) seems not give one solution for my request (Genesis-Embodied-AI/Genesis#1367). I hope to get your further help.

Thank you for your time to process my question.

Dexin Jiang

University of Chinese Academy of Sciences 

---- Replied Message ----
| From | Alexis ***@***.***> |
| Date | 7/8/2025 21:10 |
| To | ***@***.***> |
| Cc | ***@***.***> ,
***@***.***> |
| Subject | Re: [Genesis-Embodied-AI/Genesis] [Bug]: (Issue #1367) |
duburcqa left a comment (Genesis-Embodied-AI/Genesis#1367)

Installing a C++ is actually necessary to build Genesis from source. See #12 (comment)

duburcqa, 2025-07-09:
Did you managed to install MSVC ?