Genesis-Embodied-AI/genesis-world #1377: [Feature]: Add a simple API to add links

MRiabov, 2025-07-09:
### What feature or enhancement are you proposing?

Hello,
It seems there is no simple API to add a rigid link to a body. We can create MJCFs/UDRFs with just mesh and the link

### Motivation

As a RL researcher working on robotic assembly
I (sometimes) want to track positions that are not a center of a body. E.g., on an electronics connector, it is useful to track not the center of the connector, but its mating position. 
Because this enables tracking of bodies with interesting position different from meshes' `base_link`. 

### Potential Benefit

Currently I have to assign such links through MJCF, or rely on `base_link`. `base_link` does not necessarily mean the position of interest. MJCFs are verbose.

### What is the expected outcome of the implementation work?

- Using a simple API I can add a link to a body, relative to its base link.  
  - I won't have to use dynamic constraints for it.
  - I won't have to create MJCFs for it.
- I will be able to get its position using `get_links_pos`.

### Additional information

This should be fairly straightforward as MJCFs already support it, the only thing necessary is to add the API.

MRiabov, 2025-07-10:
> This is currently not supported
Hello,
Isn't links_to_keep=True meant for this? This is on MJCF, not URDF, but nevertheless.

duburcqa, 2025-07-10:
It is related, but not exactly the same. In one case it must be hard-coded in the URDF file, in the case it can be dynamically adapted to the entity without having to edit the URDF file. The latter is more flexible and less invasive.