Genesis-Embodied-AI/genesis-world #1407: [Bug]: URDF vs XML

ax-anoop, 2025-07-13:
### Bug Description

Hey I'm comparing the URDF loading vs XML (for unitree G1) 

[image]

I noticed that with latest update there's bene some improvements in correct parsing and matching the different loading processes, but there still seems to be some gaps. 

[g1_files.zip](https://github.com/user-attachments/files/21206528/g1_files.zip)

- Link data has been unified recently from a few months ago when I pulled (AWESOME) 
- Force limits different (but because xml does not have it, and urdf has "effort") 
- Kp / Kv defaults different, I'm not sure why this is the case ? Neither have these backed into the data files 
**- Damping, URDF data not being loaded** 
- Inverse weight is super different BUT once armature is matched manually, its roughly the same. 

[image]

URDF Data
  <joint name="left_knee_joint" type="revolute">
    <parent link="left_knee_motor"/>
    <child link="left_knee_link"/>
    <origin xyz="0 -0.043 0" rpy="0 0 0"/>
    <axis xyz="0 1 0"/>
    <limit lower="-1" upper="2.8798" effort="139" velocity="20"/>
    <dynamics damping="0.05" friction="0.2"/>
  </joint>

So the main issue I am seeing right now is damping not being loaded, but great job on the recent improvements guys. 

### Steps to Reproduce

If possible, provide a script triggering the bug, e.g.
```python
# code snippet triggering the bug
import genesis as gs
gs.init()
scene = gs.Scene(show_viewer=False)
broken = scene.add_entity(
    gs.morphs.MJCF(file='attachement.xml'),
)
scene.build()
```
Make sure to attached any needed assets (here `attachement.xml`)!

### Expected Behavior

Matching data

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 24.04, Windows 11 24H2]
- GPU/CPU [e.g. A100, RTX 4090, M3pr, Intel I9-9900k, Ryzen 5900x] (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

Latest

duburcqa, 2025-07-14:
Yes, a lot of effort has been made to unify loading a UDF and MJCF files. There is no a single loader under the hood, but some fields are still ignored for URDF. Regarding the difference for invweights, this is expected because it must be computed differently for URDF and MJCF files unless the robot has a fixed base and no armature inertia. There is no way around.

ax-anoop, 2025-09-18:
I belive this is resolved now & we can close ?

duburcqa, 2025-09-18:
Great!

duburcqa, 2025-09-18:
Wait no, damping and armature fields are still ignored in URDF files.