Genesis-Embodied-AI/genesis-world #1417: [Bug]: unuusal dof vel of robots when using terrain

Wataru-Oshima-Tokyo, 2025-07-15:
### Bug Description

When training Unitree Go2 on terrain, the value of base_lin_vel sometimes becomes abnormally large.
This tends to occur especially when the robot falls over or tilts, and since it didn’t happen in previous versions, I suspect that recent changes related to terrain are causing this issue.

### Steps to Reproduce

ussing the below script (go2_env.py) can reporduce the error.
I simply 
1. changed the termination degree from 10 to 150
2. replaced plane with terrain
3. increse the number of visualized robots 
4. added arrows to visualize dof_vel of each robot and given commands

It could happen regardless of the number of visualized robots or wih/without arrows.

----------------------------- #go2_train.py ----------------------------------
import argparse
import os
import pickle
import shutil
from importlib import metadata

try:
    try:
        if metadata.version("rsl-rl"):
            raise ImportError
    except metadata.PackageNotFoundError:
        if metadata.version("rsl-rl-lib") != "2.2.4":
            raise ImportError
except (metadata.PackageNotFoundError, ImportError) as e:
    raise ImportError("Please uninstall 'rsl_rl' and install 'rsl-rl-lib==2.2.4'.") from e
from rsl_rl.runners import OnPolicyRunner

import genesis as gs

from go2_env import Go2Env

def get_train_cfg(exp_name, max_iterations):
    train_cfg_dict = {
        "algorithm": {
            "class_name": "PPO",
            "clip_param": 0.2,
            "desired_kl": 0.01,
            "entropy_coef": 0.01,
            "gamma": 0.99,
            "lam": 0.95,
            "learning_rate": 0.001,
            "max_grad_norm": 1.0,
            "num_learning_epochs": 5,
            "num_mini_batches": 4,
            "schedule": "adaptive",
            "use_clipped_value_loss": True,
            "value_loss_coef": 1.0,
        },
        "init_member_classes": {},
        "policy": {
            "activation": "elu",
            "actor_hidden_dims": [512, 256, 128],
            "critic_hidden_dims": [512, 256, 128],
            "init_noise_std": 1.0,
            "class_name": "ActorCritic",
        },
        "runner": {
            "checkpoint": -1,
            "experiment_name": exp_name,
            "load_run": -1,
            "log_interval": 1,
            "max_iterations": max_iterations,
            "record_interval": -1,
[... 451 lines cut ...]

dof_vel should be stable 

### Screenshots/Videos

https://github.com/user-attachments/assets/f4b4b693-0f83-4b52-bb5a-2a29b049d302

### Relevant log output

```shell

```

### Environment

- OS: Ubuntu 22.04,
- GPU/CPU: RTX 4090
- GPU-driver version: 550.120
- CUDA / CUDA-toolkit version:12.4

### Release version or Commit ID

The bug confrimed: the latest version

The bug not confirmed:  382cf4ca12c0c142adcf2fa7675eef65caf0c661

Wataru-Oshima-Tokyo, 2025-09-05:
still happens

ax-anoop, 2025-09-18:
I wonder if this is related to https://github.com/Genesis-Embodied-AI/Genesis/issues/1727 

@duburcqa 

I can confirm training with terrain's hasn't worked as it should