Genesis-Embodied-AI/genesis-world #1425: [Bug]: Rigid Collision Detection is unable to return collision information

jolley-x, 2025-07-16:
### Bug Description

Rigid Collision Detection is unable to return collision information.

### Steps to Reproduce

import numpy as np

import genesis as gs

def init_genesis_backend():
    """Initialize the Genesis backend, prioritizing the CPU backend"""
    try:
        gs.init(backend=gs.cpu)
        return True
    except AttributeError:
        print("Warning: Unable to initialize CPU backend, using default configuration")
        return False
    except Exception as e:
        print(f"Failed to initialize Genesis: {str(e)}")
        return False

def load_stl_model(path: str, position: tuple = (0, 0, 0)) -> gs.morphs.Mesh:
    """
    Load an STL model and create a mesh object with collision detection
    
    Parameters:
        path: STL file path
        position: Initial position coordinates of the model (x, y, z)
    
    Returns:
        Initialized mesh object
        
    Exceptions:
        Throws an exception if loading fails
    """
    try:
        # Create mesh object with collision detection and visualization enabled
        return gs.morphs.Mesh(
            file=path,
            scale=1.0,
            pos=position,
            collision=True,       # Enable collision detection
            visualization=True,   # Enable visualization
            is_free=False,        # Not a free object
            fixed=True            # Position is fixed
        )
    except FileNotFoundError:
        raise FileNotFoundError(f"STL file not found: {path}")
    except Exception as e:
        raise RuntimeError(f"Failed to load STL file ({path}): {str(e)}")

def check_collision(entity1, entity2, scene) -> tuple[bool, list]:
    """
    Check if two entities are colliding
    
    Parameters:
        entity1: First entity
        entity2: Second entity
        scene: Current scene object
        
    Returns:
        Tuple (collision occurred, list of collision points)
    """
    collision_occurred = False
    collision_points = []
    
    # Get geometry IDs of entities
    geoms1 = entity1.geoms if hasattr(entity1, 'geoms') else []
    geoms2 = entity2.geoms if hasattr(entity2, 'geoms') else []
[... 223 lines cut ...]
        print(f"Program execution error: {str(e)}")

if __name__ == "__main__":
    main()

### Expected Behavior

The return value of Rigid Collision Detection can be obtained.

### Relevant log output

```shell

```

### Environment

OS: window11
GPU/CPU: RTX3060
GPU-driver version: 550.120
CUDA / CUDA-toolkit version:12.4

### Release version or Commit ID

The bug confrimed: the latest version

SonSang, 2025-07-16:
Hi, in the provided script, I see that you are querying contact information directly in `check_collision3` function. Specifically, you are trying to access contact data through `rigid_solver` at line 224 and 225. However, `rigid_solver` does not have such fields.

Rather than that, I recommend querying the contact data through `get_contacts` function in the `Collider` ([here](https://github.com/Genesis-Embodied-AI/Genesis/blob/867a86870d17c5280892f1996a901390da4029cd/genesis/engine/solvers/rigid/collider_decomp.py#L2865)). You can do like `rigid_solver.collider.get_contacts()`. Then you do not have to directly access the fields, which could be vulnerable to future code refactoring.

duburcqa, 2025-07-16:
The recommended public API to check collision between 2 entities this one
```python
entity1.get_contacts(with_entity=entity2, exclude_self_contact=True)
```

Accessing Taichi fields directly is not officially supported.

jolley-x, 2025-07-17:
Is collision detection only possible under the rigid body motion state? The collision point can only be returned when fixed=false.
```python
import numpy as np

import genesis as gs

def init_genesis_backend():
    """初始化Genesis后端，优先使用CPU后端"""
    try:
        gs.init(backend=gs.cpu)
        print("成功初始化CPU后端")
        return True
    except AttributeError:
        print("警告：无法初始化CPU后端，使用默认配置")
        return False
    except Exception as e:
        print(f"初始化Genesis失败: {str(e)}")
        return False

def load_stl_model(path: str, position: tuple = (0, 0, 0)) -> gs.morphs.Mesh:
    """加载STL模型并创建带碰撞检测的网格对象"""
    try:
        mesh = gs.morphs.Mesh(
            file=path,
            scale=1.0,
            pos=position,
            collision=True,       # 启用碰撞检测
            visualization=True,   # 启用可视化
            is_free=False,       
            fixed=False           
        )
        
        # 启用碰撞形状可视化
        if hasattr(mesh, 'show_collision'):
            mesh.show_collision = True
            
        print(f"成功加载模型: {path}")
        return mesh
    except FileNotFoundError:
        raise FileNotFoundError(f"STL文件未找到: {path}")
    except Exception as e:
        raise RuntimeError(f"加载STL文件失败 ({path}): {str(e)}")

def check_collision(entity1, entity2, scene) -> tuple[bool, list]:

    collision_occurred = False
    collision_points = []
    
    try:
        # 确保碰撞检测流程已执行
        if hasattr(scene, 'rigid_solver') and hasattr(scene.rigid_solver.collider, 'detection'):
            scene.rigid_solver.collider.detection()
        
        # 获取两个实体间的接触信息
        contacts = entity1.get_contacts(with_entity=entity2, exclude_self_contact=True)
        
        # 检查是否有碰撞点（处理字典形式的返回结果）
        if isinstance(contacts, dict) and 'position' in contacts:
            # 检查位置张量是否包含数据
            if contacts['position'].numel() > 0:
                collision_occurred = True
                
                # 将PyTorch张量转换为NumPy数组并提取坐标
                positions = contacts['position'].cpu().numpy()
                
                # 遍历所有碰撞点
                for pos in positions:
                    if len(pos) >= 3:  # 确保是三维坐标
                        collision_points.append((float(pos[0]), float(pos[1]), float(pos[2])))
        
[... 145 lines cut ...]
[Genesis] [10:57:25] [INFO] Consider setting 'performance_mode=True' in production to maximise runtime speed, if significantly increasing compilation time is not a concern.
[Genesis] [10:57:25] [INFO] Running on [AMD Ryzen 7 5800H with Radeon Graphics] with backend gs.cpu. Device memory: 15.87 GB.
[Genesis] [10:57:25] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
成功初始化CPU后端
成功加载模型: D:\ModelData\20250708_145143\link7.stl
成功加载模型: D:\ModelData\20250708_145143\link7.stl
[Genesis] [10:57:26] [INFO] Scene <ec59a0d> created.
[Genesis] [10:57:26] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <0867c10>, morph: <gs.morphs.Mesh(file='D:\ModelData\20250708_145143\link7.stl')>, material: <gs.materials.Rigid>.
[Genesis] [10:57:26] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <9546c5c>, morph: <gs.morphs.Mesh(file='D:\ModelData\20250708_145143\link7.stl')>, material: <gs.materials.Rigid>.
[Genesis] [10:57:26] [INFO] Building scene <ec59a0d>...
[Genesis] [10:57:33] [INFO] Compiling simulation kernels...
[Genesis] [10:57:51] [INFO] Building visualizer...
[Genesis] [10:57:53] [INFO] Viewer created. Resolution: 1600×1200, max_FPS: 60.
场景构建成功
开始碰撞检测仿真...
第0步: 未检测到碰撞
[Genesis] [10:57:53] [INFO] Running at 29.27 FPS.
[Genesis] [10:57:53] [INFO] Running at 30.01 FPS.
[Genesis] [10:57:53] [INFO] Running at 30.76 FPS.
[Genesis] [10:57:54] [INFO] Running at 31.51 FPS.
[Genesis] [10:57:54] [INFO] Running at 32.25 FPS.
[Genesis] [10:57:54] [INFO] Running at 32.96 FPS.
[Genesis] [10:57:54] [INFO] Running at 33.68 FPS.
[Genesis] [10:57:54] [INFO] Running at 34.43 FPS.
[Genesis] [10:57:54] [INFO] Running at 35.14 FPS.

SonSang, 2025-07-17:
This is because we ignore collisions between two fixed objects as shown [here](https://github.com/Genesis-Embodied-AI/Genesis/blob/d2017358463e3f10f1a02177f48a2156245f4ea2/genesis/engine/solvers/rigid/collider_decomp.py#L206). We skip collision test for the fixed pair of objects. You should set at least one of the objects to be not fixed.

jolley-x, 2025-07-17:
Our requirement is to simulate the movement of two entities by loading table data, so I need both entities to be fixed. If they are not fixed, their positions will shift due to factors like gravity, making it impossible to obtain accurate collision data. Are there any alternative solutions currently available? For example, disabling the physical properties of both entities while retaining only the collision calculation, or are there any other fixing methods?

duburcqa, 2025-07-17:
Using a physics engine for collision detection only is far from ideal as it was not designed for this. If you want to use genesis, you should expect having to do some hack. 

At this point, I still don't understand why you are trying to achieve. Why are you running multiple simulation steps if the entities are fixed ? Without will change anyway so a single one should be fine?

duburcqa, 2025-07-18:
> it is only used during testing to wait for the UI to display; otherwise, the program will end before the UI can show up.

You should rather wait for user input: `input()`

> Under normal circumstances, position information is input via a table, and each position is calculated only once. 

In this case, you are lucky because there is an issue solution:

Right after building the scene, you should do:
```
scene.rigid_solver.collider._collider_info.collision_pair_validity.full(True)
```

duburcqa, 2025-07-18:
> Now, I also want to get points that are less than distance X. For example, a certain point of model A and model B is less than distance X (Euclidean distance), but they are not colliding, just like a safety distance for train braking. In this case, what should I do?

This should be possible with GJK I think, what do you think @SonSang ?

SonSang, 2025-07-18:
> Now, I also want to get points that are less than distance X. For example, a certain point of model A and model B is less than distance X (Euclidean distance), but they are not colliding, just like a safety distance for train braking. In this case, what should I do?

It's possible to get distance through GJK, but right now GJK is mostly dedicated to detecting collisions. That is, if we confirm that the two objects are separated, we just finish the algorithm without further refining the distance between them. It is one of our future works to implement it, but it would take some time for it.

duburcqa, 2025-07-21:
But do we get some lower- or upper-bound at this stage?

jolley-x, 2025-07-28:
def check_gjk_proximity(entity1, entity2, scene, threshold=0.01) -> tuple[bool, list]:
    """
    使用 GJK 检测两个刚体之间是否碰撞或接近。
    返回：是否距离小于阈值，和最近点对列表。
    """
    is_close = False
    closest_points = []

    try:
        # 初始化 GJK（只需一次）
        if not hasattr(scene.rigid_solver, "gjk"):
            from genesis.engine.solvers.rigid.gjk_decomp import GJK
            scene.rigid_solver.gjk = GJK(scene.rigid_solver)

        gjk = scene.rigid_solver.gjk

        i_ga = entity1.idx
        i_gb = entity2.idx
        i_b = 0  # batch index

  
        @ti.kernel
        def run_gjk():
            gjk.func_gjk_contact(i_ga, i_gb, i_b)

        run_gjk()  # 正确调用方式

        dist = gjk.distance[i_b]

        if dist < threshold:
            is_close = True
            for i in range(gjk.n_witness[i_b]):
                w1 = gjk.witness[i_b, i].point_obj1
                w2 = gjk.witness[i_b, i].point_obj2
                closest_points.append((
                    float(w1[0]), float(w1[1]), float(w1[2]),
                    float(w2[0]), float(w2[1]), float(w2[2])
                ))

    except Exception as e:
        print(f"GJK检测失败: {e}")

    return is_close, closest_points

collision, points = check_gjk_proximity(entity1, entity2, scene,10000)

Hello, is my usage of GJK incorrect? Why is there no result returned? Also, _n_contacts_per_pair = 5, which seems a bit few. Can this number be increased, and where can I set it?

jolley-x, 2025-07-29:
I have used get_contacts before and got the results, but only points after collision can be returned, and points with a distance less than X will not be returned. So, following your suggestion, I added gjk, which is here in gjk_decomp.py ([here](https://github.com/Genesis-Embodied-AI/Genesis/blob/81e91e6c99cc883b8a57e058b56fdc080e67a2d6/genesis/engine/solvers/rigid/gjk_decomp.py#L176)). However, there is no return value after I run it. As for more contact points, it's because our model is relatively large, and we hope to obtain the collision area more accurately.  

For the first time, following the documentation's recommendations, the configuration used was that RigidOptions is set to True.

In Genesis, it is enabled when use_gjk_collision option in RigidOptions is set to be True. Also, Genesis enhances the robustness of GJK with following measures.
```python
def check_collision(entity1, entity2, scene, threshold: float = 0.0) -> tuple[bool, list]:
    """检查两个实体是否发生碰撞，仅返回距离小于阈值的碰撞点
    
    参数:
        entity1, entity2: 要检测的两个实体
        scene: 场景对象
        threshold: 距离阈值，小于等于该值的碰撞点才会被返回
        
    返回:
        tuple[bool, list]: 碰撞状态和符合阈值条件的碰撞点列表
    """
    collision_occurred = False
    collision_points = []

    try:
        # 确保启用GJK碰撞检测（根据Genesis文档要求）
        if hasattr(scene, "rigid_options") and hasattr(scene.rigid_options, "use_gjk_collision"):
            if not scene.rigid_options.use_gjk_collision:
                print("启用GJK碰撞检测（根据配置要求）")
                scene.rigid_options.use_gjk_collision = True
                
                # 重置求解器使配置生效
                if hasattr(scene, "rigid_solver") and hasattr(scene.rigid_solver, "reset"):
                    scene.rigid_solver.reset()

        # 强制刷新碰撞检测
        if hasattr(scene, "rigid_solver") and hasattr(scene.rigid_solver.collider, "detection"):
            # 清除碰撞缓存
            if hasattr(scene.rigid_solver.collider, "_contacts_info_cache"):
                scene.rigid_solver.collider._contacts_info_cache.clear()
            # 执行GJK检测流程
            scene.rigid_solver.collider.detection()

        # 获取接触对信息（用于距离计算）
        contact_pair = None
        if hasattr(scene.rigid_solver.collider, "get_contact_pair"):
            contact_pair = scene.rigid_solver.collider.get_contact_pair(entity1.id, entity2.id)

        # 获取接触点信息
        contacts = entity1.get_contacts(with_entity=entity2, exclude_self_contact=True)

        # 解析并筛选符合阈值的碰撞点
        if isinstance(contacts, dict) and "position" in contacts and contacts["position"].numel() > 0:
            positions = contacts["position"].cpu().numpy()
            
            # 检查是否有有效接触对信息用于距离判断
            if contact_pair is not None and hasattr(contact_pair, "separation_distance"):
                separation_dist = float(contact_pair.separation_distance)
                # 对于碰撞状态（分离距离 <= 0）或接近状态但在阈值内的情况
                if separation_dist <= threshold:
                    collision_occurred = True
                    for pos in positions:
                        if len(pos) >= 3:
                            collision_points.append((float(pos[0]), float(pos[1]), float(pos[2])))
            else:
                # 没有距离信息时，默认所有接触点都符合条件
                collision_occurred = True
                for pos in positions:
                    if len(pos) >= 3:
                        collision_points.append((float(pos[0]), float(pos[1]), float(pos[2])))

        # 调试：AABB重叠状态
        if hasattr(entity1, "aabb") and hasattr(entity2, "aabb"):
            aabb1_min, aabb1_max = entity1.aabb
            aabb2_min, aabb2_max = entity2.aabb
            aabb_overlap = (
                aabb1_min[0] <= aabb2_max[0] and aabb1_max[0] >= aabb2_min[0] and
                aabb1_min[1] <= aabb2_max[1] and aabb1_max[1] >= aabb2_min[1] and
                aabb1_min[2] <= aabb2_max[2] and aabb1_max[2] >= aabb2_min[2]
            )
            if not aabb_overlap and hasattr(scene, "time_step"):
                print(f"调试: AABB已分离（步骤{scene.time_step}）")

    except AttributeError as ae:
        print(f"碰撞检测属性错误: {str(ae)}")
    except Exception as e:
        print(f"碰撞检测错误: {str(e)}")

    return collision_occurred, collision_points
      collision, points = check_collision(entity1, entity2, scene,10000)
``` 

For the second time, following the documentation's recommendations, the one used was func_gjk_contact.

```python
def check_gjk_proximity(entity1, entity2, scene, threshold=0.01) -> tuple[bool, list]:
    """
    使用 GJK 检测两个刚体之间是否碰撞或接近。
    返回：是否距离小于阈值，和最近点对列表。
    """
    is_close = False
    closest_points = []

    try:
        # 初始化 GJK（只需一次）
        if not hasattr(scene.rigid_solver, "gjk"):
            from genesis.engine.solvers.rigid.gjk_decomp import GJK
            scene.rigid_solver.gjk = GJK(scene.rigid_solver)

        gjk = scene.rigid_solver.gjk

        i_ga = entity1.idx
        i_gb = entity2.idx
        i_b = 0  # batch index

  
        @ti.kernel
        def run_gjk():
            gjk.func_gjk_contact(i_ga, i_gb, i_b)

        run_gjk()  # 正确调用方式

        dist = gjk.distance[i_b]

        if dist < threshold:
            is_close = True
            for i in range(gjk.n_witness[i_b]):
                w1 = gjk.witness[i_b, i].point_obj1
                w2 = gjk.witness[i_b, i].point_obj2
                closest_points.append((
                    float(w1[0]), float(w1[1]), float(w1[2]),
                    float(w2[0]), float(w2[1]), float(w2[2])
                ))

    except Exception as e:
        print(f"GJK检测失败: {e}")

    return is_close, closest_points

collision, points = check_gjk_proximity(entity1, entity2, scene,10000)
``` 
Neither of the two code snippets returns a value, and the safety distance I set was also very large.

jolley-x, 2025-07-30:
Thank you for your explanation, but after setting it up according to your suggestions, there are still no results returned.

I have already set the threshold extremely high.

The output results of my code

```python
collision, points = check_collision(entity1, entity2, scene,10000000000)
def check_collision(entity1, entity2, scene, threshold: float = 0.0) -> tuple[bool, list]:
    """检查两个实体是否发生碰撞，仅返回距离小于阈值的碰撞点
    
    参数:
        entity1, entity2: 要检测的两个实体
        scene: 场景对象
        threshold: 距离阈值，小于等于该值的碰撞点才会被返回
        
    返回:
        tuple[bool, list]: 碰撞状态和符合阈值条件的碰撞点列表
    """
    collision_occurred = False
    collision_points = []

    try:
        # 确保启用GJK碰撞检测（根据Genesis文档要求）
        if hasattr(scene, "rigid_options") and hasattr(scene.rigid_options, "use_gjk_collision"):
            if not scene.rigid_options.use_gjk_collision:
                print("启用GJK碰撞检测（根据配置要求）")
                scene.rigid_options.use_gjk_collision = True
                
                # 重置求解器使配置生效
                if hasattr(scene, "rigid_solver") and hasattr(scene.rigid_solver, "reset"):
                    scene.rigid_solver.reset()

        # 强制刷新碰撞检测
        if hasattr(scene, "rigid_solver") and hasattr(scene.rigid_solver.collider, "detection"):
            # 清除碰撞缓存
            if hasattr(scene.rigid_solver.collider, "_contacts_info_cache"):
                scene.rigid_solver.collider._contacts_info_cache.clear()
            # 执行GJK检测流程
            scene.rigid_solver.collider.detection()

        # 获取接触对信息（用于距离计算）
        contact_pair = None
        if hasattr(scene.rigid_solver.collider, "get_contact_pair"):
            contact_pair = scene.rigid_solver.collider.get_contact_pair(entity1.id, entity2.id)

        # 获取接触点信息
        contacts = entity1.get_contacts(with_entity=entity2, exclude_self_contact=True)
        print(dir(contacts))
        print("Contacts keys:", contacts.keys())  # 查看实际包含的键
        # 新增：打印position的元素数量和是否为空
        if "position" in contacts:
            print(f"position numel: {contacts['position'].numel()}")  # 查看是否有数据
            if contacts["position"].numel() > 0:
                print(f"position data: {contacts['position'].cpu().numpy()}")  # 查看具体数据
            else:
                print("position 为空，无接触点数据")
        # 解析并筛选符合阈值的碰撞点
        if isinstance(contacts, dict) and "position" in contacts and contacts["position"].numel() > 0:
            positions = contacts["position"].cpu().numpy()
            
            # 检查是否有有效接触对信息用于距离判断
            if contact_pair is not None and hasattr(contact_pair, "separation_distance"):
                separation_dist = float(contact_pair.separation_distance)
                # 对于碰撞状态（分离距离 <= 0）或接近状态但在阈值内的情况
                if separation_dist <= threshold:
                    collision_occurred = True
                    for pos in positions:
                        if len(pos) >= 3:
                            collision_points.append((float(pos[0]), float(pos[1]), float(pos[2])))
[... 30 lines cut ...]
[Genesis] [13:30:35] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [13:30:35] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [13:30:35] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [13:30:38] [INFO] Consider setting 'performance_mode=True' in production to maximise runtime speed, if significantly increasing compilation time is not a concern.
[Genesis] [13:30:38] [INFO] Running on [AMD Ryzen 7 5800H with Radeon Graphics] with backend gs.cpu. Device memory: 15.87 GB.
[Genesis] [13:30:38] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
成功初始化CPU后端
成功加载模型: D:\ModelData\20250708_145143\UP.stl
成功加载模型: D:\ModelData\20250708_145143\PG-L.stl
[Genesis] [13:30:39] [INFO] Scene <315152c> created.
[Genesis] [13:30:39] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <98e390a>, morph: <gs.morphs.Mesh(file='D:\ModelData\20250708_145143\UP.stl')>, material: <gs.materials.Rigid>.
[Genesis] [13:30:42] [INFO] Convex hull is not accurate enough for collision detection (1.981). Falling back to more expensive convex decomposition (see FileMorph options).
[Genesis] [13:30:43] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <c69d0ce>, morph: <gs.morphs.Mesh(file='D:\ModelData\20250708_145143\PG-L.stl')>, material: <gs.materials.Rigid>.
[Genesis] [13:30:45] [INFO] Convex hull is not accurate enough for collision detection (1.708). Falling back to more expensive convex decomposition (see FileMorph options).
[Genesis] [13:30:46] [INFO] Building scene <315152c>...
[Genesis] [13:30:53] [INFO] Compiling simulation kernels...
[Genesis] [13:31:13] [INFO] Building visualizer...
[Genesis] [13:31:53] [INFO] Viewer created. Resolution: 1600×1200, max_FPS: 60.
场景构建成功
开始碰撞检测仿真...
['__class__', '__class_getitem__', '__contains__', '__delattr__', '__delitem__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__getitem__', '__getstate__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__ior__', '__iter__', '__le__', '__len__', '__lt__', '__ne__', '__new__', '__or__', '__reduce__', '__reduce_ex__', '__repr__', '__reversed__', '__ror__', '__setattr__', '__setitem__', '__sizeof__', '__str__', '__subclasshook__', 'clear', 'copy', 'fromkeys', 'get', 'items', 'keys', 'pop', 'popitem', 'setdefault', 'update', 'values']
Contacts keys: dict_keys(['link_a', 'link_b', 'geom_a', 'geom_b', 'penetration', 'position', 'normal', 'force_a', 'force_b'])
position numel: 0
position 为空，无接触点数据
```

SonSang, 2025-07-30:
I see. As we said above, currently Genesis is not optimized to give your desired information. To suit your requirements, I made a branch in my repo, you can check out [here](https://github.com/SonSang/Genesis/tree/gjk_distance). There is an [example script](https://github.com/SonSang/Genesis/blob/gjk_distance/examples/gjk_distance/main.py), where I wrote a small demo to print out contact information if there is. If there is no contact, we compute the minimum distance between the objects and return the distance and witness points (the points that form the minimum distance). You can tweak [this line](https://github.com/SonSang/Genesis/blob/5a5a9fcbc53ce207b0b17718f6d26587482ac10a/examples/gjk_distance/main.py#L8) to see how the results change when the balls in the scene collide or not. 

When the balls are in the contact at the initial state, the output is like:
```
========== Step 0 ==========
Contact detected
Contact 0:
  Position: tensor([0.4500, 0.0000, 0.0000])
  Normal: tensor([-0.9999,  0.0093, -0.0093])
  Penetration: 0.09999141097068787
========== Step 1 ==========
[Genesis] [11:56:37] [INFO] Running at 12.33 FPS.
Contact detected
Contact 0:
  Position: tensor([0.4500, 0.0000, 0.0000])
  Normal: tensor([-0.9999,  0.0093, -0.0093])
  Penetration: 0.0749964788556099
========== Step 2 ==========
[Genesis] [11:56:37] [INFO] Running at 12.77 FPS.
Contact detected
Contact 0:
  Position: tensor([0.4500, 0.0000, 0.0000])
  Normal: tensor([-9.9983e-01, -1.8567e-02, -5.8204e-10])
  Penetration: 0.050003305077552795
========== Step 3 ==========
[Genesis] [11:56:37] [INFO] Running at 13.30 FPS.
Contact detected
Contact 0:
  Position: tensor([0.4500, 0.0000, 0.0000])
  Normal: tensor([-9.9999e-01, -2.8694e-05, -4.6534e-03])
  Penetration: 0.025011925026774406
========== Step 4 ==========
[Genesis] [11:56:37] [INFO] Running at 13.83 FPS.
Contact detected
Contact 0:
  Position: tensor([0.4500, 0.0000, 0.0000])
  Normal: tensor([-1.0000,  0.0025,  0.0013])
  Penetration: 1.735846308292821e-05
========== Step 5 ==========
[Genesis] [11:56:37] [INFO] Running at 14.39 FPS.
No contact detected, computing distance
Minimum distance: 0.04997355863451958
Witness point 1: [ 4.25013214e-01  1.16518349e-05 -1.16518931e-05]
Witness point 2: [ 4.74986762e-01 -1.16518349e-05  1.16518931e-05]
========== Step 6 ==========
[Genesis] [11:56:37] [INFO] Running at 11.75 FPS.
No contact detected, computing distance
Minimum distance: 0.07496925443410873
Witness point 1: [ 4.12515372e-01  1.99188653e-05 -1.99188944e-05]
Witness point 2: [ 4.87484604e-01 -1.99188653e-05  1.99188944e-05]
========== Step 7 ==========
[Genesis] [11:56:37] [INFO] Running at 12.24 FPS.
No contact detected, computing distance
Minimum distance: 0.09996489435434341
Witness point 1: [ 4.00017560e-01  2.96646613e-05 -2.96647195e-05]
Witness point 2: [ 4.99982417e-01 -2.96646613e-05  2.96647195e-05]
========== Step 8 ==========
[Genesis] [11:56:37] [INFO] Running at 12.75 FPS.
No contact detected, computing distance
Minimum distance: 0.12496056407690048
Witness point 1: [ 3.87519747e-01  4.07905027e-05 -4.07907064e-05]
Witness point 2: [ 5.12480259e-01 -4.07905027e-05  4.07907064e-05]
========== Step 9 ==========
[Genesis] [11:56:38] [INFO] Running at 13.27 FPS.
No contact detected, computing distance
Minimum distance: 0.14995618164539337
Witness point 1: [ 3.75021935e-01  5.32064296e-05 -5.32067206e-05]
Witness point 2: [ 5.24978042e-01 -5.32064296e-05  5.32067206e-05]
```

You can see that the initial collision is resolved because of our physics solver. When the contact is resolved, the code computes the exact minimum distance between them and gives the witness points.

I hope this code helps. However, note that this branch is not an official release, which could be erroneous and slower. I believe this is the best we can do for you in the current stage.

jolley-x, 2025-07-31:
Hello, thank you for providing the example. Does this have any restrictions on the model? I encounter issues when loading larger models. For instance, with a model containing 800,000 vertices, the result returned is problematic. Additionally, currently, only one point is returned.
```shell
========== 第 0 次迭代 ==========
position1: (0.0, 0.0, 800.0)
position2: (280.0, 399.99999999999994, 150.0)
开始碰撞检测仿真...
------ 步骤 0 ------
检测到 1 个碰撞点:
碰撞点 0:
  distance: 812.86328125
  witness_point1: (0.0375153087079525, 0.0792456790804863, 800.0767211914062)
  witness_point2: (279.9842834472656, 399.9737548828125, 150.10679626464844)

========== 第 1 次迭代 ==========
position1: (0.0, 0.0, 799.9582648)
position2: (280.0, 411.42242031747844, 150.0)
开始碰撞检测仿真...
------ 步骤 0 ------
检测到 1 个碰撞点:
碰撞点 0:
  distance: 818.509765625
  witness_point1: (0.0375153087079525, 0.0792456790804863, 800.0349731445312)
  witness_point2: (279.97906494140625, 411.38433837890625, 150.09722900390625)

========== 第 2 次迭代 ==========
position1: (0.0, 0.0, 799.833056)
position2: (280.0, 422.83552434510716, 150.0)
开始碰撞检测仿真...
------ 步骤 0 ------
检测到 1 个碰撞点:
碰撞点 0:
  distance: 824.2052612304688
  witness_point1: (0.0375153087079525, 0.0792456790804863, 799.9097900390625)
  witness_point2: (279.97906494140625, 422.79742431640625, 150.09722900390625)

========== 第 3 次迭代 ==========
position1: (0.0, 0.0, 799.6243644)
position2: (280.0, 434.23000339153594, 150.0)
开始碰撞检测仿真...
------ 步骤 0 ------
检测到 1 个碰撞点:
碰撞点 0:
  distance: 829.9435424804688
  witness_point1: (0.0375153087079525, 0.0792456790804863, 799.7011108398438)
  witness_point2: (279.97906494140625, 434.1919250488281, 150.09722900390625)

========== 第 4 次迭代 ==========
position1: (0.0, 0.0, 799.3321746)
position2: (280.0, 445.59656395621624, 150.0)
开始碰撞检测仿真...
------ 步骤 0 ------
检测到 1 个碰撞点:
碰撞点 0:
  distance: 835.718017578125
  witness_point1: (0.0375153087079525, 0.0792456790804863, 799.4088745117188)
  witness_point2: (279.97906494140625, 445.5584716796875, 150.09722900390625)

========== 第 5 次迭代 ==========
position1: (0.0, 0.0, 798.9564654)
position2: (280.0, 456.925935309314, 150.0)
开始碰撞检测仿真...
------ 步骤 0 ------
检测到 1 个碰撞点:
碰撞点 0:
  distance: 841.52197265625
  witness_point1: (0.0375153087079525, 0.0792456790804863, 799.033203125)
  witness_point2: (279.97906494140625, 456.8878479003906, 150.09722900390625)

========== 第 6 次迭代 ==========
position1: (0.0, 0.0, 798.4972093)
[... 130 lines cut ...]
  witness_point2: (0.3468244671821594, 1878.596923828125, 822.4879150390625)

========== 第 8 次迭代 ==========
position1: (0.0, 0.0, 797.3279182)
position2: (280.0, 490.5987069703058, 150.0)
开始碰撞检测仿真...
------ 步骤 0 ------
检测到 1 个碰撞点:
碰撞点 0:
  distance: 0.07905066013336182
  witness_point1: (0.42587512731552124, 1878.59716796875, 821.8615112304688)
  witness_point2: (0.3468244969844818, 1878.5970458984375, 821.8615112304688)

========== 第 9 次迭代 ==========
position1: (0.0, 0.0, 796.6178)
position2: (280.0, 501.6873336773947, 150.0)
开始碰撞检测仿真...
------ 步骤 0 ------
检测到 1 个碰撞点:
碰撞点 0:
  distance: 0.07905066758394241
  witness_point1: (0.42587515711784363, 1878.59716796875, 821.1513671875)
  witness_point2: (0.3468245267868042, 1878.59716796875, 821.1513671875)

```

SonSang, 2025-07-31:
Yes, GJK returns the "minimum" distance between the objects, so there is only one pair of points. It seems that it is working as expected.

jolley-x, 2025-08-05:
Hello, you're right. Large 3D models will be split when loaded. Here is the output.
```python
    # 获取实体1的geoms数量
    geoms1_count = len(entity1.geoms)

    # 获取实体2的geoms数量
    geoms2_count = len(entity2.geoms)

    # 打印结果
    print(f"实体1的几何组件数量: {geoms1_count}")
    print(f"实体2的几何组件数量: {geoms2_count}")
``` 
```shell
实体1的几何组件数量: 14
实体2的几何组件数量: 1
``` 
But I want to get the minimum point between entity1 and entity2, not between entity1 and itself.
Here are the changes I've made, but none of them have worked. Are there any other solutions available?
```python
contacts = entity1.get_contacts(with_entity=entity2, exclude_self_contact=True)

        mesh = gs.morphs.Mesh(
            file=path,
            scale=1.0,
            pos=position,
            collision=True,
            visualization=True,
            is_free=False,
            decimate=False,
            convexify=False,
            euler=(-90,0,0),
            fixed=True
 
        )
``` 
```shell
========== 第 0 次迭代 ==========
position1: (0.0, 0.0, 800.0)
position2: (280.0, 399.99999999999994, 150.0)
开始碰撞检测仿真...
------ 步骤 0 ------
检测到 1 个碰撞点:
碰撞点 0:
  distance: 0.07905066013336182
  witness_point1: (0.42587512731552124, 1878.596923828125, 824.5335693359375)
  witness_point2: (0.3468244671821594, 1878.596923828125, 824.5335693359375)
```

jolley-x, 2025-08-05:
According to your suggestion, each geom is calculated separately, but the calculation result is incorrect and the speed is relatively slow.
```python
def check_collision(entity1, entity2, scene, threshold: float = 0.0) -> tuple[bool, list]:
    """检查两个实体是否发生碰撞，仅返回距离小于阈值的碰撞点
    处理非凸模型：遍历实体1和实体2的所有几何组件组合，计算最小距离
    """
    # 将阈值改为float类型以匹配实际使用场景
    threshold = float(threshold)
    
    # 获取实体1和实体2的geoms数量
    geoms1_count = len(entity1.geoms)
    geoms2_count = len(entity2.geoms)
    
    # 打印结果
    print(f"实体1的几何组件数量: {geoms1_count}")
    print(f"实体2的几何组件数量: {geoms2_count}")
    
    # 存储所有符合条件的碰撞信息
    collision_points = []
    min_distance = float('inf')
    min_w1 = None
    min_w2 = None
    min_geom1_idx = -1
    min_geom2_idx = -1
    
    try:
        # 获取实体ID
        entity1_id = int(entity1._idx_in_solver)
        entity2_id = int(entity2._idx_in_solver)
        
        # 遍历实体1的每个几何组件
        for geom1_idx in range(geoms1_count):
            # 遍历实体2的每个几何组件
            for geom2_idx in range(geoms2_count):
                # 计算当前几何组件对之间的距离
                dist, w1, w2 = scene.rigid_solver.collider.compute_distance(
                    entity1_id, 
                    entity2_id, 
                    geom1_idx,  # 指定实体1的几何组件索引
                    geom2_idx   # 指定实体2的几何组件索引
                )
                dist = float(dist)
                
                # 跟踪最小距离及其对应的组件索引
                if dist < min_distance:
                    min_distance = dist
                    min_w1 = w1
                    min_w2 = w2
                    min_geom1_idx = geom1_idx
                    min_geom2_idx = geom2_idx
        
        # 检查最小距离是否小于等于阈值
        if min_distance <= threshold:
            collision_points.append({
                'distance': min_distance,
                'witness_point1': tuple(min_w1),
                'witness_point2': tuple(min_w2),
                'geom1_index': min_geom1_idx,  # 记录最小距离对应的实体1组件
                'geom2_index': min_geom2_idx   # 记录最小距离对应的实体2组件
            })
            return (True, collision_points)
        else:
            return (False, [])
            
    except Exception as e:
        print(f"距离计算错误: {e}")
        return (False, [])
```

SonSang, 2025-08-05:
Hi. So you mean you want to find the exact minimum distance between the input triangle meshes? If that is the case, we do not support such functionality. The GJK algorithm that we use for computing minimum distance assumes that the input shapes are convex. To guarantee such convexity, we run convex decompositions for the input meshes and obtain convex approximations of the input mesh, and then put them into GJK algorithm. If you want to find the exact minimum distance between two input meshes, you have to compute the distance between every triangle pair of the two objects and choose the minimum one, which could be accelerated with BVH. However, Genesis does not support that functionality. You should use another library.

jolley-x, 2025-08-06:
Hello, I've separated them now, and each has generated its own mesh convex surface. Then each convex surface is calculated separately, with the minimum value taken from the results. However, there are still issues with the results. From the visualization interface, I found that 93-98 do not overlap, but the returned result is 0. Below is the code and output.

Model Settings
```python
        mesh = gs.morphs.Mesh(
            file=path,
            scale=1.0,
            pos=position,
            collision=True,
            visualization=True,
            is_free=False,
            decimate=False,

            euler=(-90, 0, 0),
            fixed=True,
        )
``` 

Core computation code

```python
def check_collision(entity1, entity2, scene, threshold: float = 0.0) -> tuple[bool, list]:
    """
    检查两个实体间的碰撞，仅返回距离小于等于阈值的最小距离信息
    实体1有多个几何组件，实体2有1个几何组件
    返回：(是否碰撞, [符合阈值的最小距离信息])
    """
    threshold = float(threshold)
    min_distance = float("inf")  # 初始化最小距离为无穷大
    min_collision_info = None    # 存储最小距离的详细信息

    try:
        # 获取实体的几何组件
        geoms1 = entity1.geoms
        geoms2 = entity2.geoms

        print(f"实体1的几何组件数量: {len(geoms1)}")
        print(f"实体2的几何组件数量: {len(geoms2)}")

        # 遍历实体1的所有几何组件与实体2的所有几何组件
        for geom1_idx, geom1 in enumerate(geoms1):
            for geom2_idx, geom2 in enumerate(geoms2):
                # 计算当前几何组件对之间的距离
                dist, w1, w2 = scene.rigid_solver.collider.compute_distance(
                    geom1._idx,  # 实体1几何组件索引
                    geom2._idx,  # 实体2几何组件索引
                    0            # 碰撞对索引
                )
                dist = float(dist)

                # 仅更新小于当前最小距离的信息
                if dist < min_distance:
                    min_distance = dist
                    min_collision_info = {
                        "distance": dist,
                        "witness_point1": tuple(w1),  # 实体1上的最近点
                        "witness_point2": tuple(w2),  # 实体2上的最近点
                        "geom1_index": geom1_idx,     # 实体1中参与的几何组件索引
                        "geom2_index": geom2_idx,     # 实体2中参与的几何组件索引
                    }

        # 整理结果：仅返回距离小于等于阈值的信息
        if min_collision_info is not None and min_distance <= threshold:
            # 补充碰撞标识（此时一定为True）
            min_collision_info["is_collision"] = True
            return (True, [min_collision_info])
        else:
            # 距离超过阈值或无有效信息时，返回空列表
            return (False, [])

[... 4 lines cut ...]
 # 执行碰撞检测
 collision_state, points = check_collision(entity1, entity2, scene, 30)
``` 

output

```shell
第 82 次迭代结果: {'collision': False, 'min_distance': None, 'details': []}
第 83 次迭代结果: {'collision': False, 'min_distance': None, 'details': []}
第 84 次迭代结果: {'collision': False, 'min_distance': None, 'details': []}
第 85 次迭代结果: {'collision': False, 'min_distance': None, 'details': []}
第 86 次迭代结果: {'collision': False, 'min_distance': None, 'details': []}
第 87 次迭代结果: {'collision': True, 'min_distance': 29.04899024963379, 'details': [{'distance': 29.04899024963379, 'witness_point1': (709.75537109375, 238.923583984375, -3133.54150390625), 'witness_point2': (716.943359375, 210.78292846679688, -3134.07177734375), 'geom1_index': 13, 'geom2_index': 0, 'is_collision': True}]}
第 88 次迭代结果: {'collision': True, 'min_distance': 23.568145751953125, 'details': [{'distance': 23.568145751953125, 'witness_point1': (709.7554321289062, 232.04995727539062, -3133.541748046875), 'witness_point2': (715.587158203125, 209.21875, -3133.971435546875), 'geom1_index': 13, 'geom2_index': 0, 'is_collision': True}]}
第 89 次迭代结果: {'collision': True, 'min_distance': 18.114654541015625, 'details': [{'distance': 18.114654541015625, 'witness_point1': (709.75537109375, 225.1317138671875, -3133.54150390625), 'witness_point2': (714.2377319335938, 207.58346557617188, -3133.8720703125), 'geom1_index': 13, 'geom2_index': 0, 'is_collision': True}]}
第 90 次迭代结果: {'collision': True, 'min_distance': 12.687143325805664, 'details': [{'distance': 12.687143325805664, 'witness_point1': (709.75537109375, 218.1709747314453, -3133.54150390625), 'witness_point2': (712.8947143554688, 205.8805389404297, -3133.77294921875), 'geom1_index': 13, 'geom2_index': 0, 'is_collision': True}]}
第 91 次迭代结果: {'collision': True, 'min_distance': 7.283968925476074, 'details': [{'distance': 7.283968925476074, 'witness_point1': (709.75537109375, 211.16976928710938, -3133.54150390625), 'witness_point2': (711.5577392578125, 204.11358642578125, -3133.67431640625), 'geom1_index': 13, 'geom2_index': 0, 'is_collision': True}]}
第 92 次迭代结果: {'collision': True, 'min_distance': 1.9034790992736816, 'details': [{'distance': 1.9034790992736816, 'witness_point1': (709.7554931640625, 204.13035583496094, -3133.5419921875), 'witness_point2': (710.2265014648438, 202.28640747070312, -3133.57666015625), 'geom1_index': 13, 'geom2_index': 0, 'is_collision': True}]}
第 93 次迭代结果: {'collision': True, 'min_distance': 0.0, 'details': [{'distance': 0.0, 'witness_point1': (718.2117919921875, 202.65093994140625, -3126.556884765625), 'witness_point2': (718.2118530273438, 202.6509552001953, -3126.556884765625), 'geom1_index': 13, 'geom2_index': 0, 'is_collision': True}]}
第 94 次迭代结果: {'collision': True, 'min_distance': 0.0, 'details': [{'distance': 0.0, 'witness_point1': (675.9568481445312, 187.63885498046875, -3137.23583984375), 'witness_point2': (675.9568481445312, 187.63885498046875, -3137.23583984375), 'geom1_index': 13, 'geom2_index': 0, 'is_collision': True}]}
[... cut ...]
第 97 次迭代结果: {'collision': True, 'min_distance': 0.0, 'details': [{'distance': 0.0, 'witness_point1': (729.418212890625, 181.4499969482422, -3117.2177734375), 'witness_point2': (729.418212890625, 181.4499969482422, -3117.2177734375), 'geom1_index': 13, 'geom2_index': 0, 'is_collision': True}]}
第 98 次迭代结果: {'collision': True, 'min_distance': 0.0, 'details': [{'distance': 0.0, 'witness_point1': (592.1908569335938, 172.85134887695312, -4026.490234375), 'witness_point2': (592.1908569335938, 172.85134887695312, -4026.490234375), 'geom1_index': 12, 'geom2_index': 0, 'is_collision': True}]}
```

SonSang, 2025-08-07:
> Hello, I have always kept this option enabled.
entity1 = scene.add_entity(mesh1, vis_mode="collision")
entity2 = scene.add_entity(mesh2, vis_mode="collision")
However, currently there is no collision, so it can't be displayed.

No, the option means to render collision geoms instead of the original geometry. It is not affected by the collision status -- whether or not there is contact, collision geoms are always rendered. 

> If the convex surface is not accurate enough, does that mean we can't obtain accurate collision information?

Yes. The convex decomposition is always an approximation to the original geometry, the collision also has an error. As I told you above, Genesis does not provide functionality for computing distance between "surface" meshes. We only provide distance computation for "convex" geoms.

jolley-x, 2025-08-07:
Hello, is it possible to achieve this through MPR, where only information about precise collisions is returned? Because I notice that MPR does not simplify the model, so what should I do?

SonSang, 2025-08-07:
No, MPR also requires convex geometries. There is no functionality in Genesis that solves your problem precisely.

duburcqa, 2025-08-11:
@jolley-x I'm sorry but Genesis is not the right tool for what you are trying to achieve. Genesis is about fast, batched, multi-physics simulation. As such, collision detection has to be fast rather than exact. This means that it is designed to provide approximate result via convex decomposition and decimation. The default parameters have been tuned to give the best performance in average for most users. Beyond the default parameters, performance and accuracy is not guaranteed anymore. We have limited bandwidth as our team is small, so we cannot provide support for use-cases that are too far from what Genesis has been designed for.

What is supported and what is not has already been clarified by @SonSang, and you got assistance for minimum bounds distance computation, so I'm closing this discussion.

closed 2025-08-11 by duburcqa