Genesis-Embodied-AI/genesis-world #1437: [Bug]: Wrong IK result on simple free joints

yifan-hou, 2025-07-19:
### Bug Description

I have a scene with a floating robot hand. To test the inverse_kinematics function, I set a desired SE3 hand pose, solved IK with it, then use the IK solution to set the hand dofs. I expect the hand pose to be exactly the same as the desired pose, however, the resulted hand pose is very different from the desired pose in orientation, while the position part is the same:

[image]

I attached a 100 line script to reproduce this error. 
[test_ik.zip](https://github.com/user-attachments/files/21326221/test_ik.zip)

Please let me know if I am missing anything. Thanks!

### Steps to Reproduce

* Unzip the attached file
* Further unzip the asset folder
* Change line 69 of test_ik.py to the actual path pointing to `wsg50_finray.xml`
* `python test_ik.py`

### Expected Behavior

* The two visualized frames should coincide.
* The plotted frame poses should be the same.

### Relevant log output

```shell

```

### Environment

System info:
* genesis-world=0.2.1
* python 3.10.16
* Ubuntu 24.04.1 LTS

### Release version or Commit ID

genesis-world=0.2.1

yifan-hou, 2025-07-19:
I see. Thanks for the quick reply. Can I get around the problem by replacing the free joint with 6 manually created joints?

yifan-hou, 2025-07-22:
I have some end effector poses, I want to control the robot to move to these poses. My understanding is that I need to compute the robot dof positions (thus IK), then send those position command via set_dof_positions.

YilingQiao, 2025-08-01:
IK is used to find the configuration of your kinematic chain. Since your robot has a moving base, you can fix the kinematic chain and move only the base. This way, you obtain a constant transformation T from the end-effector to the base. To reach a target position, you just need to move the base to that position with the offset  T

closed 2025-08-01 by YilingQiao