Genesis-Embodied-AI/genesis-world #1453: [Feature]: Setting joint frictionloss

chaitanya1chawla, 2025-07-24:
### What feature or enhancement are you proposing?

API call supports setting friction_ratio for geoms, but I couldn't find information on how to set frictionloss for joints.

Mujoco example: model.dof_frictionloss[dof_id] = 0.2  # Nm

### Motivation

As a roboticist working on sim2real applications, I want to reduce gap between simulated joints and real-world joints.
However, I couldn't find any api calls to set frictionloss.

### Potential Benefit

Better real-world modelling of joints.

### What is the expected outcome of the implementation work?

- [ Applied torque on joints reduced by amount set by frictionloss]

YilingQiao, 2025-07-24:
I can implement this after maybe in 1-2 weeks after an ongoing refactoring.

closed 2025-08-05 by chaitanya1chawla

Yoakshat, 2025-08-12:
Instead of having to specify friction loss in our xml, is it possible to dynamically set it?

For example, Genesis currently has methods like 
entity.set_dofs_damping
entity.set_dofs_armature
entity.set_dofs_stiffness

I looked through the changed files, but I couldn't find any method like this.

duburcqa, 2025-08-13:
@Yoakshat Could you create some dedicated issue for this?