Genesis-Embodied-AI/genesis-world #1466: [Bug]: Finite Element Analysis

1040862575, 2025-07-28:
### Bug Description

After loading an STL model using gs.morphs.Mesh, I want to fix one side of the model,model is in a suspended state.Besides adding gravity to the gs.Scene, I also want to apply a force at a specific position on the top surface of the model to observe the stress distribution. I only know how to load the model and add gravity, but I’m unsure how to add a fixed constraint to a face and apply a force on the top surface.

### Steps to Reproduce

```python
#初始化Genesis，优先使用cpu
    gs.init(backend=gs.cpu)
    #加载stl文件
    mesh1 = gs.morphs.Mesh(
        file="Part1.stl",
        scale=1.0,
        pos=(0,0,0),
    )
#创建场景并添加实体
    scene=gs.Scene(
        show_viewer=True,
        sim_options=gs.options.SimOptions(
        dt=0.01,
        gravity=(0, 0, -9.81),
        ),
    )
entity1 = scene.add_entity(mesh1)
    scene.add_entity(gs.morphs.Plane())
    #scene.add_force_field()
    #构建场景
    scene.build()
    for i in range(10000):
        scene.step()
``` 

### Expected Behavior

want to apply a force at a specific position on the top surface of the model to observe the stress distribution.want to fix one side of the model.

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 24.04, Windows 11 24H2]
- GPU/CPU [e.g. A100, RTX 4090, M3pr, Intel I9-9900k, Ryzen 5900x] (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

main

Milotrince, 2025-07-30:
@YilingQiao could we relabel this as support instead of bug?

Milotrince, 2025-07-30:
@1040862575 FEM fixed constraint was very recently added, you can view the example here:
https://github.com/Genesis-Embodied-AI/Genesis/blob/7c95aaa474b265d59069feace6b015e52bf3316c/examples/fem_hard_and_soft_constraint.py#L114-L117

as of this moment it's for explicit solver only, support for implicit is coming soon
To apply the force, `scene.add_force_field()` is probably not what you want because that applies acceleration on everything. @Libero0809 ?

duburcqa, 2025-07-30:
Closing as a duplicate of https://github.com/Genesis-Embodied-AI/Genesis/discussions/1464. Issues are not meant for support.

closed 2025-07-30 by duburcqa