Genesis-Embodied-AI/genesis-world #1504: [Bug]: render_pointcloud reshape error

mstoisa, 2025-08-05:
### Bug Description

Running `render_pointcloud` method raises error `ValueError: maximum supported dimension for an ndarray is currently 64, found 321`.

### Steps to Reproduce

```
import genesis
genesis.init()
scene = genesis.Scene(
    sim_options=genesis.options.SimOptions(
        gravity=(0.0, 0.0, 0.0),
    )
)
scene.add_entity(genesis.morphs.Sphere())
camera = scene.add_camera(pos=(2.5, 2.5, 2.5))
scene.build()
point_cloud, mask = camera.render_pointcloud()
points = point_cloud[mask].reshape(-1, 3)
```

### Expected Behavior

In `camera.py` lines 544 and 547 (`backproject_depth_to_pointcloud`) the depth `ndarray` is used as reshape argument instead of its shape. Replacing `.reshape((*depth, 3))` with `.reshape((*depth.shape, 3))` in both cases solves the problem for me.

### Relevant log output

```shell
Traceback (most recent call last):
  File "/genesis_bug.py", line 17, in <module>
    point_cloud, mask = camera.render_pointcloud()
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Genesis/genesis/utils/misc.py", line 146, in wrapper
    return method(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Genesis/genesis/vis/camera.py", line 561, in render_pointcloud
    pc, mask = backproject_depth_to_pointcloud(intrinsic_K, depth_arr, cam_pose, world_frame, self.near, self.far)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Genesis/genesis/vis/camera.py", line 544, in backproject_depth_to_pointcloud
    point_cloud_world = point_cloud_world[:, :3].reshape((*depth, 3))
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: maximum supported dimension for an ndarray is currently 64, found 321

[Genesis] [08:20:42] [ERROR] ValueError: maximum supported dimension for an ndarray is currently 64, found 321
[Genesis] [08:20:42] [INFO] 💤 Exiting Genesis and caching compiled kernels...
```

### Environment

- devcontainer from Genesis Dockerfile
- GPU, cuda 12.1

### Release version or Commit ID

current sha: d88faa91b94b144b4e1704b4789fdd64bc37a6c8 (Aug 5th 2025)

ceasor-mao, 2025-08-06:
Hi @duburcqa 
I’d love to help close this issue and have opened PR #1512 
I’ve submitted the patch, but I didn’t include a test case yet.
Could you point me to the best way to add one?
Should I create a new file `tests/test_pointcloud.py`?
Or is there an existing file where this back-projection test should live?
Any guidance on naming conventions or sample assertions would be appreciated—happy to update the PR once I know the right pattern.

ceasor-mao, 2025-08-06:
By the way, this is literally my very first PR on GitHub—this issue looked like the perfect bite-sized task to help me learn the ropes. I’m grateful for any feedback you can share, and I’m excited to keep contributing to the community going forward!

duburcqa, 2025-08-06:
Thank you for your contribution, it is appreciated!

I wrote a small unit test on my side and it turns out the output of the function is actually "wrong", so unfortunately it is not the perfect issue for you to learn how to contribute to an open-source project I'm afraid. I'm going to take over at this point.

> Could you point me to the best way to add one? Should I create a new file tests/test_pointcloud.py? Or is there an existing file where this back-projection test should live?

I would suggest adding the test in `test_render.py`, since it is somewhat related to rendering, even if it is mostly post-processing.

> Any guidance on naming conventions or sample assertions would be appreciated—happy to update the PR once I know the right pattern.

I would suggest that you have a look to my PR https://github.com/Genesis-Embodied-AI/Genesis/pull/1515. The idea is to find to very basic setup where the output could be validated analytically. I decided to go for sphere and box primitive shapes, because box would not be generic enough, and sphere is not very easy to debug in case of issues.