Genesis-Embodied-AI/genesis-world #1511: [Bug]: `envs_idx` is not supported for non-parallelized scene

mahao18cm, 2025-08-06:
### Bug Description

After updating my Genesis, I encountered an error when running my previous code `scene.step()`. The specific error is as follows: `scene.step()`
[Genesis] [10:32:36] [ERROR] GenesisException: `envs_idx` is not supported for non-parallelized scene.

### Steps to Reproduce

If possible, provide a script triggering the bug, e.g.
```python
# code snippet triggering the bug
import genesis as gs
gs.init()
scene = gs.Scene(show_viewer=False)
broken = scene.add_entity(
    gs.morphs.MJCF(file='attachement.xml'),
)
scene.build()
scene.step()
```
Make sure to attached any needed assets (here `attachement.xml`)!

### Expected Behavior

I hope to be able to integrate previous methods or provide new solutions.

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 24.04, Windows 11 24H2]
- GPU/CPU [e.g. A100, RTX 4090, M3pr, Intel I9-9900k, Ryzen 5900x] (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

I am not sure about this. 

### Additional Context

Nothing

duburcqa, 2025-08-06:
Could you share a reproduction script, or at least the entire traceback of the error? At this point I don't have enough information to fix it.

anashoussaini, 2025-08-19:
File "miniconda3/envs/dex/lib/python3.10/site-packages/genesis/vis/visualizer.py", line 201, in update
    self._viewer.update(auto_refresh=auto)
  File "miniconda3/envs/dex/lib/python3.10/site-packages/genesis/vis/viewer.py", line 153, in update
    self.context.update()
  File "miniconda3/envs/dex/lib/python3.10/site-packages/genesis/vis/rasterizer_context.py", line 845, in update
    self.visualizer.update_visual_states()
  File "miniconda3/envs/dex/lib/python3.10/site-packages/genesis/vis/visualizer.py", line 216, in update_visual_states
    camera.move_to_attach()
  File "miniconda3/envs/dex/lib/python3.10/site-packages/genesis/vis/camera.py", line 236, in move_to_attach
    self.set_pose(transform=transform, env_idx=self._env_idx)
  File "miniconda3/envs/dex/lib/python3.10/site-packages/genesis/vis/camera.py", line 567, in set_pose
    env_idx = self._visualizer._scene._sanitize_envs_idx(env_idx)
  File "miniconda3/envs/dex/lib/python3.10/site-packages/genesis/engine/scene.py", line 1236, in _sanitize_envs_idx
    gs.raise_exception("`envs_idx` is not supported for non-parallelized scene.")
  File "miniconda3/envs/dex/lib/python3.10/site-packages/genesis/utils/misc.py", line 39, in raise_exception
    raise gs.GenesisException(msg)
genesis.GenesisException: `envs_idx` is not supported for non-parallelized scene.

[Genesis] [01:20:43] [ERROR] GenesisException: `envs_idx` is not supported for non-parallelized scene.

I get the same issue.

duburcqa, 2025-08-28:
Without reproduction script I cannot really help you. I checked the source code of Genesis, and could not find any bug that could explain your error. I suspect that you are passing `env_idx=0` to `add_camera` when you are building the scene, whereas you build the scene without batching. This is fail by design. In such a case, you cannot specify `env_idx` at all.

closed 2025-08-28 by duburcqa