Genesis-Embodied-AI/genesis-world #1516: [Bug]: draw_debug_line draws wrongly

yifan-hou, 2025-08-06:
### Bug Description

draw_debug_line(A,B) draws the line at wrong location. Also it ignores the color specification. Instead of drawing the line between A and B, it draws a line segment from B to the other direction, as shown below.
``` py
Tl = np.array([-0.12, -0.1975, 0.35  ] )
Tr = np.array([-0.12,  0.1975, 0.35  ] )
Nl = np.array([-0.12, -0.1975, 0.239 ] )
Nr = np.array([-0.12,  0.1975, 0.239 ] )

scene.draw_debug_sphere(Tl, radius=0.03, color=(1.0, 0.0, 0.0, 0.5))
scene.draw_debug_sphere(Tr, radius=0.03, color=(0.0, 1.0, 0.0, 0.5))
scene.draw_debug_sphere(Nl, radius=0.03, color=(1.0, 0.0, 0.0, 0.5))
scene.draw_debug_sphere(Nr, radius=0.03, color=(0.0, 1.0, 0.0, 0.5))

scene.draw_debug_line(Tl, Nl, radius=0.015, color=(0.0, 0.0, 1.0, 0.2))
scene.draw_debug_line(Tr, Nr, radius=0.015, color=(0.0, 1.0, 0.0, 0.2))
scene.draw_debug_line(Nl, Nr, radius=0.01, color=(1.0, 0.0, 0.0, 0.2))
scene.draw_debug_line(Tl, Tr, radius=0.01, color=(1.0, 0.0, 0.0, 0.2))
```

[image]

### Steps to Reproduce

Full script to reproduce:
```python
import numpy as np

import genesis as gs
import numpy as np

gs.init(backend=gs.cpu)

scene = gs.Scene(show_viewer=True,
        viewer_options = gs.options.ViewerOptions(
        res           = (1280, 960),
        camera_pos    = (0, -3.0, 2.5),
        camera_lookat = (0.0, 0.0, 0.5),
        camera_fov    = 40,
        max_FPS       = 60,
    ),
    vis_options = gs.options.VisOptions(
        show_world_frame = True, # visualize the coordinate frame of `world` at its origin
        world_frame_size = 1.0, # length of the world frame in meter
        show_link_frame  = False, # do not visualize coordinate frames of entity links
        show_cameras     = False, # do not visualize mesh and frustum of the cameras added
        plane_reflection = True, # turn on plane reflection
        ambient_light    = (0.1, 0.1, 0.1), # ambient light setting
    ),
    renderer = gs.renderers.Rasterizer(), # using rasterizer for camera rendering
)

plane = scene.add_entity(gs.morphs.Plane())
scene.profiling_options.show_FPS=False
scene.build()

Tl = np.array([-0.12, -0.1975, 0.35  ] )
Tr = np.array([-0.12,  0.1975, 0.35  ] )
Nl = np.array([-0.12, -0.1975, 0.239 ] )
Nr = np.array([-0.12,  0.1975, 0.239 ] )

scene.draw_debug_sphere(Tl, radius=0.03, color=(1.0, 0.0, 0.0, 0.5))
scene.draw_debug_sphere(Tr, radius=0.03, color=(0.0, 1.0, 0.0, 0.5))
scene.draw_debug_sphere(Nl, radius=0.03, color=(1.0, 0.0, 0.0, 0.5))
scene.draw_debug_sphere(Nr, radius=0.03, color=(0.0, 1.0, 0.0, 0.5))

scene.draw_debug_line(Tl, Nl, radius=0.015, color=(0.0, 0.0, 1.0, 0.2))
scene.draw_debug_line(Tr, Nr, radius=0.015, color=(0.0, 1.0, 0.0, 0.2))
scene.draw_debug_line(Nl, Nr, radius=0.01, color=(1.0, 0.0, 0.0, 0.2))
scene.draw_debug_line(Tl, Tr, radius=0.01, color=(1.0, 0.0, 0.0, 0.2))

while True:
    scene.step()

```

### Expected Behavior

The script should draw a  closed square, with lines connecting the four corners.

### Relevant log output

```shell

```

### Environment

- OS: Ubuntu 24.04
- GPU/CPU RTX 4090/Ryzen 9 7950X
- Driver Version: 550.144.03
- CUDA Version: 12.4

### Release version or Commit ID

Genesis commit 91b099cb22678807949013b443497b84f1ac1c83

duburcqa, 2025-08-08:
Thanks for the report, we will look into this!

yifan-hou, 2025-08-09:
That is weird, I just pulled and installed genesis main (I see it is 0.3.0 now) and ran the above script again, I still got the wrong result:

[image]

I wonder if it is related to any dependency version? I have the following version in my env:

Name: PyOpenGL
Version: 3.1.9

Name: opencv-python
Version: 4.11.0.86

duburcqa, 2025-08-11:
I can confirm this is broken.