Genesis-Embodied-AI/genesis-world #1529: [Feature]: human tele operations support

yanan1116, 2025-08-09:
### What feature or enhancement are you proposing?

hi authors, 

i haven’t found any related documents about how human can do tele operation for collecting demonstrative trajectories in the scene, either via keyboard or some other devices. 
as you know, the demos can be used for downstream imitation learning. 

Will you support it in the future? 
are there any workarounds in current version ? 

thanks 

### Motivation

after customized scenes creation, we may need to collect expert demos via devices and control the robots in the simulation environment 

### Potential Benefit

provide interface for expert trajectories collection 

### What is the expected outcome of the implementation work?

similar to issac sim https://isaac-sim.github.io/IsaacLab/main/source/overview/teleop_imitation.html

Kashu7100, 2025-08-09:
Can you check this?: https://github.com/Genesis-Embodied-AI/Genesis/blob/main/examples/keyboard_teleop.py

closed 2025-08-13 by yanan1116

yanan1116, 2026-01-15:
the code to collect the dataset above:

```python

"""
https://github.com/Genesis-Embodied-AI/Genesis/blob/main/examples/keyboard_teleop.py

Keyboard Controls:
↑	- Move Forward (North)
↓	- Move Backward (South)
←	- Move Left (West)
→	- Move Right (East)
n	- Move Up
m	- Move Down
j	- Rotate Counterclockwise
k	- Rotate Clockwise
u	- Reset Scene
space	- Press to close gripper, release to open gripper
esc	- Quit
"""

import os,sys,argparse,time
import random
import threading,uuid

import genesis as gs
import numpy as np
from pynput import keyboard
from scipy.spatial.transform import Rotation as R
from PIL import Image
print('gs ver:', gs.__version__, gs.__file__)
assert gs.__version__ == '0.3.10'
from colorama import Fore, Style
from utils import *

parser = argparse.ArgumentParser()
parser.add_argument(
        "--save",
        action='store_true',
    )

parser.add_argument(
        "--save_image",
        action='store_true',
    )

parser.add_argument(
        "--vla",
        action='store_true',
    )

parser.add_argument(
        "--delta",
        action='store_true',
    )

parser.add_argument(
        "--cpu",
        action='store_true',
    )

parser.add_argument(
    "--ws",
    type=str,
    default="10.225.68.29",
    # required=True
)
parser.add_argument(
    "--horizon",
    type=int,
[... 399 lines cut ...]

    
    
def main():

    client_keyboard = KeyboardDevice()
    client_keyboard.start()  

    if args.vla:
        from openpi_client import websocket_client_policy
        policy = websocket_client_policy.WebsocketClientPolicy(
            host= args.ws,
            port=8000,
            api_key=None,
        )
        args.save=False
    else:
        policy = None 

    run_sim(client_keyboard, policy)

if __name__ == "__main__":
    main()

```