Genesis-Embodied-AI/genesis-world #1569: [Bug]: MJCF custom XML joint with frictionloss causes global objects to pass through Plane and lose collisions

KzzzzzO, 2025-08-15:
### Bug Description

When importing the SoArm100 model provided by the mujoco_menagerie project into a Genesis scene, all global entities (including the robotic arm itself and blocks) fall through the plane, and the robotic arm does not exhibit proper collision behavior. After investigation, it was found that the issue is caused by the frictionloss parameter defined on the joints of the robotic arm in the XML. Removing the frictionloss parameter restores normal collision behavior.

### Steps to Reproduce

Use the following Genesis code:
```python
import genesis as gs

gs.init(backend=gs.gpu)

scene = gs.Scene(show_viewer=True,)
plane = scene.add_entity(morph = gs.morphs.Plane(),)
cube = scene.add_entity(
    material=gs.materials.Rigid(),
    morph=gs.morphs.Box(pos=(0.0, 0.0, 1.0), size=(0.2, 0.2, 0.2)),
    surface=gs.surfaces.Default(color=(1.0, 0.0, 0.0), vis_mode='visual')
)

RigidEntity = scene.add_entity(gs.morphs.MJCF(file='./trs_so_arm100/so_arm100.xml', pos=(0.5, 0.5, 0.0)),)

scene.build(n_envs=0, env_spacing=(0.5, 0.5))

while True:
    scene.step()
```
Content of ./trs_so_arm100/so_arm100.xml. Use the following so_arm100.xml model:
```xml
<mujoco model="so_arm100">
  <compiler angle="radian" meshdir="assets/"/>

  <option cone="elliptic" impratio="10"/>

  <asset>
    <material name="white" rgba="1 1 1 1"/>
    <material name="black" rgba="0.1 0.1 0.1 1"/>

    <mesh name="Base" file="Base.stl"/>
    <mesh name="Base_Motor" file="Base_Motor.stl"/>
    <mesh name="Rotation_Pitch" file="Rotation_Pitch.stl"/>
    <mesh name="Rotation_Pitch_Motor" file="Rotation_Pitch_Motor.stl"/>
    <mesh name="Upper_Arm" file="Upper_Arm.stl"/>
    <mesh name="Upper_Arm_Motor" file="Upper_Arm_Motor.stl"/>
    <mesh name="Lower_Arm" file="Lower_Arm.stl"/>
    <mesh name="Lower_Arm_Motor" file="Lower_Arm_Motor.stl"/>
    <mesh name="Wrist_Pitch_Roll" file="Wrist_Pitch_Roll.stl"/>
    <mesh name="Wrist_Pitch_Roll_Motor" file="Wrist_Pitch_Roll_Motor.stl"/>
    <mesh name="Fixed_Jaw" file="Fixed_Jaw.stl"/>
    <mesh name="Fixed_Jaw_Motor" file="Fixed_Jaw_Motor.stl"/>
    <mesh name="Fixed_Jaw_Collision_1" file="Fixed_Jaw_Collision_1.stl"/>
    <mesh name="Fixed_Jaw_Collision_2" file="Fixed_Jaw_Collision_2.stl"/>
    <mesh name="Moving_Jaw" file="Moving_Jaw.stl"/>
    <mesh name="Moving_Jaw_Collision_1" file="Moving_Jaw_Collision_1.stl"/>
    <mesh name="Moving_Jaw_Collision_2" file="Moving_Jaw_Collision_2.stl"/>
    <mesh name="Moving_Jaw_Collision_3" file="Moving_Jaw_Collision_3.stl"/>
  </asset>

  <default>
    <default class="so_arm100">
      <joint frictionloss="0.1" armature="0.1"/>
      <position kp="50" dampratio="1" forcerange="-3.5 3.5"/>
      <default class="Rotation">
        <joint axis="0 1 0" range="-1.92 1.92"/>
      </default>
      <default class="Pitch">
        <joint axis="1 0 0" range="-3.32 0.174"/>
      </default>
      <default class="Elbow">
        <joint axis="1 0 0" range="-0.174 3.14"/>
[... 137 lines cut ...]
[Genesis] [12:51:20] [INFO] Building scene <111b995>...
[Genesis] [12:51:23] [INFO] Compiling simulation kernels...
[Genesis] [12:51:29] [INFO] Building visualizer...
[Genesis] [12:51:30] [INFO] Viewer created. Resolution: 684×513, max_FPS: 60.
[Genesis] [12:51:30] [INFO] Running at 59.70 FPS.
[Genesis] [12:51:30] [INFO] Running at 59.64 FPS.
......
```

### Environment

- OS: WSL Ubuntu 24.04
- GPU/CPU RTX 3060, Intel I9-9900k
- GPU-driver version Driver Version: 580.88
- CUDA / CUDA-toolkit version CUDA11.8

### Release version or Commit ID

genesis-world 0.3.0 （clone in main branch）

### Additional Context

Tried this on other exmple XML model (by adding frictionloss="0.1" to panda.xml), same issue comes out.
Here's an image showing the expected result.
[image]

KzzzzzO, 2025-08-15:
I have seen other issues where someone successfully ran a model with `frictionloss` ([https://github.com/Genesis-Embodied-AI/Genesis/issues/1559]). According to their description, they did not encounter any problems at all. Therefore, I also suspect that the issue might be caused by some mistake I made while modifying the XML file.

duburcqa, 2025-08-15:
Can you try with the latest main branch? This issue has been fixed very recently.

closed 2025-08-15 by duburcqa