Genesis-Embodied-AI/genesis-world #1583: [Bug]: Stuck on this step all the time：[Genesis] [16:26:41] [INFO] Building visualizer...

ting-zheng, 2025-08-17:
### Bug Description

Stuck on this step all the time：[Genesis] [16:26:41] [INFO] Building visualizer...
when i input ctrl+c,i get the error:
^CTraceback (most recent call last):
  File "/home/zzw/Genesis/test/test.py", line 10, in <module>
    scene.build()
  File "/home/zzw/miniconda3/envs/genesis/lib/python3.10/site-packages/genesis/utils/misc.py", line 136, in wrapper
    return method(self, *args, **kwargs)
  File "/home/zzw/miniconda3/envs/genesis/lib/python3.10/site-packages/genesis/engine/scene.py", line 708, in build
    self._visualizer.build()
  File "/home/zzw/miniconda3/envs/genesis/lib/python3.10/site-packages/genesis/vis/visualizer.py", line 177, in build
    self._viewer.build(self._scene)
  File "/home/zzw/miniconda3/envs/genesis/lib/python3.10/site-packages/genesis/vis/viewer.py", line 89, in build
    self._pyrender_viewer = pyrender.Viewer(
  File "/home/zzw/miniconda3/envs/genesis/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 398, in __init__
    self._initialized_event.wait()
  File "/home/zzw/miniconda3/envs/genesis/lib/python3.10/threading.py", line 607, in wait
    signaled = self._cond.wait(timeout)
  File "/home/zzw/miniconda3/envs/genesis/lib/python3.10/threading.py", line 320, in wait
    waiter.acquire()
KeyboardInterrupt

[Genesis] [16:28:01] [ERROR] KeyboardInterrupt: 
[Genesis] [16:28:01] [INFO] 💤 Exiting Genesis and caching compiled kernels...

### Steps to Reproduce

python3 examples/tutorials/control_your_robot.py 

### Expected Behavior

Have a graphical interface that appears

### Relevant log output

```shell

```

### Environment

- OS: [Ubuntu 22.04]
- GPU/CPU [RTX 3060, Intel I7] (N/A if no GPU/CPU)
- GPU-driver version    NVIDIA-SMI 580.76.05    (N/A if no GPU)
- CUDA / CUDA-toolkit version  cuda_12.1.r12.1/compiler.32415258_0 (N/A if non-Nvidia)

### Release version or Commit ID

1011712f1873e5644cc0b5462188a5272eaa8e93

kentliukx, 2025-08-17:
Ctrl+c means stop the program! If you want to paste, use Ctrl+Shift+c instead.

bk-poudel, 2025-08-29:
I am having the same issue

duburcqa, 2025-08-30:
@bk-poudel this issue has been fixed in the latest version available on pypi (release yesterday). Could you check if it works now?

bk-poudel, 2025-08-30:
> [@bk-poudel](https://github.com/bk-poudel) this issue has been fixed in the latest version available on pypi (release yesterday). Could you check if it works now?

It works now thank you

duburcqa, 2025-08-31:
Yeah! Good news!