Genesis-Embodied-AI/genesis-world #1601: [Bug]: Robot dosent collide with the ground falls down even with gravity disabled

vybhav-ibr, 2025-08-19:
### Bug Description

I tried to load the apptronik appolo from an MJCF file and the robot doesn't collide with the ground, and falls down indefinitely even with gravity disabled.

### Steps to Reproduce

import genesis as gs

gs.init(logging_level="debug")

scene = gs.Scene(
    show_viewer=True,
    viewer_options=gs.options.ViewerOptions(
        res=(1280, 960),
        camera_pos=(3.5, 0.0, 2.5),
        camera_lookat=(0.0, 0.0, 0.5),
        camera_fov=40,
        max_FPS=60,
    ),
    vis_options=gs.options.VisOptions(
        show_world_frame=True,
        world_frame_size=1.0,
        show_link_frame=False,
        show_cameras=False,
        plane_reflection=True,
        ambient_light=(0.1, 0.1, 0.1),
    ),
    rigid_options=gs.options.RigidOptions(
    enable_joint_limit=False,
    enable_collision=False,
    gravity=(0, 0, 0),
),
    # renderer=gs.renderers.RayTracer(),
    renderer=gs.renderers.Rasterizer(),
)

plane = scene.add_entity(
    gs.morphs.Plane(),
)
applolo = scene.add_entity(
    gs.morphs.MJCF(file="/apptronik_appolo/apptronik_apollo.xml",
                   pos=(0,0,5),
                   collision=True),
)

cam = scene.add_camera(
    res=(640, 480),
    pos=(3.5, 0.0, 2.5),
    lookat=(0, 0, 0.5),
    fov=30,
    GUI=True,
)

scene.build()

# render rgb, depth, segmentation, normal
rgb, depth, segmentation, normal = cam.render(rgb=True, depth=True, segmentation=True, normal=True)

cam.start_recording()
import numpy as np

for i in range(1200):
    scene.step()
    # cam.set_pose(
    #     pos=(3.0 * np.sin(i / 60), 3.0 * np.cos(i / 60), 2.5),
    #     lookat=(0, 0, 0.5),
    # )
    cam.render()
cam.stop_recording(save_to_filename="video.mp4", fps=60)

MJCF file:
[https://github.com/google-deepmind/mujoco_menagerie/tree/main/apptronik_apollo](url)

### Expected Behavior

The robot must ragdoll on the ground

### Relevant log output

```shell
[Genesis] [20:02:33] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [20:02:33] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [20:02:33] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [20:02:33] [INFO] Consider setting 'performance_mode=True' in production to maximise runtime speed, if significantly increasing compilation time is not a concern.
[Genesis] [20:02:34] [INFO] Running on [NVIDIA GeForce RTX 3050 Laptop GPU] with backend gs.cuda. Device memory: 3.80 GB.
[Genesis] [20:02:34] [DEBUG] [Taichi] version 1.7.4, llvm 15.0.4, commit b4b956fd, linux, python 3.10.12
[Genesis] [20:02:34] [DEBUG] [Taichi] Starting on arch=cuda
[Genesis] [20:02:34] [INFO] 🚀 Genesis initialized. 🔖 version: 0.3.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [20:02:34] [INFO] Scene <f2b2ac9> created.
[Genesis] [20:02:34] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <05fe0ff>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [20:02:34] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 0.
[Genesis] [20:02:34] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <0539e18>, morph: <gs.morphs.MJCF(file='/home/vybhav/Videos/apptronik_appolo/apptronik_apollo.xml')>, material: <gs.materials.Rigid>.
[Genesis] [20:02:37] [INFO] Applying offset to base link's pose with user provided value in morph.
[Genesis] [20:02:37] [INFO] Applying offset to base link's pose with user provided value in morph.
[Genesis] [20:02:38] [INFO] Building scene <f2b2ac9>...
[Genesis] [20:02:39] [DEBUG] Constraint solver time constant not specified. Using default value (`0.02`).
[Genesis] [20:02:39] [DEBUG] Constraint solver time constant not specified. Using default value (`0.02`).
[Genesis] [20:02:42] [INFO] Compiling simulation kernels...
[Genesis] [20:02:45] [INFO] Building visualizer...
[Genesis] [20:02:45] [DEBUG] Trying to create OpenGL Context for PYOPENGL_PLATFORM='native'...
[Genesis] [20:02:59] [INFO] Viewer created. Resolution: 1280×960, max_FPS: 60.
[Genesis] [20:03:00] [INFO] Running at 18.52 FPS.
[Genesis] [20:03:00] [INFO] Running at 18.43 FPS.
```

### Environment

- OS: Ubuntu 22.04
- GPU/CPU RTX 3050 MOBILE
- CUDA / CUDA-toolkit version 12.4

### Release version or Commit ID

0.3.1

### Additional Context

https://github.com/user-attachments/assets/80ece0b9-97d7-484f-aac1-37af1efee378

Kashu7100, 2025-08-19:
Why are you setting `enable_collision=False`?

vybhav-ibr, 2025-08-19:
That was just for testing to see if it made any difference(It did not). The robot exhibits the same behavior even with the collisions enabled

Kashu7100, 2025-08-20:
Ok, I found a bug. It seems gravity setting from RigidOption is not taking an effect (SimOption is used currently).

vybhav-ibr, 2025-08-20:
Looks like the collision issue and the gravity issue are unrelated. The collision issue seems to be a MJCF file issue or the MJCF loader issue in genesis, I don't have much experience using MJCF to pinpoint the issue, could you investigate further and provide a solution?. Thanks!

closed 2025-08-22 by vybhav-ibr