Genesis-Embodied-AI/genesis-world #1605: MuJoCo Cable doesn't work in Genesis

Yoakshat, 2025-08-19:
### Bug Description

I am trying to transfer my cable attached to a ball from Mujoco to Genesis. It works in MuJoCo but acts really wonky in Genesis, specifically when I set initial="free" in my cable composite. 

Here are the side-to-side videos: 

MuJoCo: 
https://github.com/user-attachments/assets/430b49c8-ee1c-4ba5-a1bd-f6a1507ea9a5

Genesis: 
https://github.com/user-attachments/assets/589b21a2-ab1c-423c-a44c-a77bf14e7313

### Steps to Reproduce

The stand-alone cable xml is here. 

<mujoco model="ball_cable">
    <extension>
        <plugin plugin="mujoco.elasticity.cable"/>
    </extension>
    <compiler angle="radian" meshdir="meshes/" autolimits="false"/>
    <option impratio="20" timestep="0.001" jacobian="dense" />

    <default>
        <geom contype="0" conaffinity="0" group="0" />
        
        <default class="conty">
            <geom contype="1" conaffinity="0" condim="6" group="1" friction="10 10 10" solref="0.004 1" solimp="9.88 8 0.001"/>
        </default>
        <default class="conaff">
            <geom contype="0" conaffinity="1" condim="6" group="1" friction="0.4 0.002 0.00008" solref="0.004 1" />
        </default>
    </default>

    <worldbody>
        
        <body name="obj" pos="-0.38 -0.26 0.05">
            <joint name="obj" type="free" />
            <geom name="obj" priority="10" class="conty" rgba="0.2 0.2 0.9 1" type="sphere" size=".01" mass="0.0005"/>
            <site name='end2' rgba="0.2 0.2 0.9 1" type="sphere" size="0.002"/>
        </body>

        <body name="anchor" pos="-0.38 -0.26 1.07"> 
            <geom type="sphere" size="0.01" rgba="1 0 0 1" contype="0" conaffinity="0" /> 
        </body> 
            
        <composite type="cable" count="26 1 1" curve="0 0 s" offset="-0.38 -0.26 0.06" initial="free">
            <joint kind="main" damping="0.01" armature="0.001"/>
            <geom type="capsule" size=".005 .007" rgba="0.8 0.6 0.4 1" mass="0.001"/>
        </composite> 
    </worldbody>
  
    <equality>
        <weld body1="B_last" body2="anchor"/> 
        <weld body1="B_first" body2="obj"/>
    </equality>
</mujoco>

### Expected Behavior

Expected to act like a regular cable attached to a ball, that doesn't go flying.

### Relevant log output

```shell

```

### Environment

- Mac M2 (on Monterey)

### Release version or Commit ID

Version 0.3.1

duburcqa, 2025-08-22:
I can reproduce the issue. Working on a fix.

Yoakshat, 2025-08-22:
I don't know why, but I get "ERROR: Could not find a version that satisfies the requirement gstaichi==1.0.1 (from genesis-world)" after checking out this pr and installing dependencies.

duburcqa, 2025-08-24:
Can you share a complete reproduction script? Because your cable simulation is working just fine on my side.

duburcqa, 2025-08-25:
This is unfortunate! This scenario was never unit tested indeed. I will have a look asap.