Genesis-Embodied-AI/genesis-world #1622: [Bug]: No genesis-world version compatible with gstaichi==1.0.1

Yoakshat, 2025-08-22:
### Bug Description

When installing genesis 0.3.1, I get the error: 'can't find a genesis-version compatible with gstaichi==1.0.1.'

### Steps to Reproduce

Create a virtual env with python==3.10, activate, and then run following commands

`conda install pytorch torchvision torchaudio -c pytorch`
`cd Genesis`
`pip install -e .`

### Expected Behavior

Expected to install without failure

### Relevant log output

```shell
INFO: pip is looking at multiple versions of genesis-world to determine which version is compatible with other requirements. This could take a while.
ERROR: Could not find a version that satisfies the requirement gstaichi==1.0.1 (from genesis-world) (from versions: none)
ERROR: No matching distribution found for gstaichi==1.0.1
```

### Environment

- Mac OS M2 (Monterey)
- Python 3.10

### Release version or Commit ID

Genesis 0.3.1

closed 2025-08-27 by duburcqa

Yoakshat, 2025-08-27:
Oh I would, but I already upgraded to MacOS 15

duburcqa, 2025-08-28:
Damn, I was expecting that! Too bad.