Genesis-Embodied-AI/genesis-world #1629: [Feature]: Should mesh conversion include pre-checks or automatic repair for tetrahedralization?

ceasor-mao, 2025-08-25:
### What feature or enhancement are you proposing?

The current tetrahedralization function in `genesis/utils/mesh.py` directly passes the mesh to tetgen without any prior validity check:
```Python
def tetrahedralize_mesh(mesh, tet_cfg):
    # Importing pyvista and tetgen are very slow and not used very often. Let's delay import.
    import pyvista as pv
    import tetgen

    pv_obj = pv.PolyData(
        mesh.vertices, np.concatenate([np.full((mesh.faces.shape[0], 1), mesh.faces.shape[1]), mesh.faces], axis=1)
    )
    tet = tetgen.TetGen(pv_obj)
    # Build and apply the switches string directly, since
    # the Python wrapper sometimes ignores certain kwargs
    # (e.g. maxvolume). See: https://github.com/pyvista/tetgen/issues/24
    switches = make_tetgen_switches(tet_cfg)
    verts, elems = tet.tetrahedralize(switches=switches)
    # visualize_tet(tet, pv_obj, show_surface=False, plot_cell_qual=False)
    return verts, elems
```

### Motivation

While investigating issue #1093, I found that the provided test.obj file was not watertight (had 3270 boundary edges) and consisted of many disconnected components. This caused the tetrahedralization to fail silently or with unclear error messages.
This raises an important question: Should Genesis perform automatic mesh validity checks (or even repairs) before attempting FEM/MPM simulation setup?

### Potential Benefit

Improved usability

### What is the expected outcome of the implementation work?

- Users will receive clear, actionable error messages

duburcqa, 2026-05-23:
Now that a robust mesh repair routine has been added to Genesis, it should be straightforward to fix this issue.

jeetrex17, 2026-08-17:
Hi, I'd like to work on this.

 I traced the current tetrahedralization paths first. FEM entities and the SAP/hydroelastic coupler use `mesh_to_elements()` and its `.tet` cache, while `PBD3DEntity` calls `Mesh.tetrahedralize()`, which reaches the lower-level TetGen wrapper without that cache.

My proposed approach is to add a shared pre-TetGen validation step, keeping the current path unchanged for surfaces that already satisfy the tetrahedralization requirements. Invalid inputs should receive actionable diagnostics, such as open boundaries or disconnected components. When repair applies, I would reuse the pipeline from #2816, validate its output, and ensure the tet cache is keyed on the geometry actually passed to TetGen.

Two design questions:

1. Should repair happen automatically for tetrahedralization, or should it be opt-in through a new tetrahedralization-specific option? I would avoid silently reusing the existing `watertighten` option because it is documented as rigid-collision-only.

2. FEM-specific: its visual-to-simulation mapping assumes the original surface vertices remain present and ordered, while repair replaces that topology. Should the repaired surface also become the render mesh, or should the original visual mesh be retained and explicitly re-mapped onto the repaired geometry?